realsenseai/librealsense #10093: D415 reports an error and cannot be used

gaoxiaohuiandy, 2021-12-22:
I have two D415, both with shells. I have been using them normally for a while. Today, when I use one of them, it keeps reporting errors and the camera is unusable. I found that its serial number has changed to ffffffffffff, it was not such a serial number before. I switched the cables, the other one can be used normally, but this one still cannot be used. There should be no problem with the cable. Is there any solution?

| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D400|
|**Serial Number**|ffffffffffff|
|**Firmware Version**|05.12.15.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|041323020901|
|**Firmware Update Id**|041323020901|

RealSense error calling:
Invalid Value in rs2_open_multiple(sensor:000002A9F9CD2AB0,
profiles:000002A9FA465AB8, count:1):
hwmon command 0x2c( 9 0 0 0 ) failed (response -9= No expected user action)

MartyG-RealSense, 2021-12-22:
Hi @gaoxiaohuiandy  This is a very rare error.  A good starting point for investigating your case will be to direct you to a similar D415 case at the link below so you can see whether there are any aspects of that case that are similiar to your own situation.

https://support.intelrealsense.com/hc/en-us/community/posts/4411008428307-D415-lost-its-brain-

gaoxiaohuiandy, 2021-12-22:
@MartyG-RealSense Thank you very much for your reply! Yes, my situation is the same as his, let me try again to see if there is a solution.

closed 2022-01-10 by MartyG-RealSense