realsenseai/librealsense #10094: a problem running two cameras in CLion IDE - address exception

ilyak93, 2021-12-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | SDK | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.0                           | 
| Language                            |  C++                          | 
 

### Issue Description
I'm using the simple part of the code from realsense examples solution for two my realsense cameras D455:

[image]

this is the code from the examples:

    rs2::context                          ctx;        // Create librealsense context for managing devices
    
    std::map<std::string, rs2::colorizer> colorizers; // Declare map from device serial number to colorizer (utility class to convert depth data RGB colorspace)
    
    std::vector<rs2::pipeline>            pipelines;
    
    // Capture serial numbers before opening streaming
    std::vector<std::string>              serials;
    for (auto&& dev : ctx.query_devices())
        serials.push_back(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

When I run "multicam" example from Visual Studio 2019 IDE it works and I see both cameras,
but when I run my code from CLion IDE, I get the following error:
  `Exception: Exception 0xe06d7363 encountered at address 0x7ffe94404f69`

on the second iteration on variable `devs`

What can be the problem and how can I solve it ?
Thank you in advance.

ilyak93, 2021-12-22:
It happens when there is a bad connection of the usb, closing this issue.

closed 2021-12-22 by ilyak93

MartyG-RealSense, 2021-12-22:
Hi @ilyak93  It's great to hear that you quickly found a solution - thanks very much for the update!