realsenseai/librealsense #10095: a problem running two cameras in CLion IDE - address exception

ilyak93, 2021-12-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | SDK | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.0                           | 
| Language                            |  C++                          | 
 

### Issue Description
I'm trying to use the code from realsense examples solution for two of mine realsense cameras D455:

    window app(1280, 960, "CPP Multi-Camera Example");
    
        rs2::context                          ctx;        // Create librealsense context for managing devices
    
        std::map<std::string, rs2::colorizer> colorizers; // Declare map from device serial number to colorizer (utility class to convert depth data RGB colorspace)
    
        std::vector<rs2::pipeline>            pipelines;
    
        // Capture serial numbers before opening streaming
        std::vector<std::string>              serials;
        for (auto&& dev : ctx.query_devices())
            serials.push_back(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
    
        // Start a streaming pipe per each connected device
        for (auto&& serial : serials)
        {
            rs2::pipeline pipe(ctx);
            rs2::config cfg;
            cfg.enable_device(serial);
            pipe.start(cfg);
            pipelines.emplace_back(pipe);
            // Map from each device's serial number to a different colorizer
            colorizers[serial] = rs2::colorizer();
        }

  

I want to run my two cameras with defined callbacks as it is done in the example of callbacks:
    
    struct Frame {
            Frame(const void *pVoid, double d, int i, int w, int h, short bpp,
                  int sib) : ts(d), data_size(i),
                             width(w), height(h), bytes_per_pixel(bpp),
                             stride_in_bytes(sib){
                frame_data = new uint8_t[w*h*bytes_per_pixel];
                assert(frame_data != nullptr);
                memcpy(frame_data, pVoid, w*h*bytes_per_pixel);
            }
    
            //~Frame(){
            //    delete this->frame_data;
            //}
    
            void* frame_data = NULL;
            double ts = -1;
            int data_size = -1;
            int width = -1;
            int height = -1;
            short bytes_per_pixel = -1;
            int stride_in_bytes = -1;
    
        };
    
        rs2::context ctx;
    
        std::vector<rs2::pipeline> pipelines;
    
        std::vector<std::string> serials;
        auto devs = ctx.query_devices();
        for (auto&& dev : devs)
            serials.push_back(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
    
        //std::map<int, std::vector<Frame>> frames;
        std::map<int, std::vector<Frame>> frames;
        std::mutex mutex;
        auto callback = [&](const rs2::frame& frame)
        {
            //std::lock_guard<std::mutex> lock(mutex);
            if (rs2::frameset fs = frame.as<rs2::frameset>()) {
                //rs2::disparity_transform disparity2depth(false);
                //fs = fs.apply_filter(disparity2depth);
                // With callbacks, all synchronized stream will arrive in a single frameset
                for (const rs2::frame f : fs) {
                    auto vf = f.as<rs2::video_frame>();
                    Frame my_f = Frame(vf.get_data(), vf.get_timestamp(),
                                       vf.get_data_size(), vf.get_width(),
                                       vf.get_height(), vf.get_bytes_per_pixel(),
                                       vf.get_stride_in_bytes());
                    frames[f.get_profile().unique_id()].push_back(my_f);
                }
            } //else {
                // Stream that bypass synchronization (such as IMU) will produce single frames
                //frames_bypass.push_back(frame);
            //}
        };
        std::map<int, std::vector<Frame>> frames2;
        std::mutex mutex2;
        auto callback2 = [&](const rs2::frame& frame)
        {
            //std::lock_guard<std::mutex> lock(mutex);
            if (rs2::frameset fs = frame.as<rs2::frameset>()) {
                //rs2::disparity_transform disparity2depth(false);
                //fs = fs.apply_filter(disparity2depth);
                // With callbacks, all synchronized stream will arrive in a single frameset
                for (const rs2::frame f : fs) {
                    auto vf = f.as<rs2::video_frame>();
                    Frame my_f = Frame(vf.get_data(), vf.get_timestamp(),
                                       vf.get_data_size(), vf.get_width(),
                                       vf.get_height(), vf.get_bytes_per_pixel(),
                                       vf.get_stride_in_bytes());
                    frames2[f.get_profile().unique_id()].push_back(my_f);
                }
            } //else {
            // Stream that bypass synchronization (such as IMU) will produce single frames
            //frames_bypass.push_back(frame);
            //}
        };
    
    
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        cfg.enable_device(serials[0]);
        rs2::pipeline_profile profiles = pipe.start(cfg, callback);
        pipelines.emplace_back(pipe);
    
        rs2::pipeline pipe2(ctx);
        rs2::config cfg2;
        cfg2.enable_device(serials[1]);
        rs2::pipeline_profile profiles2 = pipe2.start(cfg, callback2);
        pipelines.emplace_back(pipe2);
    
    

When I run "multicam" example from Visual Studio 2019 IDE it works and I see both cameras,
but when I run my code from CLion IDE, I get the following error:
  `Exception: Exception 0xe06d7363 encountered at address 0x7ffe94404f69`

on the pipe2 start:
`rs2::pipeline_profile profiles2 = pipe2.start(cfg, callback2);`

What can be the problem and how can I solve it ?
Thank you in advance.

MartyG-RealSense, 2021-12-22:
Hi @ilyak93  Is this the same question that you posted in https://github.com/IntelRealSense/librealsense/issues/10094 and closed or is it a different question, please?  Thanks!

ilyak93, 2021-12-22:
@MartyG-RealSense, the error somehow is the same, but its a different. I've encountered this right after the previous one solved by re-plugging the usb. Now I debug it, so nearly I'll edit it with additional information. Thank you.

ilyak93, 2021-12-22:
I've figure it out, the unique identifiers of the Depth and Color frames has changed, sorry :)
If you do everything genuinely, it works.

closed 2021-12-22 by ilyak93

MartyG-RealSense, 2021-12-22:
Thanks again @ilyak93   :)