realsenseai/librealsense #10097: Ubuntu 18.04, registering public key fails.

Gaurav37, 2021-12-23:
I am using a ROS Melodic on Ubuntu with kernel 5.4.
I have tried reinstalling librealsense and realsense SDK with almost all methods prescribed on website.

**Question 1:**
If I install by command: sudo apt-get install ros-melodic-realsense2-camera  
and try realsense-viewer after connecting realsense, it says: "Timestamp: 1640229877414.929443
Severity: Warn
Description: Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!
"
Which of the above files should I remove?
[image]

**Question 2**
First step registering public key fails

[image]

Gaurav37, 2021-12-23:
As suggested I tried "export http_proxy="http://<my_proxy>:<my_port>"" in ./bashrc, it still gave same error processed:1 unchanged: 1.

Gaurav37, 2021-12-23:
I follow next step but before that I " sudo apt-get remove librealsense2-dkms"  I get error: 

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.3.18
completely from the DKMS tree.
------------------------------
Done.
**Reloading original modules into kernel...modprobe: ERROR: could not insert 'uvcvideo': Operation not permitted**
Current status:
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d

Gaurav37, 2021-12-23:
Anyways I try moving ahead with "**sudo apt-get install librealsense2-dkms**" and I get:

Loading the modified modules into kernel...modprobe: **FATAL: Module videodev is in use**.
Current status:
librealsense2-dkms, 1.3.18: added
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d

I move ahead with " **sudo apt-get install librealsense2-utils**" 
I first sudo apt-get remove librealsense2-utils then install.

Then I do remove and install : 
**sudo apt-get install librealsense2-dev
sudo apt-get install librealsense2-dbg**

Gaurav37, 2021-12-23:
modinfo uvcvideo | grep "version:"
version:        1.1.1
srcversion:     CB4E2CA126755A54981CC64

Although earlier times when I installed librealsense, it showed realsense in this output but now it doesn't.

MartyG-RealSense, 2021-12-23:
Hi @Gaurav37  Starting with your first question: you may end up with multiple udev rules on your computer if librealsense was installed from both packages and from source code.  The **/etc** rule is the one that should be removed - as advised in https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547 - as removing the /lib rule can break Debian packages.

closed 2021-12-26 by Gaurav37