realsenseai/librealsense #10098: L515 LiDAR not working

fortuitousanonymous, 2021-12-23:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Hi Intel staff,

After a recent WIN10 re-installation, L515 cannot be correctly loaded (USB2.1, instead of USB3.0 is shown as the connection) and depth data can be acquired, i.e. depth sensor cannot be turned on the Viewer (v.2.50.0). However, the RGB cam & IMU are functioning.

Could you please help me with this issue?

Many thanks,
Dahe

MartyG-RealSense, 2021-12-23:
Hi @fortuitousanonymous   Could you first unplug the micro-sized end of the USB cable from the base of the camera, reverse the connector's orientation and plug it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end). 
 Does doing so make any difference?

fortuitousanonymous, 2021-12-23:
Thanks Marty for the prompt reply.

I restored the depth camera partially. I updated the firmware using Viewer
v2.40. However, it still shows that it is connecting via USB2.1. I can
only run the depth camera at the lowest resolution.

Any advice?

Many thanks,
Dahe

MartyG-RealSense, 2021-12-23:
The symptoms that you describe can occur when the USB connector in the base of the camera is inserted in a certain direction.  Can you confirm please whether flipping the insertion direction of the micro-sized end of the connector in the camera base resolves the problem, please?

fortuitousanonymous, 2021-12-23:
I flipped the insertion direction and the other direction results in no
detection of L515 by the Viewer at all. I repeated this by three times.

fortuitousanonymous, 2021-12-23:
Furthermore, on my laptop, I used L515 3 days ago at full resolution and
nothing changed between back then and now.

I have used L515 since May 2021 and this is the first time I encountered
this issue.

MartyG-RealSense, 2021-12-23:
Thanks very much for testing the connector.

Next, please open the Windows Device Manager and go to the **Cameras** category.  There should be two drivers listed, RGB and Depth, when the L515 is plugged in.  

Right-click on each of them and select **Uninstall device** from the menu until both drivers are removed.  Then unplug the camera from the USB port, wait a couple of seconds and re-insert it.  The drivers should automatically re-install and be listed again in the Device Manager.

Does this resolve the problem?

fortuitousanonymous, 2021-12-23:
Thanks again Marty.

I did that and also flipped the connector. The problem still persists -
USB2.1 and flip does not work.

Cheers,
Dahe

MartyG-RealSense, 2021-12-23:
Thanks for your continued patience.  Next, could you try installing the SDK on Windows again using the **Intel.RealSense.SDK-WIN10** automated installer file in the 'Assets' list of the SDK **Releases** page?  As installing with this method completely overwrites the previous SDK installation made using this method, it can correct problems that might have ocurred during the previous SDK installation.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

fortuitousanonymous, 2021-12-23:
Still the same problem, unfortunately.

Could it be the problem w/ my WIN10 OS as it has recently been upgraded?

Strange enough that 3 days ago, it worked ... Same setup and conditions
that I am aware of ... Could it be a WIN update killed it?

Cheers,
Dahe

MartyG-RealSense, 2021-12-23:
Windows updates can sometimes cause conflicts with RealSense cameras that worked fine before the update, yes.

fortuitousanonymous, 2021-12-23:
Any suggestion that I could revert back? Many thanks

MartyG-RealSense, 2021-12-23:
If your PC has got enabled the saving of recovery points (which are created when updates are performed) then you can roll Windows back to a time before the update.

https://www.windowscentral.com/how-use-system-restore-windows-10

fortuitousanonymous, 2021-12-23:
Thanks Marty. Problem solved, I re-installed WIN10. Interestingly, this
time one side of the USB is 2.1 and another is 3.1.

Many thanks,
Dahe

MartyG-RealSense, 2021-12-23:
That's excellent news, @fortuitousanonymous 

Don't worry about the 2.1 / 3.1 status when flipping the camera end of the USB connector.  That is a known hardware trait of the L515 model.

closed 2021-12-23 by fortuitousanonymous

zhanglepy, 2023-04-25:
How to perform hand eye calibration on the realsense L515 camera, with the eyes in hand？