realsenseai/librealsense #10099: No such file when I catkin_make --pkg librealsense following ROS wiki instructions.

Gaurav37, 2021-12-23:
I am following instructions on [http://wiki.ros.org/realsense_camera/Tutorials/Building_librealsense_from_Sources](url)  but when I catkin_make, it doesn't recognise any librealsense package although its successfully downloaded from github and moved to stabel branch by: git checkout -b stable-branch stable as written on ROS instructions link page.

MartyG-RealSense, 2021-12-23:
Hi @Gaurav37  The Instructions that you linked to for building the ROS wrapper from source code are for the old **realsense_camera** wrapper for the original generation of RealSense cameras from 2014.  It does not have support for modern RealSense camera models such as the 400 Series, L515 and T265.  

The current wrapper for modern RealSense cameras is **realsense2_camera** and instructions for building it from source are at the link below.  The realsense2_camera wrapper does not support old RealSense camera models such as F200 and R200.

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

Which RealSense camera model are you using, please?

MartyG-RealSense, 2021-12-30:
Hi @Gaurav37  Do you require further assistance with this case, please?  Thanks!

Gaurav37, 2021-12-31:
Thanks @MartyG-RealSense I was able to install with commands:
```
sudo apt-get install ros-melodic-realsense2-camera
sudo apt-get install ros-melodic-realsense2-description
```

closed 2021-12-31 by Gaurav37

MartyG-RealSense, 2021-12-31:
That's great to hear, @Gaurav37 - thanks very much for the update and for sharing your solution  :)