realsenseai/librealsense #10100: Volume measurement is not accurate.

HyeYumKim, 2021-12-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Platform                                 | PC |
| Language                            |  python | 

Hello
As I said before, I'm conducting an experiment to measure the volume of an object.

When the code is executed, it works well up to the calibration of the chessboard, but when measuring the volume of an object, the camera keeps contouring similar to the size of the chessboard.

I want to know why this problem occurs.

[image]
[image]
[image]
[image]

I want to know why this problem occurs.

MartyG-RealSense, 2021-12-23:
This is a follow-up of case https://github.com/IntelRealSense/librealsense/issues/10054

Hi @HyeYumKim  It looks as though the program is measuring the chessboard instead of the box that is placed on the chessboard.  

In the original program, the process that it goes through is:

1.  Calibrate the camera to the chessboard.
2.  Prompt the user to place a box onto the chessboard area once calibration is completed.
3.  Take measurements of the box once it is placed in the chessboard area.

If the program has been adapted so that it does not prompt for the box to be placed on the board before taking the measurements of the box then I wonder if the program immediately assumes that the chessboard is the box when the program is run and measures it before you have the opportunity to place a box into the chessboard area.

HyeYumKim, 2021-12-23:
The contents I captured were experimented by correcting the camera according to the chessboard board to see if the volume measurement was good at a distance of 1.5m.

After that, I tried to save the correction value and run it again without a chessboard. However, you can see that they continue to hold the chessboard even though it was well calibrated before running it without a chessboard.

I succeeded in loading what I saved the correction value at a closer distance. So I thought I would succeed even at 1.5m distance.

There is no capture, but if I save and load the calibration value at a distance of 1.5m, assume that the chessboard is a box immediately when the program runs. The chessboard is turned upside down, that is, the chessboard is measured on white paper, and the chessboard is held on the floor without the chessboard.

MartyG-RealSense, 2021-12-24:
The box_dimensioner_multicam program tends to have a 'sweet spot' ideal distance from the chessboard, and positioning the camera closer to or further from that ideal position usually results in a poor / wrong outcome.  A case at the link below is a good example, where the program would put a bounding box around the chessboard instead of the object (like with your program) or the bounding box would be diagonal instead of fitting closely around the box edges.

https://support.intelrealsense.com/hc/en-us/community/posts/360044001733-How-to-find-the-actual-size-of-the-object-dimension-from-the-RGB-image-acquired-using-Intel-Realsense-

[image]

[image]

HyeYumKim, 2021-12-26:
Then, should I find the best location?
Do I have to try everything until I find the best location?

MartyG-RealSense, 2021-12-27:
1 meter from the chessboard (not 0.5 and not 1.5) is likely to be the best distance from the board for the box_dimensioner_multicam application.

If you test with the original unmodified version of the program at 1 meter and it gives good results and then you test at the same camera position with your modified program and get wrong results then this would confirm whether or not the problem is in your modified code.

HyeYumKim, 2021-12-28:
Yes, I'll try it.
Thank you.

MartyG-RealSense, 2022-01-17:
Case closed due to no further comments received.

closed 2022-01-17 by MartyG-RealSense

raivil, 2022-03-09:
> 1 meter from the chessboard (not 0.5 and not 1.5) is likely to be the best distance from the board for the box_dimensioner_multicam application.
>
> If you test with the original unmodified version of [...]

Hey @MartyG-RealSense how can I increase the distance from camera to chessboard and still keep it accurate? 
I'm trying to scan bigger objects and I'd like to have cameras at 2 meters high and apart from each other. (2x2m setup)

MartyG-RealSense, 2022-03-09:
@raivil  You can edit the box_dimensioner_multicam script to change the chessboard values if you create a larger custom chessboard.  This particular program tends to be sensitive to distance though, with a 'sweet spot' position for the camera that causes problems on the image such as mis-aligned or diagonal bounding boxes if the camera is too close or too far from the chessboard.

If you were able to adapt your project to only require one camera then it is possible to edit the program so that the chessboard is not required, as described at https://github.com/IntelRealSense/librealsense/issues/9951

raivil, 2022-03-09:
@MartyG-RealSense 
Using a single camera is indeed an option and would simplify the whose setup especially if I can get rid of the chessboard and the kabsch calculations.
However, I haven't been able to adapt the demo to work with a single camera. 

Do you have any recommendations where/how to start to modify it?

MartyG-RealSense, 2022-03-09:
The RealSense user at https://github.com/IntelRealSense/librealsense/issues/9951 apparently made some progress at https://github.com/IntelRealSense/librealsense/issues/10054 regarding implementing a boardless version of the application, so it may be worth leaving a comment for them on that discussion to ask how they achieved it.