realsenseai/librealsense #10102: Realsense viewer memory leak issue with L515

jtkhair, 2021-12-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Win 11  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.50. 0.3785                         | 
| Language                            |                            | 
| Segment			|  VR                          | 

### Issue Description
Windows 11 experience memory leaks when running the intel realsense viewer with L515

I've tested on 2 windows 11 machines and 1 windows 10 machine. I received the same memory leaks problem on both windows 11 machine.  

What actually has happen;
1. I connect L515 to windows 11 machine running the realsense viewer
2. I turn on the L500 depth sensor
3. The viewer is able to display the streaming data but only for less than 1 minutes before experiencing frame drops
4. I check the system manager and notice that the memory usage has gone to 97% due to high usage by the realsense viewer

Please advice if there is a workaround or solution to this

MartyG-RealSense, 2021-12-23:
Hi @jtkhair  I will pass your report on to Intel.  We do not currently have advice for Windows 11 users of the RealSense SDK due to the newness of the OS at the time of writing this, though we have received a couple of previous reports of different issues with Windows 11 in https://github.com/IntelRealSense/librealsense/issues/9981 and https://github.com/IntelRealSense/librealsense/issues/10029

jtkhair, 2021-12-23:
Looking forward to the solution

MartyG-RealSense, 2021-12-23:
In the meantime you could try downloading and installing the **Intel.RealSense.Viewer.exe** standalone Viewer version in the 'Assets' file list on the SDK **Releases** page.  This .exe file should be launched with a double-leftclick from the folder location that it was downloaded to and not with the launch shortcut icon on the Windows desktop.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

jtkhair, 2021-12-24:
I've tried with the standalone viewer and instruction on a windows 11 machine. However, the issue is still persist in my case.

MartyG-RealSense, 2021-12-24:
Hi @jtkhair  Intel performed Windows 11 testing after reading your report.  They tested with four D435 with depth+RGB at 848x480 and 30 FPS but were unable to replicate the memory leak when monitoring the Windows **Task Manager** interface, though did observe some frame drops.

jtkhair, 2021-12-24:
Hi.. That is good to know. Hopefully intel would performed the test on L515 as well.

MartyG-RealSense, 2021-12-30:
Hi @jtkhair  Do you require further assistance with this case, please?  Thanks!

jtkhair, 2021-12-30:
Hi, yes. Please help to verify problem with L515

MartyG-RealSense, 2021-12-30:
Please run the **rs-capture** program on Windows using the instructions at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656 and monitor the memory.  This will help to identify whether your leak is occurring with the RealSense Viewer specifically or with RealSense programs in general.

jtkhair, 2021-12-31:
Hi @MartyG-RealSense . I confirm that running the rs-capture.exe does not cause the memory leak.

MartyG-RealSense, 2021-12-31:
Thanks very much for testing rs-capture.  Next, please try disabling the two **GLSL** options in the settings of the RealSense Viewer to see whether it stops the memory leak.  Instructions for doing this can be found at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

jtkhair, 2022-01-01:
Hi @MartyG-RealSense . I confirm that disabling GLSL options works as workaround for the current memory leak issue.

MartyG-RealSense, 2022-01-01:
That's great to hear, @jtkhair - thanks very much for the update!

jtkhair, 2022-01-08:
Hi @MartyG-RealSense, you may close this issue. Thanks for the support

MartyG-RealSense, 2022-01-08:
Thanks very much @jtkhair for the update!  As you suggested, I will close the case.  Thanks again  :)

closed 2022-01-08 by MartyG-RealSense