realsenseai/librealsense #10104: "Write Table" button is disabled | Can't write calibration table

ravijo, 2021-12-24:
Dear RealSense Team,

I am trying to write calibration data. This file was actually created from the RealSense Viewer application, via Calibration Data -> Save As). 

Please see the following screen recording which shows that the "Write Table" button is disabled.

https://user-images.githubusercontent.com/1439011/147343235-753e31c3-98d2-4d58-9256-dcb3db397c2b.mp4

-----------------------------------------------------------
| Required Info               |                           |
|-----------------------------|---------------------------|
| Camera Model                | D415                      | 
| Firmware Version            | 05.13.00.50              | 
| Operating System & Version  | Ubuntu 18.04.6 LTS        | 
| Kernel Version (Linux Only) | 5.4.0-86-generic          | 
| Platform                    | PC                        |
| SDK Version                 | 2.50.0  | 
| Language                    | None                      | 
| Segment			          | Robot                     | 

The `CustomRW.exe` tool shows a different error at the same time:

```
C:\Program Files\Intel\CalibrationToolAPI\2.11.1.0\bin>Intel.Realsense.CustomRW.exe -w -f  after.json
CustomRW for Intel RealSense D400, Version: 2.11.1.0

  Device PID: 0AD3
  Device name: Intel RealSense D415
  Serial number: 120222062385
  Firmware version: 05.13.00.50

Error (2001): left and right camera resolution is not supported.
```

MartyG-RealSense, 2021-12-24:
Hi @ravijo  To unlock the **Write Table** button, you have to enable the depth stream first before selecting the **Calibration Data** menu option.

[image]

MartyG-RealSense, 2021-12-30:
Hi @ravijo  Do you require further assistance with this case, please?  Thanks!

ravijo, 2021-12-30:
Nope. Thank you very much. Closing this issue!

closed 2021-12-30 by ravijo

MartyG-RealSense, 2021-12-30:
Thanks again @ravijo  :)