realsenseai/librealsense #10105: Can you please provide CustomCalibration project for Ubuntu?

ravijo, 2021-12-24:
Dear RealSense Team,

-----------------------------------------------------------
| Required Info               |                           |
|-----------------------------|---------------------------|
| Camera Model                | D415                      | 
| Firmware Version            | 05.13.00.50              | 
| Operating System & Version  | Ubuntu 20.04.3 LTS        | 
| Kernel Version (Linux Only) | 5.4.0-91-generic          | 
| Platform                    | PC                        |
| SDK Version                 | 2.50.0  | 
| Language                    | None                      | 
| Segment			          | Robot                     | 

### Issue Description

The current CustomCalibration project is given for WIndows OS as it contains the following files:

```
$ ll
total 116
drwxr-xr-x  2 ravi ravi  4096 Mar 23  2021 ./
drwxrwxr-x 11 ravi ravi  4096 Mar 23  2021 ../
-rw-r--r--  1 ravi ravi  2549 Mar 23  2021 540x480_60mm.pdf
-rw-r--r--  1 ravi ravi 18193 Mar 23  2021 CalibrationManager.cpp
-rw-r--r--  1 ravi ravi  3976 Mar 23  2021 CalibrationManager.h
-rw-r--r--  1 ravi ravi  4749 Mar 23  2021 CustomCalibration.cpp
-rw-r--r--  1 ravi ravi  2899 Mar 23  2021 CustomCalibration.h
-rw-r--r--  1 ravi ravi   790 Mar 23  2021 CustomCalibrationTest.sln
-rw-r--r--  1 ravi ravi  6511 Mar 23  2021 CustomCalibrationTest_static.vcxproj
-rw-r--r--  1 ravi ravi 11361 Mar 23  2021 CustomCalibrationTest.vcxproj
-rw-r--r--  1 ravi ravi  5718 Mar 23  2021 CustomCalibrationWrapper.cpp
-rw-r--r--  1 ravi ravi  1888 Mar 23  2021 CustomCalibrationWrapper.h
-rw-r--r--  1 ravi ravi  3956 Mar 23  2021 FileOp.h
-rw-r--r--  1 ravi ravi  1729 Mar 23  2021 Main.cpp
-rw-r--r--  1 ravi ravi  1553 Mar 23  2021 makefile
-rw-r--r--  1 ravi ravi   323 Mar 23  2021 readme.txt
-rw-r--r--  1 ravi ravi 10253 Mar 23  2021 Rs400Dev.cpp
-rw-r--r--  1 ravi ravi  4083 Mar 23  2021 Rs400Dev.h

```
**Can you please provide CustomCalibration project for Ubuntu?** so that we can have CMake compatible CMakeLists.txt file.

MartyG-RealSense, 2021-12-24:
Hi @ravijo  You should be able to obtain the CustomCalibration files on Linux by installing the **Calibration API** for Linux.  Instructions for Linux installation are on page 14 onwards of the tool's user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

Page 63 onward of this guide (Section **7.  Developing Custom Calibration Apps with Calibration API**) describes how to use a compressed package called **Dynamic CalibrationAPI-Linux-2.11.0.0.tar.gz** that is installed by the calibration tool to compile the calibration examples with CMake (on page 64).

[image]

ravijo, 2021-12-24:
@MartyG-RealSense 

Thank you very much. Please see attached screenshot below:

[image]

It shows contents of the Dynamic CalibrationAPI-Linux-2.11.0.0.tar.gz file.

I plan to do Section 4.3 in [Intel RealSense D400 Series Custom Calibration Whitepaper](https://dev.intelrealsense.com/docs/d400-series-custom-calibration-white-paper), so I need CustomCalibrationTest. 

The above screenshot shows that CustomCalibrationTest project is designed for WIndows.

MartyG-RealSense, 2021-12-24:
It would seem unusual to me to put files in a Linux tar.gz file that deliberately has 'linux' in its filename if the contents could not be built on Linux.

You could perhaps use **Visual Studio Code** on Linux and build the project using the **makefile** included in the folder (a process described in the link below).

https://earthly.dev/blog/vscode-make/

ravijo, 2021-12-29:
@MartyG-RealSense 

I created [CMakeLists.txt](https://github.com/ravijo/realsense-calibration/blob/be7070b07c1e07f2758880f674330b39374baf70/CMakeLists.txt) and uploaded the complete project to GitHub under [Intel License](https://github.com/ravijo/realsense-calibration/blob/be7070b07c1e07f2758880f674330b39374baf70/license.txt).

https://github.com/ravijo/realsense-calibration

It works well!

MartyG-RealSense, 2021-12-29:
Great to hear that you achieved a solution, @ravijo   :)  Thanks so much for sharing it with the RealSense community!

closed 2022-01-04 by MartyG-RealSense