realsenseai/librealsense #10107: 如果您看到，能麻烦您尽快回复我吗？非常感谢！如何将.bag文件转化为带有深度信息文件的RGB图片？

woshicaobin, 2021-12-27:
我想获得拍摄物体的深度信息以及图片，我是通过拍摄视频的方式拍摄的，并保存为.bag格式文件，我需要将.bag的视频文件转化为具有深度信息的文件以及与之对应的图片，请问我可以怎么做呢？

MartyG-RealSense, 2021-12-27:
Translation: If you see, can you please reply to me as soon as possible?  Thank you so much!  How to convert .bag files into RGB images with depth information files?  I want to get the depth information and pictures of the shooting object.  I took the video by shooting and saved it as a .bag format file.  I need to convert the .bag video file into a file with depth information and the corresponding picture.  What can I do?

*********************

Hello @woshicaobin  If you want to extract RGB and depth frames from the bag as files named by their frame timestamp in order to help to match them together then you can do this with a RealSense SDK tool called 'rs-convert' that can be found at the link below.  

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

rs-convert enables depth to be extracted to file formats such as .raw that preserve the depth detail that would be lost by saving depth as a png image file.

The rs-convert program in the above link is in C++ language, though there is an already built executable version of the tool included in the RealSense SDK.  

On Windows, if the full Windows version of the RealSense SDK has been installed using the automated installer file then the executable .exe version of the rs-convert tool can be found at this folder location on the computer:

C:\Program Files (x86)\Intel RealSense SDK 2.0\tools

If librealsense was built from source code with CMake then the build term below should be included in the CMake build instruction in order to build the SDK with tools and example programs included:

-DBUILD_EXAMPLES=true

********************

你好@woshicaobin 如果您想從包中提取 RGB 和深度幀作為以幀時間戳命名的文件以幫助將它們匹配在一起，那麼您可以使用名為“rs-convert”的實感 SDK 工具來完成此操作，該工具可以找到在下面的鏈接。

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

rs-convert 可以將深度提取為 .raw 等文件格式，保留將深度保存為 png 圖像文件會丟失的深度細節。

上面鏈接中的 rs-convert 程序是用 C++ 語言編寫的，儘管實感 SDK 中已經包含了該工具的可執行版本。

在 Windows 上，如果已使用自動安裝程序文件安裝了完整的 Windows 版本的實感 SDK，則可以在計算機上的以下文件夾位置找到 rs-convert 工具的可執行 .exe 版本：

C:\Program Files (x86)\英特爾實感 SDK 2.0\工具

如果 librealsense 是使用 CMake 從源代碼構建的，那麼下面的構建術語應包含在 CMake 構建指令中，以便使用包含的工具和示例程序構建 SDK：

-DBUILD_EXAMPLES=真

woshicaobin, 2021-12-28:
[image]

感谢您的答疑，但是我在使用的时候，出现这个错误，一直无法解决，无法转换，请问您这是什么愿意呢？
关于您上面的解答，我使用SDK成功的导出了RAW格式和PNG图像，但是与PNG图片一起导出的CSV文件好像有很多缺失，只有几个信息值，而我需要的正是PNG图片和对应的无缺失CSV世界坐标文件，请问我可以怎么做呢？

MartyG-RealSense, 2021-12-28:
Translation: Thank you for your Q&A, but this error occurred when I was using it, and it has been unable to be resolved and cannot be converted.  What do you wish to do?

Regarding your answer above, I used the SDK to successfully export the RAW format and the PNG image, but the CSV file exported with the PNG image seems to have a lot of missing, only a few information values, and what I need is the PNG image and the corresponding.  There is no missing CSV world coordinate file.  What can I do?

**************

Regarding the **Error opening file** message, the rs-convert tool may not be able to find the bag file because it does not have a folder path to the bag location defined, only the filename of the bag.  So the rs-convert launch instruction would be expecting to find the bag file located in the same folder as the tool (the **tools** SDK folder).  Could you try copying your bag files into the tools folder and then repeating the conversion commands, please?

 Regarding the export of the CSV file, are you doing this by using the Viewer's **Snapshot** option on the depth stream to export a PNG, RAW and CSV?

[image]

The CSV exported from the Viewer is providing camera metadata information about the saved image instead of a text representation of the depth image itself.

************

關於**Error opening file** 消息，rs-convert 工具可能無法找到包文件，因為它沒有定義包位置的文件夾路徑，只有包的文件名。 因此 rs-convert 啟動指令會期望找到與工具位於同一文件夾中的包文件（**tools** SDK 文件夾）。 您能否嘗試將您的包文件複製到工具文件夾中，然後重複轉換命令？

  關於 CSV 文件的導出，您是否通過在深度流上使用查看器的 **Snapshot** 選項導出 PNG、RAW 和 CSV 來執行此操作？

[image]

從查看器導出的 CSV 提供有關已保存圖像的相機元數據信息，而不是深度圖像本身的文本表示。

[image]

woshicaobin, 2021-12-28:
非常感谢您这么详细的回答！这对我这个新手来说超级有帮助。但是我还是想请问您几个问题：
1 我通过rs-convert转化为了CSV和PNG两种，我想请问CSV和PNG的命名都为同一个时间戳，是代表他们都是同一帧的数据吗？即他们是相匹配的。
2 请问CSV文件代表的是这一时间戳帧率的整张图像每一个像素点的深度信息吗？
3 请问如果我想对很多bag文件批处理的话，我可以怎么做呢？
非常感谢您的回答！

MartyG-RealSense, 2021-12-28:
Translation: Thank you very much for your detailed answer!  This is super helpful for me as a novice.  But I still want to ask you a few questions: 

1.  I used rs-convert to convert to CSV and PNG. I would like to ask that both CSV and PNG are named with the same timestamp.  Does it mean that they are the data of the same frame?  That is, they are matched.
2.  Does the CSV file represent the depth information of each pixel of the entire image at this time stamped frame rate?
3.  If I want to batch process many bag files, what can I do?

Thank you very much for your answer!

******************

1.  rs-convert originally named the files with the frame number but this was later changed to the frame timestamp.  My understanding is that naming by timestamp makes it easier to match the frames of different stream types to the time that they were generated.  So RGB and depth frames produced at the same time may not necessarly have the same frame number but can be matched accurately by their timestamp.

2.  Yes, that is my understanding that the content of a CSV represents the depth values of that frame shown as numeric values that resemble the depth image of that particular frame.

[image]

3.  It is possible to play back a set of multiple bag files automatically in order of their filename.  https://github.com/IntelRealSense/librealsense/issues/6887 is a good reference for doing this.

***********************

1. rs-convert 最初使用幀號命名文件，但後來更改為幀時間戳。 我的理解是，按時間戳命名可以更容易地將不同流類型的幀與其生成時間進行匹配。 因此，同時生成的 RGB 和深度幀不一定具有相同的幀號，但可以通過它們的時間戳準確匹配。

2. 是的，這是我的理解，CSV 的內容表示該幀的深度值，顯示為類似於該特定幀的深度圖像的數值。

[image]

3. 可以按文件名的順序自動播放一組多個包文件。 https://github.com/IntelRealSense/librealsense/issues/6887 是一個很好的參考。

MartyG-RealSense, 2022-01-03:
Hello @woshicaobin  Do you require further assistance with this case, please?  Thanks!

你好@woshicaobin 請問這個案例你需要進一步的幫助嗎？ 謝謝！

MartyG-RealSense, 2022-01-08:
Case closed due to no further comments received.

由於未收到進一步評論，案件已結案。

closed 2022-01-08 by MartyG-RealSense