realsenseai/librealsense #10108: [C#] Sensor Options are not updated after JSON load

jorpelae, 2021-12-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 / D435 | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.45.0.3212 | 
| Language                            |  C#/unity  | 
| Segment			| others | 

### Issue Description

I'm trying to modify some parameters from the color sensor on runtime but I'm not getting the expected behaviour.

My original idea is:
1. Start application
2. Load a JSON preset using AdvancedDevice.JsonConfiguration
3. Enable streams using Config.EnableDevice, Config.EnableStream and Pipeline.Start
4. Modify color parameters (exposure, etc) using FrameSet.ColorFrame.Sensor.Options[Option].Value
5. End application

Modifying a value on the JSON file modifies the color texture as expected.
Providing a value to Options[Option] modifies the color texture as expected. 
But reading the value after loading the JSON and before modifying with Options[Option] does not return the correct value. Instead, it returns the value provided on the previous app execution. It does not reflect the value being currently applied on the texture.

Maybe I'm doing something wrong?

jorpelae, 2021-12-27:
Just for the record. Disabling auto-exposure or auto-white-balance solves the problem in my use case.

closed 2021-12-27 by jorpelae

MartyG-RealSense, 2021-12-27:
I was just looking into your case, @jorpelae - thanks very much @jorpelae for the update that you achieved a solution!