realsenseai/librealsense #10109: Hi

gwspotex, 2021-12-27:
I have project work with object-detection(done) 

as you can see the yt https://youtu.be/KyT0CxR-Jws?t=210

i just want to get the some object litte offset to  fit the new  robot work space 

Is any program or project do the same thing?

[image]

MartyG-RealSense, 2021-12-27:
Hi @gwspotex  Can you confirm please whether you are trying to have the application be able to work out the positions of objects placed at random coordinates, pick up one of the objects and carry it to another location please (like with the robot arm in the YouTube video)

gwspotex, 2021-12-28:
in additonal  there some video https://www.youtube.com/watch?v=YIaQC1iHtMA

MartyG-RealSense, 2021-12-28:
It looks as though you are trying to track both the position and angle of the object (its 'pose') and need to be able to obtain the position and angle of the object even if it is rotated.

closed 2022-01-09 by MartyG-RealSense