realsenseai/librealsense #10110: std::runtime_error, Error occured during execution of the processing block! after running pointcloud.calculate()

UmutUz, 2021-12-27:
Using Windows 11
<html>
<body>

Required Info |  
-- | --
Camera Model | D415
Firmware Version | 
Operating System & Version | Windows 11
Platform | PC
SDK Version | 2.49.0.3474 (took that from the realsense viewer not sure if right)
Language | C++

</body>
</html>

When executing the code below I get the error mentioned in the title. I saw another issue here which is practically the same but there is no fix for that and it seems pretty old. When I comment the depth_to_disparity filter, this error disappears so I would think its associated with that but I have no clue how to fix it.

```
        rs2::frame depth_filtered = frames1.get_depth_frame();

	depth_filtered = dec_filter.process(depth_filtered);
	depth_filtered = spat_filter.process(depth_filtered);
	depth_filtered = temporal_filter.process(depth_filtered);
	depth_filtered = depth_to_disparity.process(depth_filtered);
 
	rs2::frame depth_filtered2 = frames2.get_depth_frame();

	depth_filtered2 = dec_filter.process(depth_filtered2);
	depth_filtered2 = spat_filter.process(depth_filtered2);
	depth_filtered2 = temporal_filter.process(depth_filtered2);
	depth_filtered2 = depth_to_disparity.process(depth_filtered2);

	rs2::frame color1 = frames1.get_color_frame();
	rs2::frame color2 = frames2.get_color_frame();

	// Map RGB to PCL
	pcl_camera1.map_to(color1);
	pcl_camera2.map_to(color2);

	// Generate the pointcloud and texture mappings
	points_camera1 = pcl_camera1.calculate(depth_filtered);
	points_camera2 = pcl_camera2.calculate(depth_filtered2);
```

MartyG-RealSense, 2021-12-27:
Hi @UmutUz  As a starting point in investigating your case, I would recommend checking whether the depth_to_disparity filter has been declared as True and whether the opposite **disparity_to_depth** filter has been declared as False.  

With depth_to-disparity and disparity_to_depth, one of them must be true and the other false.  You cannot have them set to both true or both false.  

A C++ script at https://github.com/IntelRealSense/realsense-ros/issues/1333 provides an example of these declarations.

[image]

[image]

UmutUz, 2021-12-27:
I sadly don't have the realsenses at hand right now and would try this tomorrow, didn't know I had to declare both of them! Thanks for the tip I will update this comment if I get a result

MartyG-RealSense, 2021-12-28:
Thanks very much, @UmutUz - please do let me know the outcome of your test.  Good luck!

UmutUz, 2021-12-28:
It worked! Thanks for the help.

closed 2021-12-28 by UmutUz

MartyG-RealSense, 2021-12-28:
That's great news, @UmutUz - thanks very much for the update!