realsenseai/librealsense #10111: [D435i]How to extract point cloud from .bag files.

woshixzy, 2021-12-27:
I use the command of ROS**(>rosrun pcl_ros bag_to_pcd test.bag <tpic> ./test.pcd)** to extract PCD point cloud data.
so I tried to use the bag_to_pcd  command from ROS to convert .bag file into a pcd file. 
But this method cannot create  PCD file.I can't find the extractable point cloud topic in the .bag file.
How to extract point cloud data thank you~

MartyG-RealSense, 2021-12-27:
Hi @woshixzy  My research of your question did not find many useful references, though hopefully the best one that I did find in the link below will be helpful.

**Chinese-language original article**
https://blog.csdn.net/jean7155/article/details/118999912

**English translation**
An English translation of the Chinese-language article linked to above is shown in images below.

[image]

[image]

MartyG-RealSense, 2022-01-08:
Case closed due to no further comments received.

closed 2022-01-08 by MartyG-RealSense

Martinyllim, 2026-09-07:
The bag_to_pcd command needs the exact topic name for the point cloud, and depth cameras like the D435i sometimes publish it somewhere less obvious than you'd expect. Try rosbag info yourfile.bag first, that'll list every topic actually in the bag along with its message type, look specifically for a sensor_msgs/PointCloud2 type topic (commonly something like /camera/depth/color/points on RealSense, but it varies by launch config). If nothing shows up as PointCloud2, the bag may only have raw depth and RGB images recorded rather than a computed point cloud topic, in which case you'd need to run the RealSense driver's point cloud node against the recorded depth/RGB topics first to generate one.

Once you've got that sorted and can actually see the topic, worth keeping in mind that bag_to_pcd just dumps one PCD per frame rather than merging anything, so if what you actually want is one combined map rather than a folder of individual frame captures, you'll need something doing registration between frames on top of that. Full disclosure, I built a tool called LidarFlow that runs GLIM SLAM on a bag or MCAP and hands back one merged PCD/PLY, might be worth a look once you've got the topic issue fixed and know what you're actually working with.

MartyG-RealSense, 2026-09-07:
Thanks so much @Martinyllim for sharing your experience on this topic for the benefit of future readers!