realsenseai/librealsense #10113: How to do a snapshoot while displaying video stream?

ZhongQingliang, 2021-12-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Linux (Ubuntu 20LTS)   | 
| Kernel Version (Linux Only)    | 5.11                                        | 
| Platform                                 | PC |
| SDK Version                            |      pyrealsense 2.50               | 
| Language                            |  python                       | 
| Segment			|  Others                     | 

### Issue Description
I'm using the pyrealsense for convenience to develop an application. How can I do an RGB snapshoot by a hot key meanwhile displaying video stream?

MartyG-RealSense, 2021-12-28:
Hi @ZhongQingliang  If you wanted to be able to save an RGB image snapshot whilst video streaming continues seamlessly then that might require the use of **multithreading** like the Python case linked to below, where a RealSense user had one thread for the live-captured images and another thread to write the captured images to video.

https://support.intelrealsense.com/hc/en-us/community/posts/360033158094-Recording-Video-in-60fps-with-threading-and-queue-Python-

Another Python discussion about multithreading is at https://github.com/IntelRealSense/librealsense/issues/6039

If having continuous RGB streaming wasn't a requirement then a simpler Python example at https://github.com/IntelRealSense/librealsense/issues/9448#issuecomment-883100431 demonstrates how to save an image using **imwrite**

ZhongQingliang, 2021-12-29:
@MartyG-RealSense It seems the _pipeline.wait_for_frame()_ method can be called by different threads at the same time, isn't it?

MartyG-RealSense, 2021-12-29:
You can move frames from one thread to another when using wait_for_frames(), as described in the section of Intel's **Frame Management** documentation linked to below.

https://dev.intelrealsense.com/docs/frame-management#section-frames-and-threads

closed 2022-01-11 by MartyG-RealSense