realsenseai/librealsense #10114: Where i can find 2016 R2 SDK and its doc download now？

SnowWindSaveYou, 2021-12-29:
hi, i'm looking for a hand joints recognition solution with depth information.
Since sdk 2.0 doesn't contains it, i just returned my d455 and bought a sr300 for its 2016 R2 SDK support, 
but then i fount that i can't find the download page for 2016 R2, 
all of urls i can found on internet are redirect to the sdk2.0 page

https://software.intel.com/en-us/articles/realsense-sdk-windows-eol

MartyG-RealSense, 2021-12-29:
Hi @SnowWindSaveYou  If you left-click on the link below then it will launch a 1.8 gb download in your browser of an installer file for 2016 R2.  It is a slow download, so it is best to start it and then do something else until it is finished.

http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe 

The R2 and R3 versions are very similar, with the main difference being that R2 installs the entire 2016 SDK and R3 installed a 'core' module and then required the separate installation of additional modules for particular features such as hand tracking.  So the R2 'all in one' installation is the most convenient.

I have attached a compressed zip file containing the official PDF manuals for this old SDK.

[Realsense-2014-Docs.zip](https://github.com/IntelRealSense/librealsense/files/7787920/Realsense-2014-Docs.zip)

SnowWindSaveYou, 2022-01-04:
> 

Hi, i tried this sdk last week, but it can't find my camera. _(when i open the camera explorer it shows i didn't connect any camera, and every samples in sdk are failed to run.)_
After i search on internet, i fount that maybe because i missed its driver, but i can't find its driver install link either.

On the other hand, i can open the camera by intel realsense Viewer and pyrealsense2, i'm not sure is this really the driver problem...

MartyG-RealSense, 2022-01-04:
Yes, if the SR300 can be detected by the RealSense Viewer then it sounds as though the firmware drivers and the camera drivers (RGB, Depth and Virtual drivers) are okay.

Originally in the 2016 SDK, a driver installation program called the **Depth Camera Manager (DCM) had to be run first and then the SDK installer run secondly.  This was though before Intel introduced in 2017 the ability for the SR309 camera drivers to be auto-installed over the internet simply by inserting the camera into the USB port.

Usually, if the camera could not be detected then the SDK installer would provide a non-detection error message page and exit, so it was likely able to detect the drivers okay if the SDK installation completed successfully.  

The symptoms that you describe make me suspect that you are experiencing a known issue with SR300 on some computers where there is insufficient power provided to the camera by the USB port and so the camera disconnects as soon as s program tries to access it.  This could be indicated by a green light on the front of the camera turning on for a second and then going out again.  In such a situation, using a mains electricity powered USB hub may help to resolve the issue.

SnowWindSaveYou, 2022-01-05:
> Yes, if the SR300 can be detected by the RealSense Viewer then it sounds as though the firmware drivers and the camera drivers (RGB, Depth and Virtual drivers) are okay.
> 
> Originally in the 2016 SDK, a driver installation program called the **Depth Camera Manager (DCM) had to be run first and then the SDK installer run secondly. This was though before Intel introduced in 2017 the ability for the SR309 camera drivers to be auto-installed over the internet simply by inserting the camera into the USB port.
> 
> Usually, if the camera could not be detected then the SDK installer would provide a non-detection error message page and exit, so it was likely able to detect the drivers okay if the SDK installation completed successfully.
> 
> The symptoms that you describe make me suspect that you are experiencing a known issue with SR300 on some computers where there is insufficient power provided to the camera by the USB port and so the camera disconnects as soon as a program tries to access it by enabling a stream. This could be indicated by a green light on the front of the camera turning on for a second and then going out again. In such a situation, using a mains electricity powered USB hub may help to resolve the issue but does not resolve it in every case.
> 
> If a particular computer has such a power problem then it can also manifest in the RealSense Viewer, with the camera being able to be detected and displayed in the options side-panel but disconnecting when the depth stream is enabled.

thank you! it fixed by change usb port _(:з」∠)_

closed 2022-01-05 by SnowWindSaveYou

MartyG-RealSense, 2022-01-05:
You are very welcome.  That's great news that a port change solved your problem.  Thanks very much for the update!