realsenseai/librealsense #10115: Cant write calibration parameters using depth_auto_calibration.py as an example

PierreLouisT, 2021-12-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

PierreLouisT, 2021-12-29:
Hello ! I need a little help from you, I am using the depth_auto_calibration.py from that link from github. It works well and I could write the new calibrations parameters. But I dont know how to have XML document where I can write it using Intel.Realsense.CustomRW -w -f "document calibration parameters.xml"

I used the dynamic calibration tool already and it worked perfectly for me but I also need intrinsics parameters and I use the on-chip calibrator via the script. Indeed, when I am using realsense viewer there are calibration mistakes as undefined value in run_on_chip calibrator hwmon failed. When I use the code provided in github, it works but I need to take the parameters to write them using CustomRW 

Thank you in advance, 

I searched on the differents issues posted and not found yet a solution for my problem

MartyG-RealSense, 2021-12-29:
Hi @PierreLouisT  The On-Chip calibration mode does not have an XML export function like the CustomRW tool does.

Once you have written a calibration with the On-Chip tool, you can access the current calibration and save it to file in **json** format by going to **More > Calibration Data** in the RealSense Viewer and clicking the **Save As...** button.  The saved json can then be opened in a text editor program.

[image]

[image]

[image]

PierreLouisT, 2021-12-29:
Ok, thank you very much Marty, it seems like a good idea, 
Do you know if I can convert json document to an xml and use it with the CustomRW ? 
If it is possible, I will close this issue :)

MartyG-RealSense, 2021-12-29:
You could try simply opening the json in the text editor and saving the contents with the .xml file extension instead of .json.  This should create an xml format file that you can try loading into CustomRW (though I do not know if an xml created in this way can be read by CustomRW).

[image]

PierreLouisT, 2021-12-29:
I will try on my own, thank you very much for your help, it is okay for me :+1: 
I close this issue, thank again

PierreLouisT, 2023-01-20:
Hello ! 

I need assistance about camera calibration with intrinsics parameters (radials distortions, not tangentials parameters). I made a bag file with D455 realsense camera underwater. I got a wrong pointcloud as expected. 
I would like test a calibration on this bag file but I dont know how to do it. I cannot experience underwater again, so I am looking for a  solution to do it. Do you know if it makes sense to use openCV calibration functions ? I don't think so because hardware parameters has been saved during the recording of the bag file. 
Have you experienced this before ? 
Is it possible to modify distortion parameters with a bag file, I just want to verify if my distortions calculs are good to put later again a camera underwater, 
Hope I am clear, 
Thank in advance for your help. 

PS : I tested with the viewer from Intel Realsense but without any success

MartyG-RealSense, 2023-01-20:
Hi @PierreLouisT  At https://github.com/IntelRealSense/realsense-ros/issues/1723 there is an extensively detailed discussion about underwater calibration.  

It is so long that a lot of the comments are hidden under a cut, so I recommend visiting https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-802741234 first and left-clicking on the 'Load more' link to display the entire discussion before starting to read it at the beginning.

In your case however, the depth data has already been recorded to the bag file and cannot be changed.  As you suggest, using OpenCV's camera calibration functions may be an option for you, first undistorting the image and then remapping the distortion.

https://docs.opencv.org/4.x/dc/dbb/tutorial_py_calibration.html

PierreLouisT, 2023-01-20:
Hello Marty ! 
Thank you for your answer, you seemed to understand my main problem, 
Do you think it will be possible with openCV to modify distortion radials coefficients ? Will it be executed directly on the pointcloud ? Will it be possible to visualize the "new" pointcloud ?

MartyG-RealSense, 2023-01-20:
My knowledge of distortion correction in OpenCV is limited, unfortunately. However, if you want to remove radial distortion to correct a 'barrel distortion' (bulging) in the image then googling for the term **opencv "remove radial distortion"** should provide useful research leads.  Include the quotation marks in the search term so that the most relevant search results are provided.

The OpenCV distortion functions could potentially be incorporated into the RealSense SDK's Python / OpenCV pointcloud example program **opencv_pointcloud_viewer.py**, with the script modified to use the bag file as its data source instead of a live camera.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

closed 2023-02-07 by MartyG-RealSense