realsenseai/librealsense #10116: creating a pointcloud and frames from from previously data saved using .get_data()

ilyak93, 2021-12-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D465 | 

| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.0 | 
| Language                            |  {C++ } | 

### Issue Description
Hi, I saw similar questions long ago, but didn't figure out the concrete solution for this:

My current logic of recording and saving frames is by using callbacks, when in each callback I use frame.get_data(), copy it by bytes copy (memcpy) to my storage (mmaped_file), rgb and depth.

Now, if I want to create a pointcloud from the raw data Ive collected, can I recreate frames as rs::frames with the data I got ? 

something like:

    char* rgb_data_saved_by_callback = load("path_to_color_frames_file"); //suppose I have saved 30 raw color and depth frames
    char* depth_data_saved_by_callback = load("path_to_depth_frames_file");
    vector<rs::frame> color_vec;
    vector<rs::depth_frame> depth_vec;
    
    for(int i=0; i < 30; ++i){
        rs::frame color= rs::frame(color_data_saved_by_callback[i*color_frame_size]);
        color_vec.push_back(color);
        rs::frame depth= rs::frame(depth_data_saved_by_callback[i*depth_frame_size]);
        depth_vec.push_back(depth);
    }

and then do as in the pointcloud example something like: 
    
    pc.map_to(color_vec[0]); 
    //maybe I need to pass few color_images of color and depth, I'm not an expert in realsense sdk yet.
    // but the pseudo-code I wrote i c++ style is capable to do so (pass the whole vector instead a sole frame).
    
    // Generate the pointcloud and texture mappings
    points = pc.calculate(depth_vec[0]);
    
    // Upload the color frame to OpenGL
    app_state.tex.upload(color_vec[]0);
    
    // Draw the pointcloud
    draw_pointcloud(app.width(), app.height(), app_state, points); 

So that main issue is how to construct (I didn't saw the appropriate constructor for the relevant classes) from raw data I've got from .get_data() and saved as bytes the usual rs:frame objects to use them as in the examples and e.t.c ? 

Otherwise, as I started the question, maybe my logic should be to create a pointcloud from captured color and depth frames during the callback (which sounds resource expansible and time consuming for a callback), I afraid that that will effect the number of frrames I collect and the speed, which is very important to be as much real time as it can be (now the callbacks effectively are perfect, because for example during 60 seconds approximately I collect 1800 frames approximately, which corresponds to he 30 fps I set and use), but are there any other effective alternatives ?  

Thank you in advance.

ilyak93, 2021-12-30:
Maybe another solution, if there aren't any appropriate constructors from raw data to rs::frame, is to serialize the rs::frame and rs::depth_frame during the callbacks instead of creating a point cloud on the fly and then save it (probably also by serialization, I'm no sure how).

Generally, I want to have an offline point-cloud, one that I can save on my disk, and also load back and view it with rs lib ,is this possible somehow ?

MartyG-RealSense, 2021-12-30:
Hi @ilyak93  Thanks very much for your additional comment about how you want an offline point cloud that you can save to disk and then load back.  Your original suggested approach of using memcpy is outside of my programming knowledge unfortunately, so I do not have advice that I can offer on that particular method.

You could use the **RealSense Viewer** C++ tool to record color and depth camera data to a **bag** format file (which is like a video recording containing camera data from all streams that were enabled at the time of recording), and then load the bag file into the Viewer and play the data back as a point cloud in the Viewer's **3D** mode.  

Point clouds in this mode can be exported to a **.ply** point cloud file that can be loaded into other applications, such as 3D modelling tools like MeshLab and Blender for further editing and optional conversion into solid-mesh 3D models.

https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer

It is straightforward to export a point cloud to a .ply file but typically the ply is not able to be imported back directly into librealsense.  It is advised in https://github.com/IntelRealSense/librealsense/issues/2976 to use a specialized pointcloud library for import.  The librealsense SDK has compatibility wrappers for creating applications that interface with libraries such as PCL and Open3D.

ilyak93, 2021-12-30:
@MartyG-RealSense, so as much as I understand, it is probably true not only for pointcloud, but also for frames, the fact that you can't save them and then load them back without using the .bag format, is it ?

I'll try to clarify the thing for you, because it is actually very simple, and is probably implemented in one way or another in real sense viewer and .bag format. 

So, the rs::frame class has a method .get_data() which you probably familiar with, it's signature is `const void * | get_data () const` which returns a sequence of bytes which is all the frame numbers which represent the rgb channels values.

Which means its size in bytes is the the size of the frame, suppose on my camera the height and width are 720*1280, then the size of the "data" of the frame is 720*1280*channels(which is 3) * sizeof(char*) (because the data can be reinterpreted just as a seqeunce of bytes encoding each pixel rgb values, for example for pixel in location x=0, y=0 there will be 3 encoded values for 3 rgb channels: 253, 125, 197 (values between 0-255).

The thing is I have another two cameras (not real-sense) for which I want to create manually a point-cloud, but before that I do some synchronization between real-sense camera and the others, so that's why the automatic recording into bag file and then exporting it to a point-cloud not the best fit for my use-case, I need to work on a lower level and be able to present the frames from their raw view (like what I asked bytes to rs::frame and then for example I can probably create a .bag and then view it as a ply and e.t.c.).

So, the ROS-bag format, is it the same as this: 
[ROS](http://wiki.ros.org/rosbag/Code%20API#C.2B-.2B-_API)
or is it something related exclusively to realsense API  ?

If it is, I see in this link that I can write and create my own bag files, so an idea could be to first of all save a .bag using the Real Sense Viewer, and then open it and decode in c++ API in the link, to see what structure and information there is, and maybe I would be able to create manually my bag file of the frames I saved as I told above (I wonder what data is should be composed in a .bag file about the rs::frames, so it will be a valid .bag file, so I can recreate it manually, but I'm pretty sure that the data that can be achieved by .get_data() is 100% there, maybe some-other metadata as well).

What do you think about that ? 
I hope my clarification on the data helped to understand what I do, why is this is the level of working, and maybe helped you to help me with ideas :)

ilyak93, 2021-12-30:
Maybe what is I'm actually looking for is a alternative for ROS-bag and realsense point-cloud, but something is able to translate RGBD into a pointcloud (some pointcloud in some software).

MartyG-RealSense, 2021-12-31:
You can obtain the intrinsics for a particular camera model using the librealsense SDK's **rs-enumerate-devices** tool, as described in https://github.com/IntelRealSense/librealsense/issues/8689

The SDK's **software-device** interface for defining a 'synthetic camera' enables a pointcloud to be generated from types of input other than a bag, such as loading a PNG image.  Information about doing this in C++ can be found in the link below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

**********

You can also create manual custom framesets with the  SDK's **custom processing block**.  C++ code for doing so is provided by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/5847#issuecomment-586718261

**********

The SDK's bag format is based on the ros-bag standard but has some differences that can introduce incompatibilities (for example, being unable to play back an SDK-recorded bag in ROS and vice versa).  This is referenced in https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

https://github.com/IntelRealSense/librealsense/issues/2186 details how it is possible to sync between a RealSense camera and RGB from non-RealSense cameras using camera metadata.

Further C++ information about how the SDK can record and playback bags is in the link below.

https://dev.intelrealsense.com/docs/rs-record-playback

***************

The SDK's **genlock** multiple camera synchronization system also offers the possibility of syncing between a RealSense camera and RGB produced by non-RealSense cameras.  See point **3.  Externally controlled sync** in the link below.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-4-genlock-usages

***********

The link below discusses the RealSense SDK's bag format compared to the ROS rosbag format, and has a link to a rosbag parser project.

https://github.com/IntelRealSense/librealsense/issues/7845#issuecomment-732456123 

SDK bags can also have their topic and message structure read with an SDK tool called rosbag-inspector

https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector

ilyak93, 2021-12-31:
@MartyG-RealSense, wow, thank you very much, a lot of useful links and information, I'll research it further, really helpful.
I'll keep the topic open for a while, for maybe I'll have some questions, otherwise I'll close it.

MartyG-RealSense, 2021-12-31:
You are very welcome, @ilyak93 - yes, it's fine to keep this topic open for a time and post further questions if you have them. Good luck!

Co-stoletta, 2022-02-03:
Hi @ilyak93, I have read your issue and I think that maybe you can help with a problem. I create this code:

        auto ucharptr = reinterpret_cast<unsigned char *>(const_cast<void *>(frame.get_data()));
        int w = frame.get_width();
        int h = frame.get_height();
        Frame video_frame;
        video_frame.width = w;
        video_frame.height = h;
        video_frame.data.resize(w * h * 3 /*RGB*/);
        memcpy(video_frame.data.begin(), ucharptr, size_t(w * h * 3));

That allows me to copy the frame (is a rs2::video_frame) inside the Frame video_frame that I have created.
What I want to ask you is: is there a way to get the distance information (in meters) from the Frame video_frame.data?
Inside those data are "hidden" also the depth info of each pixel in the FOV, is there a way to get this number?

Thanks!! :)

MartyG-RealSense, 2022-02-03:
Hi @Co-stoletta  A common method of getting the real-world distance in meters from the **depth** data is to use the instruction **get_distance()**.  As you use C++, the SDK example program **rs-hello-realsense** provides an example of doing this, printing the depth in meters of the center pixel of the image.

https://github.com/IntelRealSense/librealsense/blob/master/examples/hello-realsense/rs-hello-realsense.cpp#L30

If you need your solution to be based around the **RGB** image then you could consider obtaining the XYZ coodinates of specific pixels on the image with the instruction **rs2_project_color_pixel_to_depth_pixel** which converts a color pixel to a depth pixel.  This method has faster processing than performing depth to color alignment on the entire image.

https://github.com/IntelRealSense/librealsense/issues/4124 has an example of C++ code for obtaining the distance value with rs2_project_color_pixel_to_depth_pixel and printing it out.

Co-stoletta, 2022-02-03:
> Hi @Co-stoletta A common method of getting the real-world distance in meters from the **depth** data is to use the instruction **get_distance()**. As you use C++, the SDK example program [...]

Hi, regarding the first part of your comment: yes I know that I could use the instruction get_distance() but don't want to work with Realsense types (like rs2::frame, rs2::depth_frame etc...), I want to work with the data stored in those types, that's why I'm using .get_data().

Regarding the second part: ok I will read the #4124 and I will post updates.

Thank you! :)

MartyG-RealSense, 2022-02-03:
How about storing point cloud data as vertices and then retrieving the texture coordinates using **tex_coords**

https://github.com/Resays/xyz_rgb_realsense/tree/master

Co-stoletta, 2022-02-03:
> How about storing point cloud data as vertices and then retrieving the texture coordinates using **tex_coords**
> 
> https://github.com/Resays/xyz_rgb_realsense/tree/master

I'm having trouble with the function get_texcolor(...), dunno what I have to include to use this function.
I will post updates.

Thanks

Co-stoletta, 2022-02-04:
Hi @MartyG-RealSense, I was able to introduce that code in my code but now I have a question: the line

auto vertices = points.get_vertices();

provide a pointer that I could use to access all the (x,y,z) of the pixels. For example, if I wanna take the z coordinate of the pixel i,j I can use z = v(j*width + i). 

From this z value, is there a way to get the distance in meters? I thought that dividing for the depth units = 0.001 would be good but I don't know...

Thanks!

MartyG-RealSense, 2022-02-04:
You can obtain a distance in meters by multiplying the raw **uint16_t** depth value by the depth unit scale of the camera.  Though this value can be retrieved from the camera in real-time, if you only plan for your project to be used with 400 Series cameras and not the L515 then you can simply multiply the raw depth value by **0.001** (the default depth unit scale of 400 Series cameras) to obtain the distance in meters.

Co-stoletta, 2022-02-04:
> You can obtain a distance in meters by multiplying the raw **uint16_t** pixel depth value by the depth unit scale of the camera. Though this scale value can be retrieved from the camera in real-time, if you only plan for your project to be used with 400 Series cameras and not the L515 then you can simply multiply the raw depth pixel value by **0.001** (the default depth unit scale of 400 Series cameras) to obtain the distance in meters.

Hi, I solved in the sequent manner:

auto vertices = pc.calculate(allFrames.get_depth_frame()).get_vertices();

Hallow me to get a pointer to the first pixel, and then with:

vertices[y*width + x].z;

You can extract from those data the (x,y) pixel real-world depth.
Thanks!

MartyG-RealSense, 2022-02-04:
Thanks too for continuing to share your solutions  :)

MartyG-RealSense, 2022-02-10:
Hi @Co-stoletta  Do you require further assistance with this case, please?  Thanks!

Co-stoletta, 2022-02-15:
> Hi @Co-stoletta Do you require further assistance with this case, please? Thanks!

Hi, sorry for the late response... Nope, I don't need further assistance.

Thanks!

MartyG-RealSense, 2022-02-15:
Thanks very much @Co-stoletta for the update!  As you do not need further assistance, I will close this case.  Thanks again!

closed 2022-02-15 by MartyG-RealSense