realsenseai/librealsense #10118: D435 disconnect when the camera temperature is 42℃

PCCSYY, 2021-12-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
 I found in the official manual that the ambient temperature for the normal and stable operation of d435 is 0-35 degrees. After the temperature of the camera reaches 60 degrees, the laser module in the camera starts to protect its power. 

In my actual scene, the d435 works in a closed enclosure. The interior of the enclosure is not ventilated with the external environment. There are two control boards with cooling fans in the enclosure. Then, the d435 sometimes disconnects after working for 1 to several hours. It is impossible to find the depth camera device of "Intel crop" using the command of "lsusb", At this time, the log shows that the chip temperature of the camera itself is more than 40 degrees, not more than 60 degrees. 

Would you like to ask whether this disconnection phenomenon is normal in this environment, and whether this scenario or casing design is reasonable ？

thank you for apply

MartyG-RealSense, 2021-12-30:
Hi @PCCSYY  The data sheet for the 400 Series cameras recommends a maximum operating temperature of 35 degrees C.  In practice though you could likely reach 42 degrees C (the temperature that your cameras are at) before problems started manifesting above that temperature.

There is a relationship between the operating temperature and the temperature of the camera's outer casing.  Intel lab tests found that if the external casing temperature is below 44 degrees C then the operating temperature is not likely to exceed 60 degrees C.

Intel have regularly tested the 400 Series cameras in long-run sessions.  These cameras are able to operate indefinitely so long as they remain within recommended operating temperature tolerances.  If a problem does occur in a long-run session, it is likely to be related to the computer hardware, its operating system or the USB hardware (port / hub / controller / cable) rather than a problem with the camera hardware.

PCCSYY, 2022-01-04:
Thank you very much for your reply

I also want to ask again, in my actual scene, if the ambient temperature of the camera exceeds 35 degrees, is it impossible to ensure that the camera can work stably and connect continuously? If possible, what could it be?

Thank you again for your correction

MartyG-RealSense, 2022-01-04:
Once the camera reaches a temperature where problems begin to manifest then there is not much that you can do to correct it except try to reduce that temperature.  In this particular project, the camera's location within an enclosure with no heat exhaustion to the outside of it is likely the main cause, so tweaking the camera stream configuration would make little difference.

Does the camera's operating temperature start at a lower level such as 28 to 35 degrees C and then progressively rise past 40 during the initial hour, perhaps as the trapped air inside the enclosure becomes hotter?

You could consider incorporating a **heat sink** component into your housing design if you are not using one already.  A heat sink is basically a metal block with fins on it that is placed flat on a hot surface of an electronics component and draws heat away from that component.  

https://en.m.wikipedia.org/wiki/Heat_sink

With the caseless **Depth Module Kit** versions of RealSense cameras, you also have the flexibility to incorporate custom heat sink solutions into your design to provide additional cooling, using larger sinks for greater cooling.  Page 81 of the current edition of the data sheet document discusses this.  The depth module equivalent of the D435 is the **D430**.

I wonder though whether the control board cooling fans themselves could be contributing to overheating of the camera by changing airflow in the enclosure in a way that makes it more difficult for heat exhausted from the camera's grilles to be carried away from it.  It is like how a computer must not be cooled by blowing air into its casing through the ventilation grilles, as the grilles are intentionally designed to let hot air be directed **out** of them.  

Replacing the active fans with passive heat sinks might provide control board cooling without enclosure airflow disruption.

It may also be possible to exhaust hot air from the enclosure into the surrounding water by incorporating into the enclosure a **check valve** that only lets liquids or gas (like air) pass through it in one direction.

https://en.m.wikipedia.org/wiki/Check_valve

closed 2022-01-15 by MartyG-RealSense