realsenseai/librealsense #10120: Anti-spoofing detction using depth data

akanksh2kb, 2021-12-30:
| Camera Model                       | { D435 } | 
| Operating System & Version |   {Win (11) | 
| Platform                                 | PC  |
| Language                            |  {/python }                          | 

Hi,
I am Akanksh.

Presently, working Anti-spoofing 

Found this link:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib/face

But didn't understand: How nose_depth, eye_depth etc. are calculated?

And Is there  python version of it?

Thanks,
Akanksh

MartyG-RealSense, 2021-12-30:
Hi @akanksh2kb  The **validate_face.h** Dlib wrapper file contains the code for calculating variables such as nose_depth and describes how these variables are calculated.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/dlib/face/validate_face.h

The 68 facial landmark points of the RealSense Dlib face example program are provided by a trained dataset file downloaded from the link below.

http://dlib.net/files/shape_predictor_68_face_landmarks.dat.bz2

These landmarks are then applied to the RGB image that the camera is observing.  If the landmark is judged to be on a real face then it is drawn in the color green.  If the landmark is judged to be fake (e.g a face image displayed on a phone screen) then the landmark is drawn in the color red.

[image]

Looking at the code of validate_face.h, it looks as though it is not calculating depth like a typical RealSense program does, but instead compares the distances between a set of certain landmarks.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/dlib/face/validate_face.h#L63

For example, the value of the **nose_depth** variable is calculated using the face landmarks NOSE_RIDGE_2 and NOSE_TIP.  The **eye_depth** variable meanwhile is calculated by comparing a pair of landmarks on the left eye and right eye separately, and then using those two results for the left and right eye to calculate the final eye_depth value.

*************

In regard to a Python version of the Dlib example: there is not an exact equivalent for it.  The Dlib face detection library does provide support for Python though.  If you go to the link below and find the **Help / Info** section in the side-panel, expanding open the **Examples: Python** section shows a list of Dlib face sample programs for Python.

http://dlib.net/

[image]

A RealSense team member in another case also recommended the Dlib approach on Python, suggesting in https://github.com/IntelRealSense/librealsense/issues/4430#issuecomment-511660804 to google for the term **facial-landmarks-dlib-opencv-python**

akanksh2kb, 2021-12-31:
Thanks a ton..

closed 2021-12-31 by akanksh2kb