realsenseai/librealsense #10121: Windows 10 cannot detect a USB device

Marmot01, 2021-12-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| SDK Version                            |  { legacy / 2.<4>.<9> }                          | 
|                | 

### Issue Description
I connected the camera to my computer via USB, but my computer did not detect the device. The SDK version installed is 2.4.9. I tried to change the SDK version but it didn't work

MartyG-RealSense, 2021-12-31:
Hi @Marmot01   First, please check in the Windows camera privacy settings that access to the camera hardware and permission for applications to access the camera are both set to On.  Instructions for doing so are here:

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

If these two settings are already On, next try unplugging the USB cable from the base of the L515 camera, turning the connector around the opposite way (USB-C cables are two-way insertion on the micro-sized end) and plugging the connector back into the base of the camera.

Marmot01, 2022-01-04:
Hello, the privacy permission setting of the camera has been opened, and the USB cable has been reconnected as you said, but my system still cannot detect the device

MartyG-RealSense, 2022-01-04:
Is the camera plugged into a USB 3 port on the computer?  If you are using the camera in a USB 2 port then it is capable of using USB 2, but it first has to have a firmware update performed whilst in a USB 3 port.

Marmot01, 2022-01-05:
The camera is plugged into the USB 3 port. At the beginning of use, the computer once detected the camera device, but later, no matter how you try, the computer can't detect it

MartyG-RealSense, 2022-01-05:
Can you confirm whether the camera is listed in the Windows Device Manager under its **Cameras** category, please.

[image]

Also, has the camera been stored unused in a cool dry place such as a cupboard or drawer since the last time that it worked and then did not work after having been taken out of storage?

Marmot01, 2022-01-05:
The camera device is not detected in the device manager of windows. The camera has been placed in a ventilated and dry place after use
[image]

MartyG-RealSense, 2022-01-05:
How long was it stored unused in the ventilated and dry place before the problem occurred (weeks, months, etc)

Marmot01, 2022-01-05:
About half a month

MartyG-RealSense, 2022-01-05:
Thank you for the confirmation.  There have been a small number of past cases where RealSense cameras have stopped functioning correctly after being stored in a cool dry place for a couple of weeks or more.

In those cases, the cameras have tended to recover on their own a few weeks after coming out of storage.  For example, one camera was returned from India to Intel in California and by the time that it arrived in California it was working again.

So your options in this case may be:

1.  Try applying power to the camera once a day for a few weeks to see whether it recovers.  This may not be practical in your case though if the camera cannot even be detected.   Or

2.  Return the camera to the retailer that you purchased it from using the **Return Material Authorization (RMA)** returns procedure of that retailer.

MartyG-RealSense, 2022-01-11:
Hi @Marmot01  Do you require further assistance with this case, please?  Thanks!

CyborgNature, 2022-01-17:
I have the exact same issue and don't know what to do now!

CyborgNature, 2022-01-17:
@MartyG-RealSense can you kindly guide the process of returning the L515 using (RMA)

MartyG-RealSense, 2022-01-17:
Hi @CyborgNature  If you decide to return your L515 then you should contact the retailer that you purchased it from to find out what their return policies are.  

If you purchased it from the official RealSense online store then I recommend contacting them using the contact web-page at the link below.

https://www.intel.com/content/www/us/en/forms/support/intel-realsense-store-inquiry.html

CyborgNature, 2022-01-17:
Thank You @MartyG-RealSense for response.

MartyG-RealSense, 2022-01-28:
Hi @CyborgNature  Do you have an update that you can provide about how the return process is proceeding for you please?  Thanks!

MartyG-RealSense, 2022-02-03:
This case will be closed after 7 days from the time of writing this if we do not receive a further comment.  Thanks!

closed 2022-02-09 by MartyG-RealSense