realsenseai/librealsense #10122: Support Risc-V

lingffff, 2021-12-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (OE)  | 
| Kernel Version (Linux Only)    |  5.13.17  | 
| Platform                                 | riscv64 |
| SDK Version                            |  newest | 
| Language                            |  C  | 
| Segment			|  others  | 

### Issue Description

When building librealsense on Risc-V (Sifive unmatched U740), I got below CMakeError:
```
Determining if the include file pthread.h exists failed with the following output:
Change Dir: /run/media/sda1/librealsense/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make -f Makefile cmTC_47e96/fast && /usr/bin/make  -f CMakeFiles/cmTC_47e96.dir/build.make CMakeFiles/cmTC_47e96.dir/build
make[1]: Entering directory '/run/media/sda1/librealsense/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_47e96.dir/CheckIncludeFile.c.o
/usr/bin/cc   -pedantic -g -D_DEFAULT_SOURCE   -mssse3 -pthread  -fPIE -o CMakeFiles/cmTC_47e96.dir/CheckIncludeFile.c.o -c /run/media/sda1/librealsense/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
cc: error: unrecognized command-line option '-mssse3'
make[1]: *** [CMakeFiles/cmTC_47e96.dir/build.make:78: CMakeFiles/cmTC_47e96.dir/CheckIncludeFile.c.o] Error 1
make[1]: Leaving directory '/run/media/sda1/librealsense/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:127: cmTC_47e96/fast] Error 2

```
Seems like librealsense can not get correct CMAKE config on Risc-V.  Can librealsense  support Risc-V? 
Thanks.

MartyG-RealSense, 2021-12-31:
Hi @lingffff  There are no previous cases on this GitHub support forum of RISC-V being used with librealsense2.  A RISC-V website at the link below does list packages for librealsense2 though up to version **2.38.0** (librealsense is at version 2.50. at the time of writing this)

My understanding is that Risc-V is related to Arm architecture, with RISC-V being open-source and Arm proprietary.  The error **unrecognized command-line option '-mssse3'** suggests that the build is having problems due to the absence of SSE3 technology that is incorporated into Intel x86 processor architecture.

If the RISC-V librealsense packages in the above link are not helpful then I would suggest trying to build librealsense from source code using CMake with the **RSUSB backend** method that is used with Arm-based Nvidia Jetson boards.  Instructions for performing an RSUSB build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

MartyG-RealSense, 2022-01-07:
Hi @lingffff  Do you require further assistance with this case, please?  Thanks!

lingffff, 2022-01-09:
Hi, 
Sorry that I was busy with other things and I did the tests just now.
The OS on my Risc-V board is OE version from Yocto, so I can't install rpm package from Fedora.
As for second solution, it did not work and I got same error as previously mentioned.
Below is my command line:
```bash
root@unmatched:/run/media/sda1/librealsense/build# cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.50.0
-- Setting Unix configurations
-- Building libcurl enabled
-- using RS2_USE_LIBUVC_BACKEND
-- GLFW 3.3 not found; using internal version
CMake Error at /usr/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /usr/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.21/Modules/FindThreads.cmake:238 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  third-party/glfw/CMakeLists.txt:63 (find_package)

-- Configuring incomplete, errors occurred!
See also "/run/media/sda1/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/run/media/sda1/librealsense/build/CMakeFiles/CMakeError.log".
root@unmatched:/run/media/sda1/librealsense/build#
root@unmatched:/run/media/sda1/librealsense/build#
root@unmatched:/run/media/sda1/librealsense/build#
root@unmatched:/run/media/sda1/librealsense/build# cat CMakeFiles/CMakeError.log
Determining if the include file pthread.h exists failed with the following output:
Change Dir: /run/media/sda1/librealsense/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make -f Makefile cmTC_52c66/fast && /usr/bin/make  -f CMakeFiles/cmTC_52c66.dir/build.make CMakeFiles/cmTC_52c66.dir/build
make[1]: Entering directory '/run/media/sda1/librealsense/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_52c66.dir/CheckIncludeFile.c.o
/usr/bin/cc   -pedantic -g -D_DEFAULT_SOURCE   -mssse3 -pthread  -fPIE -o CMakeFiles/cmTC_52c66.dir/CheckIncludeFile.c.o -c /run/media/sda1/librealsense/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
cc: error: unrecognized command-line option '-mssse3'
make[1]: *** [CMakeFiles/cmTC_52c66.dir/build.make:78: CMakeFiles/cmTC_52c66.dir/CheckIncludeFile.c.o] Error 1
make[1]: Leaving directory '/run/media/sda1/librealsense/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:127: cmTC_52c66/fast] Error 2
```

MartyG-RealSense, 2022-01-09:
My research found a case at https://github.com/IntelRealSense/meta-intel-realsense/issues/15 where someone encountered the **error: unrecognized command-line option '-mssse3'** error when building librealsense for Intel's old RealSense flying drone system, which used Yocto as its OS via a compatibility wrapper called **meta-intel-realsense**.

In that particular case the RealSense user's workaround ws to make edits to CMakeLists.txt.  I also saw other non-RealSense cases of this error where it was resolved by editing CMakeLists.txt to remove references to SSE3.  Without detailed instructions about which CMakeLists.txt file(s) to edit and where they are located, this could be difficult to implement though.

A related case with librealsense compilation problems in Yocto is at https://github.com/IntelRealSense/meta-intel-realsense/issues/14

lingffff, 2022-01-10:
Thanks for your reply. 
It seems that meta-intel-realsense is a little difficult for me since I have not tried to build Yocto myself.  
I will try this solution later. Or, if there is any simple solution to edit CMakeLists.txt in librealsense2 to patch out the -mssse3 problem?

MartyG-RealSense, 2022-01-10:
In the meta-intel-realsense patch, it looks as though they edit a CMake instruction called **/CMake/unix_config.cmake**

https://github.com/IntelRealSense/meta-intel-realsense/blob/master/recipes-support/librealsense/files/Enable-riscv-arch-compilation.patch

The above link shows the original instruction in red and the edited version of the instruction in green.

In the librealsense SDK, this instruction is located in the **include_os.cmake** file in the 'librealsense > CMake' folder.

https://github.com/IntelRealSense/librealsense/blob/5ff27fca5aaeec4736d6bb3bfb958fee82ee198b/CMake/include_os.cmake#L6

MartyG-RealSense, 2022-01-16:
Hi @lingffff  Do you have an update about this case that you can provide, please?  Thanks!

lingffff, 2022-01-17:
Hi,
I edited **CMake/unix_config.cmake** following the meta-intel-realsense patch. 
Now cmake is passed, but get make error below.
```bash
/run/media/sda1/librealsense/third-party/glfw/include/GLFW/glfw3.h:207:13: fatal error: GL/glu.h: No such file or directory
  207 |    #include <GL/glu.h>
      |             ^~~~~~~~~~
compilation terminated.
```
I have tried to look for the switch of GLU in some CMake instructions but failed. Is there any advice? Thanks.

MartyG-RealSense, 2022-01-17:
A RealSense team member offers advice about the **fatal error: GL/glu.h: No such file or directory  #include <GL/glu.h>**   error at https://github.com/IntelRealSense/librealsense/issues/5232#issuecomment-552941255

closed 2022-01-29 by MartyG-RealSense