realsenseai/librealsense #10123: Point cloud data has a lot of noise with D435

Tom127, 2021-12-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   |   | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 |   |
| SDK Version                            |                       | 
| Language                            |  C++                          | 
| Segment			|                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
@MartyG-RealSense  Hi, I'm using [ros-driver](https://github.com/IntelRealSense/realsense-ros) to get point-cloud from D435. We were in an open field, the point cloud data obtained by the camera at a height of 15 cm, as can be seen from the video below, there are a lot of point cloud noise on the flat ground. 

https://user-images.githubusercontent.com/34730402/147808219-1e1b849f-5e30-49b8-b988-e74e37516abb.mp4

These noises bring us a lot of resistance in obstacle detection. Is there any setting to reduce or avoid this kind of noise?
Best regards,
Tom

MartyG-RealSense, 2021-12-31:
Hi @Tom127  In a project where the camera is mounted low to the ground, this may cause problems for obstacle detection as the ground directly in front of the camera will register as close-range depth and may mis-trigger the obstacle detector.  

In te librealsense SDK, it helps to set up a **threshold filter** post-processing filter, which enables a minimum and maximum observable depth sensing distance to be defined.  Depth values outside of the defined min-max range will be excluded from the depth image.

The RealSense ROS wrapper does not support the threshold filter though - it can only set a maximum observable distance wit the **clip-_distance** instruction.  It may be possible to set a minimum distance in ROS by using the 'RangeMin' setting of **rtabmap_ros** though, as described in https://github.com/IntelRealSense/realsense-ros/issues/1859

MartyG-RealSense, 2022-01-07:
Hi @Tom127  Do you require further assistance with this case, please?  Thanks!

Tom127, 2022-01-12:
> Hi @Tom127 Do you require further assistance with this case, please? Thanks!

Hi, @MartyG-RealSense Sorry for the late reply! We currently need the detection distance of the camera to be as close as possible, and the detection target to be as small as possible. Elevating the camera or tilting the installation cannot meet the application requirements. 

Best regards,
Tom

MartyG-RealSense, 2022-01-12:
These are similar issues to the ROS case https://github.com/IntelRealSense/librealsense/issues/10128 discussed during the past week.  In that case I suggested using a **RANSAC** algorithm to extract the ground data, but it did not work well for the RealSense user in that particular case.  So at their request, at https://github.com/IntelRealSense/librealsense/issues/10128#issuecomment-1010083835 I suggested some alternative approaches to obstacle detection in ROS.

closed 2022-01-29 by MartyG-RealSense