realsenseai/librealsense #10124: Syncronization strategy of the SDK when enabling multiple sensor streams

zc08, 2022-01-02:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Ubuntu 18.04.3 |
| Kernel Version (Linux Only) | 5.4.0-48-generic |
| Platform | PC |
| SDK Version | 2.48.0 |
| Language | C/python |
| Segment | Robot |

### Issue Description
I have recorded a ROS bag file using realsense-viewer, enabling all sensor streams (depth, color, gyro, accel).  The framerates of color and depth are both set to 30fps, gyro is set to 400fps, and accel set to 250fps. Now I want to obtain the recorded color/gyro/accel data from the bag file. However, the data timestamps I get when enabling multiple streams seems confusing to me.

1. When enabling both gyro and accel streams, and printing out the timestamps as below
```
config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 400)
config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 250)
```
```
frames = pipeline.wait_for_frames()
gyro_frame = frames.first(rs.stream.gyro).as_motion_frame()
print(f'{frame_cnt} -  gyro time: {gyro_frame.timestamp}')
```
I get the following results. The accel and gyro with the same index come from the same frameset.
```
100 - accel time: 1640923848565.5867
100 - gyro time: 1640923848568.465
101 - accel time: 1640923848569.5713
101 - gyro time: 1640923848568.465
102 - accel time: 1640923848569.5713
102 - gyro time: 1640923848570.9512
103 - accel time: 1640923848601.449
103 - gyro time: 1640923848570.9512
104 - accel time: 1640923848617.3884
104 - gyro time: 1640923848570.9512
105 - accel time: 1640923848617.3884
105 - gyro time: 1640923848670.3784
106 - accel time: 1640923848617.3884
106 - gyro time: 1640923848702.6907
107 - accel time: 1640923848705.0537
107 - gyro time: 1640923848702.6907
108 - accel time: 1640923848705.0537
108 - gyro time: 1640923848705.1763
109 - accel time: 1640923848705.0537
109 - gyro time: 1640923848707.662
110 - accel time: 1640923848709.0378
110 - gyro time: 1640923848707.662
```
It seems gyro or accel data are duplicated whenever a new measurement arrives.

2. When enabling both color stream and gyro stream, I get:
```
120 - gyro time:  1640923852120.2817
120 - color time: 1640923852054.2375
121 - gyro time:  1640923852127.7964
121 - color time: 1640923852087.7212
122 - gyro time:  1640923852177.8997
122 - color time: 1640923852121.2075
123 - gyro time:  1640923852177.8997
123 - color time: 1640923852154.6946
124 - gyro time:  1640923852235.5227
124 - color time: 1640923852188.1843
125 - gyro time:  1640923852255.5671
125 - color time: 1640923852221.675
126 - gyro time:  1640923852320.7153
126 - color time: 1640923852255.1685
127 - gyro time:  1640923852325.7273
127 - color time: 1640923852288.6626
128 - gyro time:  1640923852390.7961
128 - color time: 1640923852322.1597
129 - gyro time:  1640923852425.643
129 - color time: 1640923852355.808
130 - gyro time:  1640923852450.5344
130 - color time: 1640923852389.0833
```
It seems that: 
1. Some gyro data are discarded to match the framerate of the color stream (30fps). This "discard strategy" seems different from the "duplication strategy"  when gyro and accel streams are enabled.
2. The color and gyro times are not optimally matched according to the timestamps. E.g. for frameset 126, the color time is 1640923852255.1685, the corresponding gyro time is 1640923852320.7153. But obviously the gyro time 1640923852255.5671 would be a better match.

What I want to ask are:
1. The precise meaning of `frame.timestamp` field in `pyrealsense2`. 
2. The synchronization strategy adopted by the SDK (i.e. pyrealsense2).
2. How can I properly get the timestamps of all the color frames and gyro frames (No discarding), so that the closest timestamps are the correct match? (I need these data to do video stabilization)

Thanks in advance!

MartyG-RealSense, 2022-01-02:
Hi @zc08   Thanks for your questions.

1.  Intel's official pyrealsense2 documentation defines **frame.timestamp** simply as the time at which the frame was captured.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.frame.html

2.  An Intel guide to IMU data in the link below advises about IMU-depth sync that "The D435i depth camera generates and transmits the gyro and accelerometer samples independently, as the inertial sensors exhibit different FPS rates (200/400Hz for gyro, 63/250Hz for accelerometer).  Each IMU data packet is timestamped using the depth sensor hardware clock to allow temporal synchronization between gyro, accel and depth frames".

As the D455 model is an enhanced evolution of the D435i, I would assume the above statement about D435i to also be true for D455.

https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

In regard to syncing timestamps of IMU and color, a RealSense team member provides advice about this in https://github.com/IntelRealSense/librealsense/issues/3205

3.  For a closest match between timestamps of different sensors, i would recommend that **global time** is enabled.  an example of Python code for enabling global time is at https://github.com/IntelRealSense/librealsense/issues/8205#issuecomment-788159385

zc08, 2022-01-03:
Thank you for the reply!
I have enabled the global time option in the realsense-viewer when recording. And I'm trying to check out all the timestamp related fields (i.e. sensor_timestamp, frame_timestamp, back_end_timestamp, time_of_arrival_timestamp, and the timestamp attribute itself)

The color and gyro frame below come from one frameset.
[image]

From the `get_frame_timestamp_domain` function, we can see that the `timestamp` returned is in the `global_time` domain. 

But I still have two questions:
1. As mentioned in the question above, the color and gyro data in a single frameset doesn't seem to be the closest match by the `timestamp` field, even when `global time` option is enabled.
2. The global time (`timestamp`) is different from any of the four timestamps returned, I'm wondering what's the exact meaning of this field.
3. For the gyro frame, the `timestamp` is greater than both the `time_of_arrival` and `backend_timestamp`.  I have presumed that `time_of_arrival` would be the latest timestamp associated with a frame, but it seems that's not true?

MartyG-RealSense, 2022-01-03:
1.  This is a somewhat difficult question.  As mentioned above, the IMU data is timestamped with the depth sensor hardware clock.  There is not much available information about sync between IMU and **color** timestamps though.  My understanding - as described by a RealSense team member in https://github.com/IntelRealSense/librealsense/issues/1548 - is that if **wait_for_frames()** is used in a script then wait_for_frames provides best-match mapping between the color and depth sensors.  

So my speculation (not certainty) would be that if IMU is timestamped with the depth sensor clock and wait_for_frames() is used to best-match the color timestamp to depth, then overall the color, depth and IMU timestamps may end up synced. 

2.  If you would like further information about the workings of the global timestamp, https://github.com/IntelRealSense/librealsense/issues/4505#issuecomment-553590316 is an excellent reference.

3.  A RealSense team member provides explanations in https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803 of the different types of timestamp, including **time_of_arrival** and **backend_timestamp**.

zc08, 2022-01-03:
Thank you!
1. I have found an explanation about the synchronization strategy in realsense-ros issues [here](https://github.com/IntelRealSense/realsense-ros/issues/729#issuecomment-525623221). But it claims to be somehow outdated now. I'm wondering whether there are more up-to-date references.
2. Thank you for the information!
3. My major question was, according to this explanation, `time_of_arrival` basically should be the last timestamp a frame has. But the global `timestamp` is actually later (greater) than the `time_of_arrival` timestamp, which is confusing to me.

MartyG-RealSense, 2022-01-03:
1.  There was a follow-up in https://github.com/IntelRealSense/realsense-ros/issues/898 to Doronhi's interpolation explanation.  Doronhi (who is the RealSense ROS wrapper developer).  My interpretation of the information in that case is that Doronhi updated the linear_interpolation system to integrate changes suggested by another developer called Chutsu at their GitHub at the link below.

https://github.com/chutsu/rs4se
 
When using the ROS wrapper, IMU timestamp problems can occur for some RealSense ROS users that disappear when global time is disabled.  It is my understanding that in the RealSense ROS wrapper, global time is enabled by default but having global time enabled does not have as much impact as it does in the librealsense SDK.  This is because unlike librealsense, the ROS wrapper can provide time correction when global time is disabled.  This is described by Doronhi the RealSense ROS wrapper developer in https://github.com/IntelRealSense/realsense-ros/issues/796#issuecomment-497997643

2.  You are very welcome!

3.  The advice from a RealSense team member in https://github.com/IntelRealSense/librealsense/issues/7837#issuecomment-754970222 may be relevant to this question.

closed 2022-01-15 by MartyG-RealSense