realsenseai/librealsense #10125: D455 raw color image with

arainbilal, 2022-01-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                | |
|---------------------------------|------------------------------------------- | ----------------------------------|
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } |  D455  |
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 05.13.00.50
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | Ubuntu 20.04
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  | PC
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 2.50.0
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | python
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | others

### Issue Description
References: #8716 #7275 #8025 #7275 #701 #1430 @MartyG-RealSense 

The D455 distortion coefficients are non-zero, as stated in issue #1580. How to obtain the raw (unrectified) colour image using D455? The output of `rs-enumerate-devices -c` is attached for reference. Also attached the dynamic calibration results.
[device_calib.txt](https://github.com/IntelRealSense/librealsense/files/7801858/device_calib.txt)

arainbilal, 2022-01-03:
@MartyG-RealSense @RealSenseSupport 
One of the methods could be to write the write custom calibration from a text file in xml format into the D455 by setting the distortion coefficients to zero. Please confirm if this is the right method to use to be able to obtain the unrectified raw image from the RGB camera of D455? Thanks.

MartyG-RealSense, 2022-01-03:
Hi @arainbilal  The **Y16** format is unrectified, as it is used for camera calibration.  Y16 is available from the list of RGB formats supported by the D455 camera model, though it is a monochrome image.  You can test it in the RealSense Viewer tool.

[image]

[image]

If you require a color image, an alternative method would be to access RGB raw frames from the camera hardware and convert them to a usable format using the OpenCV **cvtColor** instruction.  https://github.com/IntelRealSense/librealsense/issues/7275 describes a method for doing this in Python.

arainbilal, 2022-01-03:
Thanks @MartyG-RealSense. Please note the output of the following code snippet 
`
config.enable_stream(rs.stream.color, 0, 1280, 800, rs.format.y16, 30)
rgb_intrinsic_cfg = profile.get_stream(rs.stream.color)
rgb_intrinsic = rgb_intrinsic_cfg.as_video_stream_profile().get_intrinsics()
`
is as follows
`
[ 1280x800  p[635.427 409.187]  f[640.961 639.651]  Inverse Brown Conrady [-0.0547421 0.0613225 -0.000308431 -0.000209734 -0.0195954] ]
`
This is using the **Y16** format. I wasn't hoping to find the non-zero distortion coefficient in this mode. Could you please explain why I would get non-zero distortion coefficients if this is unrectified image? Thanks.

P.S. I am happy with the grayscale unrectified image

MartyG-RealSense, 2022-01-03:
For the 400 Series cameras, most of the coefficients are **artifically** set to zero by intentional design, as described in https://github.com/IntelRealSense/librealsense/issues/1430#issuecomment-375945916

RGB on the D455 model has non-zero coefficients.  As far as I am aware though, it makes little difference compared to an RGB stream from a camera model such as D435i where the color coefficients are zero.  Coefficients and a Y16 RGB image from a D435i are shown below.

[image]

[image]

The image below shows non-zero color coefficients from a D455, though the values are so close to zero that they may as well be zero.

arainbilal, 2022-01-03:
Thanks. Will use the Y16 format as stated in the comments.

closed 2022-01-03 by arainbilal