realsenseai/librealsense #10126: Question - Reduce RGB resolution on post capture but get same results as capture

Camilochiang, 2022-01-04:
Hei.
- I'm using the input of the camera  (RGB) for two different process (3D reconstruction and machine learning). I would like to transform the RGB frame from _1920x1080_ to _640x480_  on my computer as one of the process need to work at lower resolution. If I do that using any standard library (in my case, in python) this produce a change in the aspect ratio and everything look differently.  I have no camera with me at the moment  to test if I take the images directly with setting the camera at for example _640x480_ are more or less  the same as reducing the resolution after capture.
 - Question: Is there a way/open code to check how to reduce resolution but get similar results as if I would be taking the image in that resolution?
 Example code to compare results
```
import pyrealsense2 as rs
import numpy as np
import cv2

# Get the full size image
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1920, 1080, rs.format.z16, 30)
....
color_frame_full = frames.get_color_frame()
color_frame_full = cv2.resize(cv_image, (640,480), interpolation = cv2.INTER_AREA)
....

# Get the original small image
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
....
color_frame_reduced = frames.get_color_frame()
```
Where color_frame_full and color_frame_reduced should give similar results.
  
 I hope that my question is clear. Thanks in advance!

MartyG-RealSense, 2022-01-04:
Hi @Camilochiang  A post-processing filter called **Decimation** can be used to reduce the resolution of depth and RGB streams.  The decimation filter can be tested in the RealSense Viewer tool by going to the **RGB** section of the options side-panel and expanding open the **Post-Processing** category, then expanding open the **Decimation Filter** option.  Increasing the decimation value from its default of '2' intelligently reduces the resolution.

[image]

The RGB images below, taken in the Viewer with a D435i, show a scene at the default of '2' and at the maximum of '8'.

**Decimation = 2**

[image]

**Decimation = 8**

[image]

*************

If you need the resolution to be directly changed from one to another, another approach would be to define custom stream configurations with a **cfg** instruction (as you are already doing in your script above).  Start the program with a 1920x1080 RGB resolution defined and perform the 3D reconstruction activity first.  Then stop the pipeline, set a new cfg configuration of 640x480 RGB and re-start the pipeline to continue at the low resolution to perform the machine learning activity.

Camilochiang, 2022-01-04:
Hei @MartyG-RealSense  Thanks for your answer once again!

The problem is that I need both resolutions at the same time . No suggestions how to achieve the same results than decimation on python? If you can tell me where i can find the code, I can also have a look.

Thanks!

MartyG-RealSense, 2022-01-04:
You could possibly use **multiprocessing** on Python to do two different activities at the same time.  For example, a past use of it was to stream continuously whilst also independently saving frames to file without interrupting the stream.  Details of that case are in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360033158094-Recording-Video-in-60fps-with-threading-and-queue-Python-

There are not many Python multiprocessing scripting references available though other than the ones in the links below.

https://github.com/IntelRealSense/librealsense/issues/7365

https://github.com/IntelRealSense/librealsense/issues/9085

https://github.com/IntelRealSense/librealsense/issues/7968#issuecomment-744538550

Camilochiang, 2022-01-04:
Great. Thanks Marty! i Also found that `scikit-image` has implemented a working version of downsampling  /decimation.

Thanks!

Camilochiang, 2022-01-04:
Sorry, btw. This do not solve the problem of image aspect, as decimation [preserve the aspect ratio](https://dev.intelrealsense.com/docs/post-processing-filters). Any suggestions?

MartyG-RealSense, 2022-01-04:
Another approach you could consider would be to set up two separate pipelines with their own individual cfg config.  There is a Python example of doing this in https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-390840337

closed 2022-01-04 by Camilochiang