realsenseai/librealsense #10127: rosbag data number

gaoyi-2016, 2022-01-04:
d435i    firmware version: 05.12.09.00       librealsense:2.40.0           realsense-ros:2.2.20

when i use realsense-viewer  to get rosbag :
the number of rgb and depth is same:
[image]
[image]

but when i use realsense-ros to get rosbag  (roslaunch realsense2_camera rs_rgbd.launch):
the number of rgb and depth is different:
[image]
[image]

why is it?because i must use ros to get rosbag,so i hope to get your help,thanks

MartyG-RealSense, 2022-01-04:
Hi @gaoyi-2016  The RealSense SDK (librealsense) uses a bag format that is based on the ROS rosbag standard but has some differences - as described by a RealSense team member in https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766 - and so an SDK-recorded bag may not be the same as one created in ROS with rosbag record. 

There is also the difference that your rosbag is recording an aligned topic, whilst the bag format used by the RealSense SDK and its tools such as RealSense Viewer does not save aligned data to the bag.

gaoyi-2016, 2022-01-04:
but when i  use realsense-ros to get rosbag(roslaunch realsense2_camera rs_camera.launch)
this is not aligned rgb and depth
the number is same
[image]
[image]

but if i want to get aligned rgd and depth,what can i do to insure they are the same number.

gaoyi-2016, 2022-01-04:
@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
May I confirm if you have used **roslaunch realsense2_camera rs_camera.launch align_depth:=true** to obtain an aligned topic with rs_camera.launch?

gaoyi-2016, 2022-01-04:
if i not add align_depth:=true,the number is same
[image]

if i add align_depth:=true,the number is different 
[image]

gaoyi-2016, 2022-01-04:
so,if i can understand it is a aligned  bug that i must face.@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
Depth-color alignment is a processing intensive function that may lead to dropped frames if the computer's CPU has to work harder.

What computer / computing device are you using, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc)

gaoyi-2016, 2022-01-04:
[image]

gaoyi-2016, 2022-01-04:
its my computer enviroment,does it cause this problem?@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
Okay, thank you, that is a strong specification.

MartyG-RealSense, 2022-01-04:
Let's try opening the **dynamic_reconfigure** interface after launch and disabling the RGB option **auto_exposure_priority** using the instructions in https://github.com/IntelRealSense/realsense-ros/issues/2142#issuecomment-953669299

This should force the FPS to remain constant instead of being allowed to vary if performance drops.  What mesage numbers do you get if auto_exposure_priority is disabled, please?

gaoyi-2016, 2022-01-04:
auto_exposure_priority  is  disabling  originally,so i dont  change it
[image]

it may have other causes

gaoyi-2016, 2022-01-04:
@MartyG-RealSense

gaoyi-2016, 2022-01-04:
Does it  have other cause?

gaoyi-2016, 2022-01-04:
its my rgbd dynamic_reconfigure
[image]
@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
The default RGB exposure of the 400 Series cameras is **156**.  So you could change '166' to '156' to eliminate that as a cause.  Also try disabling **global_time_enabled** to eliminate that as a cause too, please

gaoyi-2016, 2022-01-04:
i change the reconfigure  to next picture
[image]

but the problem is still there
[image]
@MartyG-RealSense

gaoyi-2016, 2022-01-04:
but i find it has reduce difference
for an example,in the same time,
in the past,it  may differ 50 picture,
but now,it may differ 20picture.

but whether i can solve the problem totally.@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
All of your other values for the settings in dynamic_reconfigure above are the correct default values. 

It may be best to judge when it is 'good enough' rather than spend much more time trying to achieve a perfect result.

Are the exposure and gain values in the **depth** section of dynamic_reconfigure exposure = 8500 and gain = 16 ?

gaoyi-2016, 2022-01-04:
i may find the cause.
in the same time,
if the camera move a little even static,the color-depth alignment requirement is little and CPU can satisfy,and frame dont drop or drop a little
[image]
[image]

but if the camera move much,the color-depth alignment requirement is much and CPU cannt satisfy,and frame  drop or drop much
[image]
[image]

is it true?@MartyG-RealSense

gaoyi-2016, 2022-01-04:
[image]

gaoyi-2016, 2022-01-04:
depth section 
[image]

MartyG-RealSense, 2022-01-04:
I recall a case at https://github.com/IntelRealSense/librealsense/issues/4958 where a RealSense user had reduced performance when the camera was being moved.  A RealSense team member in that case suggested first trying to turn off auto_exposure_priority (which you have already done).  Then they suggested turning off auto-exposure and using a manual exposure value, and that was what corrected the loss of performance during motion in that particular case.

gaoyi-2016, 2022-01-04:
i think my problem may differ from the case
i think when i use alignment function whenever in a little or much move,the CPU cannt satisfy
so it may cause this problem,
if i convert rosbag to rgb and depth.
in rgb,there are  so close two picture,
[image]
but depth dont have it,so rgb number is more the depth number

if i say my CPU cannt satisfy aligenment real time fuction ,so it cause this problem ,is it true?
@MartyG-RealSense

gaoyi-2016, 2022-01-04:
as this problem is only in color-depth alignment function, if not alignment it isnt.
so the hardware such as CPU performance may  cause it,can you give me some advice,thanks.
@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
You have a powerful CPU, so I would expect it to perform well.  You could though check CPU percentage usage when align is enabled by using a system monitoring tool to see if the CPU % usage goes near 100.  On Ubuntu a monitoring tool such as **htop** could be used.

gaoyi-2016, 2022-01-04:
Because i use two system windows and ubuntu，
ubuntu is only 200g，when use ubuntu usually，i feel a little delay.

when align,the CPU % usage goes near 100,so it may cause the problem
[image]

how can i solve it?@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
You could try building librealsense from source code with CMake and include the term **-DBUILD_WITH_OPENMP=true** in the CMake build instruction.  This allows librealsense to take advantage of multiple CPU cores when performing depth-color alignment.  OpenMP can reduce latency, though at a cost of higher CPU % usage.  

If librealsense is built again then you would also have to build the ROS wrapper again afterwards.

gaoyi-2016, 2022-01-04:
I dont  -DBUILD_WITH_OPENMP=true in librealsense, but now i use multiole CPU,wht is it?
[image]

gaoyi-2016, 2022-01-04:
I dont -DBUILD_WITH_OPENMP=true in librealsense, but now i use multiole CPU,wht is it?
[image]
@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
Looking again at your CPU code number i7-6700HQ, it is less capable than I first thought.  The '6' means that it is an Intel Core 6th Generation processor (codename Skylake), which dates back to 2015-2016.  It is a 4-core (quad core) processor with 8 threads.  

RealSense 400 Series cameras can work well with underpowered computers because the camera contains its own data processing hardware.  Alignment is processed on the computer CPU and not the camera hardware though, so an old CPU would be a disadvantage.

gaoyi-2016, 2022-01-04:
Yes,i think so,i bought it in 2016.
I do it as your above method , it solve my problem in a certain extent,
but my old CPU may the source of problem,i think i will change my computer now.
It takes your a lot of time to help me solve this small problem,thank very much.
Hope you a good luck.
@MartyG-RealSense

MartyG-RealSense, 2022-01-04:
You are very welcome, @gaoyi-2016 - it's no trouble, and I'm pleased that I could be of help.  :)

gaoyi-2016, 2022-01-04:
Through your help,the color-depth alignment frame rate changes  from 0.045 to 0.035(true:0.033,remaining is about old CPU),it seems small but important,now i will go to bed and have a good night,goodbye.
YOU ARE IN PATIENCE AND A GOOD PERSON.
@MartyG-RealSense

MartyG-RealSense, 2022-01-05:
You are very welcome.  That's great news that you achieved the desired result, @gaoyi-2016   :)

MartyG-RealSense, 2022-01-11:
Hi @gaoyi-2016  Do you require further assistance with this case, please?  Thanks!

gaoyi-2016, 2022-01-11:
No,that's all. Thanks.
You could close this issue.

MartyG-RealSense, 2022-01-11:
THanks very much @gaoyi-2016 for the update.  As you suggest, I will close the issue.  Thanks again!

closed 2022-01-11 by MartyG-RealSense