realsenseai/librealsense #10128: How to clear the ground point cloud?

ju-mingyue, 2022-01-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |  Linux (Ubuntu18.04.5)  | 
| Kernel Version (Linux Only)    |  4.9.201                                     | 
| Platform                                 | PC |
| SDK Version                            |2.50.0                        | 
| Language                            | ROS packages                           | 
| Segment			|  Robot                         | 
### Issue Description
@seebq  I want to use point cloud to laser signal to avoid obstacles, but because my camera has a bird's-eye view angle, I need to ignore the ground. At present, I am debugging the clip_distance parameter of the camera. After modifying it to 0.5m, the depth range of the camera is 0.5. m, but obstacles below 15cm cannot be detected. Do you know how to adjust this problem? Or do you know how other manufacturers solve this problem?
At the same time, I also tried to use the point cloud alone for obstacle avoidance, but unfortunately the point cloud will add the ground as an obstacle, making the robot unable to plan the path. Is there any other way to clear the ground point cloud?
Looking forward to receiving your reply!

MartyG-RealSense, 2022-01-05:
Hi @ju-mingyue  The D435 camera model has a default minimum distance of 0.1 meters / 10 cm.  When objects or surfaces become closer to the camera than that minimum distance then the area of the depth image at that range begins breaking up.

You could use the **Disparity Shift** option in the RealSense ROS wrapper to reduce the minimum distance and enable observed objects / surfaces to get closer to the camera without breaking up.  There is not a command setting for doing this in ROS though.  You would instead have to define Disparity Shift in the RealSense Viewer tool, export a **json** camera configuration file and then import the json file into ROS.  **Doronhi** the RealSense ROS wrapper developer provides advice about this in https://github.com/IntelRealSense/realsense-ros/issues/981

If you are only depth sensing up to a maximum range of 0.5 meters than a Disparity Shift value of '100' should work well.  As Disparity Shift is increased, the minimum distance reduces but the maximum observable depth sensing distance also reduces.  

Whilst '50' would normally provide a good balance between minimum distance and retaining depth that is further away, for very close range it will not matter if most of the background is excluded from the depth image with a higher Disparity Shift.

**********

In regard to the ground, the dilemma is that you need very close-range sensing but do not want the ground.  There are a couple of options to explore.

1.  Use **rtabmap_ros** and set a minimum distance with its **RangeMin** instruction, described in https://github.com/IntelRealSense/realsense-ros/issues/1859#issuecomment-836207835

2.  Use a **RANSAC** algorithm in ROS to identify and extract the floor plane.  The guide in the link below provides advice about using RANSAC ground extraction in ROS.

https://roboticsknowledgebase.com/wiki/sensing/pcl/#3d-object-detection

ju-mingyue, 2022-01-06:
@MartyG-RealSense  Thank you for your reply. I followed your link and added a program to filter the ground. I updated the specific situation to the question above.

MartyG-RealSense, 2022-01-06:
I have encountered cases with an error referring to extrapolating into the **future** before, but not the past.  Researching the issue, it seems that the solution for the 'past' and 'future' errors would be the same: to set **ros::Time::now()** instead of ros::Time(0).  But you have already done that in your script.

[image]

Perhaps ros::Time::now() needs to be in a different place in the script. 

Another approach suggested in the link below is to insert a **catch** for the extrapolation exception after the ros::Time instruction to catch and handle the exception instead of allowing the script to error and exit.

https://answers.ros.org/question/331027/realsense-d435-lookup-would-require-extrapolation-into-the-future/?answer=331028#post-id-331028

ju-mingyue, 2022-01-07:
@MartyG-RealSense  
Hello, I tested it the way you did, but it didn’t work.
I am running on the ros melodic version, and my ideas are as follows:
1. Start the program of the d435 camera, which is the rs_camera.launch file. At the same time, start the file that divides the ground. My TF coordinates are accurate in my opinion. TF_Tree is as follows:
[image]
I have added frame_id in the program:
[image]

At the same time, the ground point cloud is not cleared every time. When I modify the parameters in the figure below, the following error will be prompted when I run it again:
[image]
[image]

There will be some prompts as follows every time you use catkin_make to compile:
[image]

Look forward to your reply！

MartyG-RealSense, 2022-01-07:
This particular question is outside of my ROS knowledge, unfortunately.  I do apologize.

I recall though that you previously had problems with obstacles being cleared when you were using **depthimage-to-laserscan** and resolved it in https://github.com/IntelRealSense/librealsense/issues/9294#issuecomment-879579215 by adding  the obstacle layer of the camera to the obstacle_layer obstacle layer.  Is that relevant to what you are doing now?

ju-mingyue, 2022-01-11:
@MartyG-RealSense 
Hello, thank you for your reply, but my problem is that the point cloud cannot divide the ground, do you have a better way? If you simply use RANSAC for segmentation, the threshold is set very high, so that the lower ground position cannot be seen, so do you have other solutions? Or do you have any other solutions for the robot to use D435 for obstacle avoidance?

ju-mingyue, 2022-01-11:
Regarding the red error above, I have found the reason, which is related to the setting of the L value in the program, and the L value is related to the actual height of the camera.

MartyG-RealSense, 2022-01-11:
There are not many ground / floor segmentation methods I know of that do not use RANSAC in some form, unfortunately.

Intel have a package for ROS called **ros_object_analytics** that includes intelligent object avoidance capability.  The ROS1 version is old and based on ROS Kinetic though, so I do not know how difficult it would be to adapt it for Melodic.

https://github.com/intel/ros_object_analytics

There was also a complete hardware and software blueprint for a ROS mobile robot with avoidance that was originally published in 2017.  It was originally designed for the RealSense R200 and ZR300 stereo camera models that were the direct predecessors of the current 400 Series stereo depth cameras, though the author states that the project can be used with the 400 Series too.

https://www.intel.com/content/www/us/en/developer/articles/technical/build-an-autonomous-mobile-robot-with-the-intel-realsense-camera-ros-and-sawr.html

Another option for ROS object avoidance that officially supports the D435 camera model is **PX4-Avoidance**.

https://github.com/PX4/PX4-Avoidance

closed 2022-01-23 by MartyG-RealSense