realsenseai/librealsense #10129: How to use the L515 camera in YOLO detection at Win10?

lizhiying-2019, 2022-01-05:
I want to use YOLO to detect. How do I connect to the L515 camera in Win10? Please tell me how to do ? I have no idea.

MartyG-RealSense, 2022-01-05:
Hi @lizhiying-2019   A typical YOLO object detection configuration used with RealSense cameras is **yolov4** and **darknet**.

The darknet GitHub at https://github.com/AlexeyAB/darknet provides instructions for setting up yolov4 and darknet on Windows.

https://github.com/IntelRealSense/librealsense/issues/9098 also provides an example of Python scripting for using an L515 camera to access yolov4 and darknet in the librealsense SDK.

lizhiying-2019, 2022-01-10:
I want to detect with yolov5. When connect compute by USB3.0, I set the camera as 0, it's not working, display warning. What should I do? Please tell me. @MartyG-RealSense

MartyG-RealSense, 2022-01-10:
In a yolov4 case at https://github.com/IntelRealSense/librealsense/issues/9874#issuecomment-947631452 it was necessary to set **ZED_CAMERA=0** in the Makefile if a **ZED** camera is not being used and then afterwards define the **/video** channel to use, such as /video0.  

So it may be worth checking the Makefile of yolov5 to see if ZED_CAMERA needs to be set to '0' to allow a non-ZED camera such as RealSense to be accessed. 

A yolov5 case at https://github.com/IntelRealSense/librealsense/issues/9134 is also worth reading.

closed 2022-01-22 by MartyG-RealSense