realsenseai/librealsense #10131: Calibration

likui01, 2022-01-06:
Hi, 
I want to calibrate two L515 sensors, to make 3D reconstructions from different cameras. How I can do that?

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18  |

likui01, 2022-01-06:
@MartyG-RealSense Thank you for the reply.
can you give details on how cameras had their positions calibrated relative to each other using Apriltags? I am using Open3d for 3D reconstruction. But I want to figure out how can I use multiple cameras?

MartyG-RealSense, 2022-01-06:
The YouTube video shows that the solid 3D model resulting from the stitching was being viewed in Open3D but does not provide details of the software being used during the scanning process or how the apriltag boards were used for calibration.

[image]

There was a similar Open3D project in the link below that uses multiple Azure Kinect cameras to calibrate against apriltags and generate a combined point cloud.

https://github.com/stytim/k4a-calibration

MartyG-RealSense, 2022-01-13:
Hi @likui01  Do you require further assistance with this case, please?  Thanks!

likui01, 2022-01-13:
@MartyG-RealSense, thank you for your response. 
I am still unable to calibrate the two L515, it's a work in progress. At this time i don't have an exact question.
I am looking into ways in which i can calibrate using April tags.

MartyG-RealSense, 2022-01-13:
Another commercial 3D software tool that supports the L515 camera model is **DotProduct Dot3D**.  It is available in a Lite version and a Pro version for larger scenes that captures up to 160 million points.

https://www.dotproduct3d.com/dot3dlite.html
https://www.dotproduct3d.com/dot3dpro.html

DotProduct has a tutorial for how two separate scans can be registered together using apriltags.

https://desk.zoho.com/portal/dotproduct/en/kb/articles/dot3d-tutorial-registering-multiple-dp-scans-with-apriltags-in-dot3d-pro

closed 2022-01-26 by MartyG-RealSense