realsenseai/librealsense #10132: The depth and RGB images can't be aligned when connect the realsense camera by net_device class

JunSun-CSH, 2022-01-06:
Required Info |  
-- | --
Camera Model | D415
Firmware Version | 	5.13.0.50
Operating System & Version | Linux (Ubuntu 18.04)
Kernel Version (Linux Only) |  
Platform | Raspberry Pi 4B
SDK Version | v2.50
Language | C/C++
Segment |  

**Problem Description:**
- Use following codes to start the camera:
<code>   
rs2::align alignTo(rs2_stream::RS2_STREAM_COLOR);
rs2::config cfg;
cfg.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 1280, 720, rs2_format::RS2_FORMAT_Z16, 6);
cfg.enable_stream(rs2_stream::RS2_STREAM_COLOR, 1280, 720, rs2_format::RS2_FORMAT_BGR8, 6);
rs2::context ctx;
rs2::net_device device("192.168.99.101");
device.add_to(ctx);
pipeline = new rs2::pipeline(ctx);
pipeline->start(cfg);
auto frames = pipeline->wait_for_frames(5000);
alignTo.process(frames);
</code>

- If replace net_device codes by traditional USB device codes, the images can be aligned.
- In realsense viewer can see same issue. In 3D view, images get aligned with USB connection and not aligned with network connection.
- If see this issue is same with this: network device align not work #8966. But in v2.50, this issue is still alive.

MartyG-RealSense, 2022-01-06:
Hi @JunSun-CSH  I see that you have chosen FPS speeds that are appropriate for the limitations of a RealSense networking connection, as described in the **Software Limitations** section of the networking white-paper guide.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

In a networking case at https://github.com/IntelRealSense/librealsense/issues/8423#issuecomment-795557815 where there was a problem in the cfg definition, it was found that using the **RGB8** format for color worked.

 May I also confirm that when you built librealsense from source code in Linux with CMake on the Raspberry Pi 4, you have included the term **-DBUILD_NETWORK_DEVICE=ON** in the CMake build instruction.

Are you using a two-computer arrangement like the one in the white-paper (a Pi 4 with a camera attached as the remote computer and a conventional PC as the central host computer) or are you using a single computer please, like in https://github.com/IntelRealSense/librealsense/issues/10080

JunSun-CSH, 2022-01-07:
> Hi @JunSun-CSH I see that you have chosen FPS speeds that are appropriate for the limitations of a RealSense networking connection, as described in the **Software Limitations** section of the networking white-paper guide.
> 
> https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations
> 
> In a networking case at [#8423 (comment)](https://github.com/IntelRealSense/librealsense/issues/8423#issuecomment-795557815) where there was a problem in the cfg definition, it was found that using the **RGB8** format for color worked.
> 
> May I also confirm that if you built librealsense from source code in Linux with CMake on the Raspberry Pi 4, you have included the term **-DBUILD_NETWORK_DEVICE=ON** in the CMake build instruction.
> 
> Are you using a two-computer arrangement like the one in the white-paper (a Pi 4 with a camera attached as the remote computer and a conventional PC as the central host computer) or are you using a single computer please, like in #10080

I built a rs-server and run rs-server on a raspberry pi 4B board with Debian 11 to make the D455 work as a net device.
In other Win10 PC, wrote codes to connect D455. I was successfully connect to D455 and capture the RGB and depth images. But, the align function isn't work normally. From the following pictures, you can see the RGB and depth images were misaligned. If directly connect D455 to the Win10 PC, I can see the align function work as expected.

RGB and depth alignment effect in my codes:
[image]

RGB and depth align effect in realsense viewer:
[image]

JunSun-CSH, 2022-01-07:
In my current investigation, found the issue caused by wrong intrinsics, extrinsics from the profile and wrong depth_scale of the depth_frame instance. These value are provided by the net_device from rs_server. 

-  The values in intrinsics and extrinsics get truncated.
-  The depth_scale is an extreme wrong value.
-  After placing the hardcode correct intrinsics, extrinsics and depth_scale values in the align function, the depth and RGB images can be aligned normally. This way can be a temporary work around. Could we fix this issue in the rs_server codes in next release?
[image]

**Normal Values:**
```
-		depthIntrinsics	{width=1280 height=720 ppx=630.740906 ...}	rs2_intrinsics
		width	1280	int
		height	720	int
		ppx	630.740906	float
		ppy	362.714935	float
		fx	885.439941	float
		fy	885.439941	float
		model	RS2_DISTORTION_BROWN_CONRADY (4)	rs2_distortion
-		coeffs	0x0000009c68efef94 {0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000}	float[5]
		[0]	0.00000000	float
		[1]	0.00000000	float
		[2]	0.00000000	float
		[3]	0.00000000	float
		[4]	0.00000000	float
-		rgbIntrinsics	{width=1280 height=720 ppx=645.108337 ...}	rs2_intrinsics
		width	1280	int
		height	720	int
		ppx	645.108337	float
		ppy	356.728638	float
		fx	921.190857	float
		fy	919.487183	float
		model	RS2_DISTORTION_INVERSE_BROWN_CONRADY (2)	rs2_distortion
-		coeffs	0x0000009c68efef44 {0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000}	float[5]
		[0]	0.00000000	float
		[1]	0.00000000	float
		[2]	0.00000000	float
		[3]	0.00000000	float
		[4]	0.00000000	float

		_depth_scale	0.00100000005	float

-		depthToRgb2	{rotation=0x000000dd0d1eef08 {1.00000000, 0.00000000, 0.00000000, 0.00000000, 1.00000000, 0.00000000, ...} ...}	rs2_extrinsics
-		rotation	0x000000dd0d1eef08 {1.00000000, 0.00000000, 0.00000000, 0.00000000, 1.00000000, 0.00000000, 0.00000000, ...}	float[9]
		[0]	1.00000000	float
		[1]	0.00000000	float
		[2]	0.00000000	float
		[3]	0.00000000	float
		[4]	1.00000000	float
		[5]	0.00000000	float
		[6]	0.00000000	float
		[7]	0.00000000	float
		[8]	1.00000000	float
-		translation	0x000000dd0d1eef2c {-0.00000000, -0.00000000, -0.00000000}	float[3]
		[0]	-0.00000000	float
		[1]	-0.00000000	float
		[2]	-0.00000000	float
```

**Wrong Values:**
```
-		depthIntrinsics	{width=1280 height=720 ppx=630.000000 ...}	rs2_intrinsics
		width	1280	int
		height	720	int
		ppx	630.000000	float
		ppy	362.000000	float
		fx	885.000000	float
		fy	885.000000	float
		model	RS2_DISTORTION_BROWN_CONRADY (4)	rs2_distortion
-		coeffs	0x000000c0b9ddf5a4 {0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000}	float[5]
		[0]	0.00000000	float
		[1]	0.00000000	float
		[2]	0.00000000	float
		[3]	0.00000000	float
		[4]	0.00000000	float
-		rgbIntrinsics	{width=1280 height=720 ppx=645.000000 ...}	rs2_intrinsics
		width	1280	int
		height	720	int
		ppx	645.000000	float
		ppy	356.000000	float
		fx	921.000000	float
		fy	919.000000	float
		model	RS2_DISTORTION_INVERSE_BROWN_CONRADY (2)	rs2_distortion
-		coeffs	0x000000c0b9ddf554 {0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000}	float[5]
		[0]	0.00000000	float
		[1]	0.00000000	float
		[2]	0.00000000	float
		[3]	0.00000000	float
		[4]	0.00000000	float

		_depth_scale	-431602080.	float

-		depthToRgb2	{rotation=0x000000414deff378 {0.999985516, -0.000769818376, 0.00532204285, 0.000799249217, 0.999984384, ...} ...}	rs2_extrinsics
-		rotation	0x000000414deff378 {0.999985516, -0.000769818376, 0.00532204285, 0.000799249217, 0.999984384, -0.00553008122, ...}	float[9]
		[0]	0.999985516	float
		[1]	-0.000769818376	float
		[2]	0.00532204285	float
		[3]	0.000799249217	float
		[4]	0.999984384	float
		[5]	-0.00553008122	float
		[6]	-0.00531770289	float
		[7]	0.00553425495	float
		[8]	0.999970555	float
-		translation	0x000000414deff39c {0.0149887651, 2.59664448e-05, 0.000211531689}	float[3]
		[0]	0.0149887651	float
		[1]	2.59664448e-05	float
		[2]	0.000211531689	float

```

MartyG-RealSense, 2022-01-07:
It's great to hear that you achieved a workaround!  

Yes, your experience sounds similar to the one reported in https://github.com/IntelRealSense/librealsense/issues/8966#issuecomment-833173303 so I will highlight the issue to Intel.

MartyG-RealSense, 2022-01-07:
Intel provided feedback to me.   An ethernet side branch of the SDK will be released soon, so they recommend waiting to see whether that fixes the alignment issue.  A time schedule for release of the side branch is not currently available.

JunSun-CSH, 2022-01-10:
> Intel provided feedback to me. An ethernet side branch of the SDK will be released soon, so they recommend waiting to see whether that fixes the alignment issue. A time schedule for release of the side branch is not currently available.

Thanks for your feedback.

JunSun-CSH, 2022-01-11:
Made a temporary fixing for these issues. The fixing codes contains in the attached git patch zip file.

Fix wrong intrinsics, extrinsics and depth_scale are get from profile and depth_frame when connect to net_device.

Root Cause:
1. Wrong intrinsics values is caused by the third-party component liveMidia doesn't support float type. Before fixing, the float data value is truncated as a int data value.
2. Wrong extrinsics values is caused by using wrong to_key in ip_device::init_device_data for caching the extrinsics values.
3. Wrong depth_scale value is caused by that the frame meta data values(include depth_units) don't be cached due to the current connecting device is a net device not a UVC device.

Solution:
1. Add float type support in MediaSession class and use higher precision float string value.
2. Correct the wrong to_key  in ip_device::init_device_data and use higher preciosn float string value.
3. Add a set_units function for depth_frame. We can cache the depth sensor's depth unit to depth frame by the following codes:
```
auto frames = pipeline->wait_for_frames(5000);
auto firstDepth = frames.get_depth_frame();
firstDepth.set_units(units);
```
[0001-Fix-wrong-intrinsics-extrinsics-and-depth_scale-are-.zip](https://github.com/IntelRealSense/librealsense/files/7844368/0001-Fix-wrong-intrinsics-extrinsics-and-depth_scale-are-.zip)

MartyG-RealSense, 2022-01-11:
Thanks so much @JunSun-CSH for sharing your workaround solution with the RealSense community!

closed 2022-01-23 by MartyG-RealSense