realsenseai/librealsense #10133: D435 Camera, The edge of object problem

GHCK57, 2022-01-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) 
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi,

While working with Realsense D435 camera I faced with a problem. When the camera is close to object approximately 35-40 cm, depth frame looks good. But when I put the camera more farther approximately 70cm. The edges of object  are deformed. 

I use the "**opencv_pointcloud_viewer.py** " shared by Realsense to get the point cloud data. 

That is what I want to measure. There is a surface and there is a object on this surface. I am just looking the edges.  

[image]

This photo shows the .ply file taken from camera 70 cm far away

[image]

And also this one is taken from 35 cm far away 

[image]

As you can see, there is a clear difference in the edges of the object. 
I have to work 70cm far away. Why does this difference occur? How can I deal with this problem. I just want to flat edge.

I hope you understood me. I wait for your helping. Thanks a lot.

MartyG-RealSense, 2022-01-06:
hI @GHCK57  Whist the box looks like it has more edges detected at 35 cm, the blue edges actually likely represent empty areas of missing depth data rather than depth detail of the box edges.  The same is true for the increased numbers of blue holes on the table surface.

It also looks like at 70 cm range, the camera is having difficulty distinguishing the plain white edges of the box from the similar plain white texture of the table.

You may achieve a better quality image if you ensure that the **IR emitter** is enabled in your Python script.  When the emitter is enabled, it casts a semi-random pattern of dots onto surfaces in the scene that can be used as a 'texture' source for analyzing low-textured or no-textured surfaces (tables, doors, walls, etc) for depth detail.

You could try integrating the Python code below into your application, which enables the IR emitter with the **set_option(rs.option.emitter_enabled, 1)** instruction.

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
pipeline_profile = pipeline.start(config)
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
if depth_sensor.supports(rs.option.emitter_enabled):
depth_sensor.set_option(rs.option.emitter_enabled, 1)
```

GHCK57, 2022-01-06:
Hi MartyG,

Thanks for your reply. I insterted the these lines to **opencv_pointcloud_viewe.py**  file after this code **pipeline.start(config)** 

depth_sensor = device.query_sensors()[0]

if depth_sensor.supports(rs.option.emitter_enabled):
    depth_sensor.set_option(rs.option.emitter_enabled, 1)

Also I used a different color box.( The Realsense camera's orijinal box).  But still when I get the point cloud data from different distances, the problem is going on. 

This photo how camera is looking the object. 

[image]

This is how **opencv_pointcloud_viewe.py** shows the object.

[image]

-- Camera far away 70cm from the object--

[image]

-- Camera far away 35cm from the object--
[image]

Still, 35 cm is better than 70cm. What is the problem ?

GHCK57, 2022-01-06:
Also I used the function to enable infrared stream but it didn't change.

config.enable_stream(rs.stream.infrared, rs.format.y8, 30)

MartyG-RealSense, 2022-01-06:
It appears that enabling the IR emitter has at least closed up most of the holes in the table surrounding the box, so that is an improvement.

I am reminded of another Python box scanning case at https://github.com/IntelRealSense/librealsense/issues/6857#issuecomment-660532275 where the edge of the box seemed to be pointing diagonally like your 70 cm image above instead of 90 degrees straight downwards.  Their eventual solution - shared in https://github.com/IntelRealSense/librealsense/issues/6857#issuecomment-660540868 - was to create a custom **json** camera configuration file and load it into their project to adjust the camera settings to produce a more accurate box pointcloud.

The increased bumpiness of the cloud at 70 cm compared to 35 cm is a known phenomenon referred to as 'waviness', where the surface becomes increasingly wavy as the camera moves further away from the observed surface.  A RealSense user in https://github.com/IntelRealSense/librealsense/issues/1375#issuecomment-373561886 found that using a custom json could reduce some of the waviness.  

A RealSense temam member also suggested in https://github.com/IntelRealSense/librealsense/issues/1375#issuecomment-373726950 that applying post-processing filtering such as edge-preserving and temporal avergaing could help.  These concepts are covered in Intel's post-processing guide in the link below.  Performing a page 'find' operation for the terms **edge-preserving** and **temporal averaging** can help to find relevant sections of the guide quickly.

https://dev.intelrealsense.com/docs/depth-post-processing

GHCK57, 2022-01-06:
Hi @MartyG-RealSense ,

I saw and read the topic #6857 and #1375 before and I tried "ShortrangePreset.json" but it didn't solve my problem.  In addition to these topic and read this white paper you shared with me  https://dev.intelrealsense.com/docs/depth-post-processing . The photos shared in here before I shared them  I exported .ply and viewed on meshlab. I think there is no viewer problem. 

I tried a lot of configuration on Intel RealSense Viewer v2.48 GUI, What I did whatever, I didn't get precision point cloud in comparison with the position far away 35 cm from object.

This is a different configuration. But camera trying to match edge of object and surface. ( Camera mounted 70cm far away from object)

[image]

This is depth frame

[image]

And this is color frame 

[image]

MartyG-RealSense, 2022-01-06:
Applying a post-processing filter that has a **hole-filling** function will likely help to close up the black area surrounding the box.  You can test this in the RealSense Viewer by going to the **Post-Processing** category of the options side-panel and left-clicking on the red icon beside the **Hole-Filling** filter (which indicates a Disabled status) to turn it to blue (enabled).

[image]

If you find that hole-filling works well for you, you could implement hole-filling in your Python application by applying the **Spatial** post-processing filter.

[image]

GHCK57, 2022-01-06:
Thanks for your reply. But I don't need hole filling filter. It is not suitable for my application.  I reaaly dont know what is the problem. The depth frame is good for camera far away 35 cm but while the distance is 70cm, it measure wrong. 

I just want this depth frame when I put the camera far away from object

[image]

MartyG-RealSense, 2022-01-06:
How does the box look in the RealSense Viewer if you change the **Depth Units** option from its default of 0.001 to **0.0001**.  You can left-click on the pencil icon beside the option to type the value in on the keyboard instead of using the adjustment slider.

[image]

GHCK57, 2022-01-07:
Hi @MartyG-RealSense 

I tried it too before. 

For Depth Units --> 0.001

[image]

For Depth Units --> 0.0001

[image]

MartyG-RealSense, 2022-01-07:
If your goal is to measure the box then the SDK Python example program **box_dimensioner_multicam** may meet your needs.  It works with a single camera, despite the program's name.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

In that program, the box is placed upon a print-out of a chessboard to calibrate the camera position to the board and then a bounding-box is placed around the box on the chessboard on an RGB image and volume measurements for the box that are calculated from the depth data are displayed.

[image]

GHCK57, 2022-01-07:
Hi @MartyG-RealSense 

I worked with this code. I need 3D constraction while calculating the dimensions of the box. That's why I need a good depth frame. Due to the point cloud data taken from 70cm is bad, everything going false. As you see from photos that I shared before, The camera seems the edge of box and surface like  as same. This problem makes calculating the volume impossible. Please I need more advice about it . Is it caused for IR pattern emitting from camera or caused from something else ?

MartyG-RealSense, 2022-01-07:
If your project is able to take measurements manually instead of them needing to be automatic then you could try using the RealSense Viewer's **Measure** tool in its 3D pointcloud mode, which enables you to drag out a line between two points on the pointcloud image and receive a measurement of the real-world distance between those two points.

[image]

If you prefer to continue with the current method then I would recommend testing with the camera at a straight-ahead or straight-down angle (0 or 90 degrees) instead of the camera being tilted, as indicated by the tilted angle of the table.  The angle of the camera may be distorting the point cloud, so having the camera at a straight angle will confirm whether or not the camera angle is the cause.

GHCK57, 2022-01-07:
I did what you said. I put the camera 90 degree above of the object. The camera aproximately 60cm far from the object. The configurations are didn't changed. Everything is same. It looks like this. Still the camera connected the surface of table and edge of object.

[image]

When I  put it the camera 40cm above of the box. Everythings are fine. That is what I want. But I have to work from  70-75 cm distance 

[image]

MartyG-RealSense, 2022-01-07:
I attempted to replicate the test in the Viewer with a pack of drinks bottles but did not experience the edge loss at 70 cm.

**35 cm**

[image]

**70 cm**

[image]

Could you check whether disabling the two **GLSL** options in the Viewer's settings before taking the scan improves your edge detection, please?  This can be done with the instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

Also, try completely disabling all post-processing filters that the Viewer applies by default by left-clicking on the blue icon beside **Post-Processing** to turn the icon to red and turning off all filters.

[image]

GHCK57, 2022-01-07:
Hi again @MartyG-RealSense 

First of all your data taken from 70cm is perfect. 

I tried the disable GLSL options. It didn't work. But I explored something. I will try to explaint step by step because what have been done I dont know. I need your help to figure out the solution. 

First I want to share the photo that after GLSL options were disabled. It seems nothing changed.

[image]

After while looking the #8110 topic I saw your comment. https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754653180  

I checked my **tools** under ProgramFiles(x86)--> Intel RealSense SDK 2.0 . I tried to run exe files. After the start the **rs-measure.exe** I saw the difference. The sides of the box are starting to look good. But I didn't change anything. All parameters are same. Just I run the rs-measure.exe . After that It is a picture taken from the camera. 

There are still some noise but better than old one. 
[image]

What is changed it ?

MartyG-RealSense, 2022-01-07:
The main difference that comes to mind about rs-measure is that instead of performing depth to color alignment, it performs the much rarer **color to depth** alignment.  On the D435 and D435i models, which have a smaller field of view (FOV) size on the color sensor than on the depth sensor, the overall image is therefore stretched out as the color FOV resizes to match the larger depth FOV size.

GHCK57, 2022-01-07:
Okay. If I hardware reset the camera using Realsense Viewer, everything will be back to old. What should I do next to get the correct depth frame. 

[image]

MartyG-RealSense, 2022-01-07:
Is there any positive change in the Viewer image if you maximize the **Laser Power** option to '360' instead of the default '150'?  Increasing laser power makes the IR dot pattern more visible to the camera and can increase the amount of depth detail on an image.

[image]

GHCK57, 2022-01-10:
Hi @MartyG-RealSense 

Sorry for my late reply.  I generally set the laser power option to 330. But as I see there is no difference can see with eye.

MartyG-RealSense, 2022-01-10:
I looked at the code of **rs-measure** to see what might account for the image improving when the program is run but reverting to the worse image after a hardware reset.  When rs-measure is run, it loads and applies the **High Accuracy** camera configuration preset.  When a camera reset is performed, the camera would return to its default configuration.

Ypu could therefore try adding a preset load instruction for 'High Accuracy' to your **opencv_pointcloud_viewer.py** script using Python code in https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634 or instead selecting the preset from the presets drop-down menu in the Viewer.

[image]

GHCK57, 2022-01-10:
It looks pretty good. After a hardware reset, before open the stream, I choose the **High Accuracy** camera configuration preset. The box look like this.   

[image]

There is almost no edge connection. But if you look carefully, you can see the  try to connect edge of object and surface of table. 

[image]

MartyG-RealSense, 2022-01-10:
How does it look if you change "High Accuracy" to **"Medium Density"**

The Medium Preset provides a good balance between accuracy and the amount of detail on the image.

GHCK57, 2022-01-10:
Medium Density Preset

[image]

MartyG-RealSense, 2022-01-10:
Is the above image from 70-75 cm distance?

GHCK57, 2022-01-11:
Yes it is.

GHCK57, 2022-01-11:
Hi @MartyG-RealSense 

This is coor frame taken by camera. 

[image]

This is point cloud data. 

[image]
 
[image]

Why the shaped edge seems like there was a radius there ?

GHCK57, 2022-01-11:
Okey, I will use the 848x480 depth and colur resolution. What do you think about this one ? Is this same with **rs2_deproject_pixel_to_ point** function ? 

```

                 depth = frame.get_distance(x,y)
                 X = (x - self.color_intrinsics.ppx)/self.color_intrinsics.fx *depth
                 Y = (y - self.color_intrinsics.ppy)/self.color_intrinsics.fy *depth

```

And, I will use to get point cloud data.

```

                points = pc.calculate(depth_frame)                
                vertices = np.asarray(points.get_vertices(2)).reshape(-1, 3)  #XYZ
                texture =  np.asarray(points.get_texture_coordinates()).reshape(-1, 2) #UV

```

What about if I want to get the XYZ coordinates from the specified pixel coordinate ? Because I need it to calibrate the system. There is a "red circle" on the table. The opencv library detects the red circle from color frame of camera and it gives me the center of pixel coordinates of that red circle. After I get the pixel coordinates to calculate the XYZ world coordinate I was able to used **rs2_deproject_pixel_to_ point** function. If **rs2_deproject_pixel_to_ point** has bad accuracy how to I get the XYZ world coordinates from the pixel ?

MartyG-RealSense, 2022-01-11:
Instead of using depth to color alignment on the entire image, you can retrieve the XYZ of a single specified coordinate without using alignment by using an SDK instruction called **rs2_project_color_pixel_to_depth_pixel**. 
 https://github.com/IntelRealSense/librealsense/issues/5603 is a good reference for using this instruction in Python.  The RealSense user in that case shares their final successful script at https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

GHCK57, 2022-01-12:
Okey thanks a lot for information.  

Do these values following the mouse ​​in Realsense Viewer show us 3D XYZ points? 

Is X= 0.091  Y = 0.014  Z = 0.738  ??

[image]

MartyG-RealSense, 2022-01-12:
Yes it is showing the 3D real-world XYZ, with Z represented by real-world distance of the coordinate from the camera in meters.

GHCK57, 2022-01-12:
I have no idea about X and Y but  I think Z value is wrong.  I measured it. Viewer shows the center of red point as 0.76 meter but there is big difference with between measured by me. 

[image]
[image]
[image]

MartyG-RealSense, 2022-01-12:
On the cased 400 Series cameras, depth measurements are taken from the front glass of the camera.  So the real-world tape measure distance from the camera to the observed point on the table would be approximately 72 cm / 0.72 m.

There is a closer match of 73.8 cm / 0.738 m on the image that you posted above. 

[image]

The camera measurements will have a certain amount of error in them, as error starts at around zero at the camera lenses and increases linearly over distance (a phenomenon called **RMS error)**.  My understanding is that at 1 meter distance, a D435 would have an error factor of 2.5 mm to 5 mm.

Also, unless the IR emitter is enabled to project a dot pattern onto the scene then a plain untextured or low-texture surface such as the table would be more difficult for the camera to analyze for depth detail.  

The **High Accuracy** preset can help to filter out obviously incorrect depth values.  

The 400 Series stereo depth cameras do not have a 'confidence map' like the earlier SR300 model did, but you can achieve a similar effect using the **Second Peak Threshold** control, which in the Viewer can be found in the options side-panel under **Advanced Controls > Depth Control > DS Second Peak Threshold**.  If the slider is reduced towards zero from its default value of '325' then you can observe holes in the image disappearing, whilst holes increase if the slider is increased above the default.

GHCK57, 2022-01-12:
Thanks for advices.  The camera to the observed point on the table would be approximately 72 cm / 0.72 m. But the camera tells us it is 0.76m not 0.73m. We can ignore the photo that posted by me before. It was a different point on table. I think it is huge different betwwen 0.76 and 0.72.  Do the camera need the calibrated it  ? 
 
[image]

MartyG-RealSense, 2022-01-12:
There is certainly no harm in calibrating the camera if you have doubts about the accuracy of the depth measurement values.  It is worth bearing in mind that if **On-Chip** calibration is used then this can improve the quality of the image but a separate calibration option called **Tare Calibration** is used to improve depth accuracy.

[image]

GHCK57, 2022-01-12:
I was though the On-Chip Calibration is calibrate the depth and color . That is good news for me . It is good now.

[image]

MartyG-RealSense, 2022-01-12:
That's great news, @GHCK57   :)

GHCK57, 2022-01-12:
Hi again @MartyG-RealSense 

I think (I Hope) my problems are over. But I want to ask you a few questions. 

How can I inject RGB color to XYZ point cloud data.  

self.vertices = np.asarray(points.get_vertices(2)).reshape(-1, 3)  #XYZ
self.texture =  np.asarray(points.get_texture_coordinates(2)).reshape(-1, 2) #UV

I used to use color frame to get color when color and depth frame alignment but now there is no alignment. Therefore I can not use color frame to get RGB colors. What should I do to find true color corresponding to that own pixel.

MartyG-RealSense, 2022-01-12:
At https://github.com/IntelRealSense/librealsense/issues/1890 there is a Python case where someone was trying to map a **depth pixel to a color pixel** instead of mapping a color pixel to depth pixel using rs2_project_color_pixel_to_depth_pixel

GHCK57, 2022-01-14:
Hi again @MartyG-RealSense 

I didn't solve the problem that posted in this https://github.com/IntelRealSense/librealsense/issues/10133#issuecomment-1008899208 

At the points where the table surface meets the box, points are seen in the lower right and left corners of the box.
[image]

MartyG-RealSense, 2022-01-14:
This problem would seem to have similarities to the case https://github.com/IntelRealSense/librealsense/issues/8306 which discusses options for tidying pointcloud data that you have already tried, such as changing the depth unit scale and the Second Peak Threshold.

The RealSense user in that case also achieved some improvement by using an **Advanced Mode** control in the RealSense Viewer called **Rsm** to reduce noise such as the grey areas.

[image]

The setting that they changed was **Remove Threshold**, altering it from the default of '64' to a higher value of '93'.

GHCK57, 2022-01-17:
Hi @MartyG-RealSense 

Sorry for my late reply. I enable the advanced mode in my python project with below code. 

where config_D435 is configuration to enable streams , pipe435 is  pipeline (rs.pipeline()) 
       
 ```
        pipeline_profile = pipe_D435.start(config_D435)
        device = pipeline_profile.get_device()
        advanced_mode = rs.rs400_advanced_mode(device)
        advanced_mode.load_json(HighResHighAccuracyPreset.json)
```

Also I use "HighResHighAccuracyPreset.json" 

> {
>     "param-disableraucolor": 0,
>     "param-disablesadcolor": 0,
>     "param-disablesadnormalize": 0,
>     "param-disablesloleftcolor": 0,
>     "param-disableslorightcolor": 1,
>     "param-lambdaad": 751,
>     "param-lambdacensus": 6,
>     "param-leftrightthreshold": 10,
>     "param-maxscorethreshb": 2893,
>     "param-medianthreshold": 796,
>     "param-minscorethresha": 4,
>     "param-neighborthresh": 108,
>     "param-raumine": 6,
>     "param-rauminn": 3,
>     "param-rauminnssum": 7,
>     "param-raumins": 2,
>     "param-rauminw": 2,
>     "param-rauminwesum": 12,
>     "param-regioncolorthresholdb": 0.786380709066072,
>     "param-regioncolorthresholdg": 0.5664810046339115,
>     "param-regioncolorthresholdr": 0.9857413557742051,
>     "param-regionshrinku": 3,
>     "param-regionshrinkv": 0,
>     "param-regionspatialthresholdu": 7,
>     "param-regionspatialthresholdv": 3,
>     "param-robbinsmonrodecrement": 25,
>     "param-robbinsmonroincrement": 2,
>     "param-rsmdiffthreshold": 1.6605679586483368,
>     "param-rsmrauslodiffthreshold": 0.7269914923801174,
>     "param-rsmremovethreshold": 0.8150280066589434,
>     "param-scanlineedgetaub": 13,
>     "param-scanlineedgetaug": 15,
>     "param-scanlineedgetaur": 30,
>     "param-scanlinep1": 155,
>     "param-scanlinep1onediscon": 160,
>     "param-scanlinep1twodiscon": 59,
>     "param-scanlinep2": 190,
>     "param-scanlinep2onediscon": 507,
>     "param-scanlinep2twodiscon": 493,
>     "param-secondpeakdelta": 647,
>     "param-texturecountthresh": 0,
>     "param-texturedifferencethresh": 1722,
>     "param-usersm": 1
> }

I implemented the code you posted.  But still the problem is going on. Advanced mode is enable . The outpu is 

> Found device that supports advanced mode: Intel RealSense D435
> Advanced mode is enabled
> RAU Thresholds Control: 
>  rauDiffThresholdRed: 1007, rauDiffThresholdGreen: 578, rauDiffThresholdBlue: 802
> After Setting new value, RAU Thresholds Control: 
>  rauDiffThresholdRed: 200, rauDiffThresholdGreen: 500, rauDiffThresholdBlue: 1000

MartyG-RealSense, 2022-01-17:
I ran extensive further tests with a D435i and a similar cardboard box.  I was able to replicate the black areas shown in your image above.  They seemed to be strongest when the camera was viewing the box from a horizontal-diagonal angle and receded when the camera was placed parallel to the box off to the side of it, looking straight ahead.

[image]

[image]

GHCK57, 2022-01-17:
I don't suffer from black areas. My problem is different but I dont know how can I explain it. There is depth degradation on the contours selected on photos above.

The blue lines are different surface.
The red circle is different surface.
The green line shows the bottom boundary of box.
The yellow lines are degraded depth data.

[image]

This is color frame. 

[image]

GHCK57, 2022-01-17:
The color frame 

[image]

Bottom view of the box. (point cloud data)

[image]

Front view of the box. ( point cloud data) 

[image]

Different view

[image]

[image]

As you see, there are many points colored white. These points are belong to surface of table they are not belong to box. I hope I explained well.

MartyG-RealSense, 2022-01-17:
The kind of point cloud edge trimming required may be beyond the SDK's default point cloud capabilities and require the use of specialist point cloud libraries such as **PCL** or **Open3D** to trim off unwanted detail with filters.  For example, an outlier removal filter (which can remove unconnected 'islands' of points that are not joined to the main image) could likely deal with the corner blocks in the image above.  The RealSense SDK has compatibility wrappers for PCL and Open3D to enable their functions to be accessed from a RealSense project.

GHCK57, 2022-01-18:
I already apply different kind of filters such as outline remover, KNN ,etc. It is not about this. I think there is a problem but I actually dont know what it is.  Because if you look picture in this comment https://github.com/IntelRealSense/librealsense/issues/10133#issuecomment-1014359703 . There is a depth distortion. Is it caused due to that reason.

GHCK57, 2022-01-18:
I will share one more example. I put the 2 screwdrivers on side of the box. 

[image]

If we look at the depth frame it looks like this. 

[image]

[image]

Something seems to have gone wrong.

GHCK57, 2022-01-18:
When the curved wall has removed , the depth frame is 

[image]

[image]

MartyG-RealSense, 2022-01-18:
The above image with the screwdrivers either side of the box reminds me of when I tested creating a pointcloud of thin toothpicks in https://github.com/IntelRealSense/librealsense/issues/8228 at a similar distance range.  The toothpicks had the same kind of black area around them at first.  I was able to remove the black areas by using a depth unit scale of '0.0001' like you tried in https://github.com/IntelRealSense/librealsense/issues/10133#issuecomment-1007155390

The speckled effect on all the surfaces in the above image makes me think that the IR dot pattern might be leaking through onto the RGB image, something that is not meant to happen.  When it does happen, reducing the **Laser Power** value can reduce the visibility of the dots.

Gercek57, 2022-01-18:
Hi @MartyG-RealSense It is @GHCK57 

What is the meaning of leakage IR dot pattern onto the RGB image ? I will check the changing the Laser Power to visibility of the dots tomorrow. In your comment #8228, the depth measurement on of the toothpick is very successful. Why can't I get a healthy depth data?

MartyG-RealSense, 2022-01-18:
An infrared image is covered in a projection of semi-random dots from the camera's IR Emitter component, like your IR images above.  These dots may appear on the depth image as black dots, an undesirable noise effect called **laser speckle**.  They should not be appearing on the RGB image though, and when they do it can be described as a 'leaking through'.  I am not aware of negative effects to the RGB image of such a leak though, other than it looking wrong in terms of the real-life scene not having dots on it.

In regard to the toothpick image: a characteristic of the scene that the camera was in is that it does not have strong lighting.  I also put a plain flat background card behind the toothpick and set the depth scale to 0.0001.  The depth scale change by far made the largest impact on removing unwanted black and grey areas from the image.

I also experimented with setting the **Disparity Shift** option (under Advanced Mode > Depth Table in the Viewer) to a value of '50' instead of the default '0' to reduce the camera's minimum depth sensing distance below the default 0.1 meters of D435 / D435i.

GHCK57, 2022-01-19:
I tested the changing of temperature. It take approximately 30 minutes to reach the 44 degree temperature. While testing just color and depth frame was open and Laser power was 150mW. There was no error occurred.

[image]

The second test ,s changing the Laser Power. 

The laser Power is 0:

[image]

The laser power is 50:

[image]

The laser Power is 150:

[image]

The laser Power is 300:

[image]

MartyG-RealSense, 2022-01-19:
It can be observed that as Laser Power increases, the number of black dots increases, as shown in your '300' image.  In those particular images, the speckle effect seems to be mostly in the area away from the box rather than negatively affecting the box itself even though the dot pattern is being cast onto the box.

[image]

This is an expected effect, as the dot visibility can differ depending on the material surface and the time of day.  It has been demonstrated that earlier in the day the pattern may be invisible on one type of material in the scene and strongly visible on another type of material, whilst in late afternoon the visibility pattern may reverse, with the formerly invisible area now visible and the strongly visible area becoming invisible.

GHCK57, 2022-01-19:
Then, What sholuld I do ? What is your suggestions ?  There is huge black hole  apperings behind of the box. When the lights turn off it does not appear. Also there are lots of problem like in https://github.com/IntelRealSense/librealsense/issues/10133#issuecomment-1015261451 . The camera does not see very well the box, how can I measure of its volume ?

MartyG-RealSense, 2022-01-19:
I recommend determining the lighting level and settings in which the dots are most visible on the box's surface.  In your particular scene, that would seem to be Laser Power = 300 and with the lights turned off.  The camera can then use the dots on the surface of the box as a 'texture source' to analyze for depth detail.  This approach works well for plain surfaces with a low amount of texture detail on them like the cardboard.

As mentioned earlier in https://github.com/IntelRealSense/librealsense/issues/10133#issuecomment-1007310195 the RealSense Viewer allows manual selection of multiple points on a pointcloud to calculate the area.

The link below suggests an approach for using PCL to calculate volume.

https://stackoverflow.com/questions/55629892/2019-point-cloud-volume-estimation-with-c/55681149#55681149

Another approach would be to use a RealSense-powered commercial box measurement solution such as the ones in the links below.

https://www.mobileworxs.com/volume-dimensioning-3d-camera-system/
https://www.lips-hci.com/lipsmetric

GHCK57, 2022-01-19:
In summary, the camera can be affected by the light source, even if the light source is soft. It is necessary to store the camera so that it is not affected by external sources.

But despite everything, I couldn't get the performance I wanted from the camera. I still haven't found the solutions to the questions I wrote earlier. For example, as in comment (https://github.com/IntelRealSense/librealsense/issues/10133#issuecomment-1014584236) and https://github.com/IntelRealSense/librealsense/issues/10133#issuecomment-1013147810, I could not find why the edges of the box were trying to connect to the table surface.

MartyG-RealSense, 2022-01-19:
I cannot think of a way to deal with the table surface other than to position the camera at a top-down angle at a fixed height and set the **threshold filter** post-processing filter with a Minimum value where depth data at the table height and below is excluded from the depth image.  Even if larger boxes are used, if the table top is a fixed distance from the camera then the table under the base of the box should always be removed whilst the box remains on the image.

GHCK57, 2022-01-19:
For this job. The D415 cameras is better than D435, isn't it ?

MartyG-RealSense, 2022-01-19:
I did consistently obtain better box images with the D415 model in my own tests with a similar sized cardboard box.  The D415, though it has a smaller field of view, has around 2x better image quality than D435 and 2x less RMS error (noise over distance).  It also has an optimal depth accuracy resolution of 1280x720, compared to 848x480 on D435 / D435i.

I cannot guarantee that a D415 will resolve the issues that you are experiencing in your own particular scene though.  I can only speak for my own test results.

GHCK57, 2022-01-20:
**"The D415 excels at scanning small static objects, whilst the D435 is optimized for capturing fast motion."**  Is These diffrence thanks to shutter ? 

-> Is there any planning of these sensors EOL ?

[5 more replies]

closed 2022-01-26 by MartyG-RealSense