realsenseai/librealsense #10135: No realsense camera feed obtained while starting RGBD Handheld scanning using rtabmapviz

Rafi-UST, 2022-01-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { Intel Realsense D435i } | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |   {Ubuntu 20.04} |                                 | 
| Platform                                 | {Laptop}  |
| SDK Version                            |  { v2.50.0 }                          | 

### Issue Description
Hi Team , 

I am facing an issue while running RGBD-handheld  scanning using the realsene camera, rtabmapviz  is not able to 
show any feed from the camera. Although rviz is able to get the feed from image raw topic etc.

rafism@rafism-CNBIC-EAN1:~/catkin_ws$ roslaunch rtabmap_ros rtabmap.launch \
>     rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
>     depth_topic:=/camera/aligned_depth_to_color/image_raw \
>     rgb_topic:=/camera/color/image_raw \
>     camera_info_topic:=/camera/color/camera_info \
>     approx_sync:=true \
>     wait_imu_to_init:=true \
>     imu_topic:=/rtabmap/imu
... logging to /home/rafism/.ros/log/3740e122-6ec8-11ec-bebf-d1fc0822d0f1/roslaunch-rafism-CNBIC-EAN1-33691.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rafism-CNBIC-EAN1:37107/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmapviz/

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.13
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/expected_update_rate: 0.0
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/keep_color: False
 * /rtabmap/rgbd_odometry/max_update_rate: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rgbd_odometry/wait_imu_to_init: True
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
[... 563 lines cut ...]
 06/01 15:24:01,058 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 06/01 15:56:20,938 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 15:56:20,988 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 15:56:33,063 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 15:56:33,114 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 15:56:33,165 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:14:51,793 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 06/01 16:14:53,848 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:14:53,899 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:14:53,949 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:14:54,000 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:15:17,089 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:15:17,139 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:15:17,190 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:41:39,518 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:41:39,569 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:41:39,620 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:41:39,670 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:55:36,929 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 16:55:36,980 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 17:26:25,656 WARNING [139660516910848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/01 17:48:56,539 ERROR [139659845822208] (ds5-options.cpp:88) Asic Temperature value is not valid!
 06/01 17:48:57,543 ERROR [139659845822208] (ds5-options.cpp:88) Asic Temperature value is not valid!
 06/01 17:48:58,548 ERROR [139659845822208] (ds5-options.cpp:88) Asic Temperature value is not valid!
 06/01 18:12:13,557 ERROR [139659845822208] (ds5-options.cpp:88) Asic Temperature value is not valid!

MartyG-RealSense, 2022-01-06:
Hi @Rafi-UST  My research of your question indicates that when using rtabmapviz, it should  be set to True in the launch file whilst RViz is set to False.  And if Rviz is to be used then RViz is set to True and rtabmapviz set to False.

https://github.com/IntelRealSense/realsense-ros/issues/1653 provides an example of a RealSense ROS case with a D455 camera model, where RViz is set to True and rtabmapviz to False.

[image]

In the RealSense ROS wrapper's **opensource_tracking.launch** file, you can similarly see rtabmapviz set to False by default.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/opensource_tracking.launch#L23-L24

Rafi-UST, 2022-01-06:
Hi @MartyG-RealSense , 

Thanks for the quick response. The rtabmap launch command that i am using is :

`roslaunch rtabmap_ros rtabmap.launch \
    rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
    depth_topic:=/camera/aligned_depth_to_color/image_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false \
    wait_imu_to_init:=true \
    imu_topic:=/rtabmap/imu`

So here when I cross-checked the rtabmap.launch file , its showing that rtabmapviz parameter is set to true and rviz to false: 

[image]

Rafi-UST, 2022-01-06:
Hi @MartyG-RealSense , 

The issue is resolved , I forgot to add the imu_filter_node , which is :

` rosrun imu_filter_madgwick imu_filter_node \
    _use_mag:=false \
    _publish_tf:=false \
    _world_frame:="enu" \
    /imu/data_raw:=/camera/imu \
    /imu/data:=/rtabmap/imu`

Thanks for the support .

MartyG-RealSense, 2022-01-07:
That's great to hear that you found a solution, @Rafi-UST - thanks very much for the update!

closed 2022-01-07 by Rafi-UST