realsenseai/librealsense #10138: Intel Real Sense D465 - looking for specs and documentation.

ilyak93, 2022-01-06:
Hi,I have a camera which type is D465:

`strcmp(dev.get_info(RS2_CAMERA_INFO_NAME), "Intel RealSense D465") == 0)`

this peace of code returns True.

I couldn't find the proper documentation and specs for this type of model, does D455 is somehow related to this one ?

I think my manager told me that D465 is a limited edition models, but I'm not sure and he isn't approachable for now.

Will be grateful for any clues, thank you in advance.

MartyG-RealSense, 2022-01-06:
Hi @ilyak93  You could likely use the **rs-enumerate-devices** SDK tool to retrieve configuration information from the camera hardware, as technical documentation for this model is not available.  The tool has a range of launch modes, such as **rs-enumerate-devices -c** to retrieve calibration information such as the camera's intrinsics and extrinsics.

https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

Here are details about how to launch this tool.

**Linux**
Input **rs-enumerate-devices** (or one of its specialist launch instructions such as rs-enumerate-devices -c) in the Linux terminal.

**Windows**
Follow the instructions at https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-615365409

ilyak93, 2022-01-06:
@MartyG-RealSense, thanks. 

I wondered if this model has the ability to anticipate in a multi-sensoring system, being used a Master or a Slave or triggered by external signal generator with high frequency, as it described [here](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-e-external-trigger) and [here](https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-4-genlock-usages).

I don't see the pins as in D435:
[image]

although in my camera maybe it is the left one:
[image]
[image]

Do you have an idea how can I figure it out?

MartyG-RealSense, 2022-01-06:
On the D455 the hardware sync pins are on the top of the casing, under a small removable cover.  If there is not such a cover on the top of the D465 then the slot on the underside probably is the sync pins.  It resembles the opened sync pin recess that I have on a D435i. I cannot offer a guarantee that it is operational though.

ilyak93, 2022-01-06:
@MartyG-RealSense, thank you for your answer. 
It a little bit surprising what you are saying, it is common to all the D-400 series, that synchronization should be done only with the same devices ? 

Because my task is to synchronize indeed different cameras (real-sense with thermo-cam of different company - other resolution, other value-space: 16 unsigned values of temprature), I hoped this could be done in such way that one of them can be a Slave and the other is a Master and vice versa (I wanted to try both). 

Other-wise I probably need an external signal generator that will trigger the cameras simultaneously, don't I ? 

Something as described here:
[image]

MartyG-RealSense, 2022-01-06:
An example of hardware differences between models is that the D415's projector is always on, whilst the D435 / D435i projector pulses in line with exposure.  So such camera models need to have their IR Emitter set to Always On to replicate the D415 projector behaviour.  Also. the D415 has its RGB sensor integrated into the camera circuit board, whilst on D435 / D435i the RGB sensor is separately attached via an internal cable.  And then there are the differences such as the distance between sensors and length of the stereo baseline, etc.  So mixing models may lead to inconsistencies in performance between the cameras in the sync arrangement.

If you need to sync RealSense cameras with non-Realsense ones (such as RealSense depth and an HD RGB cam) then you can do that though, either by syncing with camera metadata or with the External Synchronization (genlock) hardware sync system.  Bear in mind that External Synchronization is considered to be an experimental tool and not considered to be mature.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

MartyG-RealSense, 2022-01-18:
Case closed due to no further comments received.

closed 2022-01-18 by MartyG-RealSense

ilyak93, 2022-02-22:
@MartyG-RealSense, Hi again.
Now I got back to this issue, how can I know what is the ground pin and what is the sync pin on my Intel RealSense D465 ? 
I have looked at the blue arrowed port (the one that I draw in the images) and it doesn't look that there are pins at all, so maybe it isn't the pins hole and there isn't one on this model of camera, is it possible ? 
Who should I try to contact to ensure this ?

MartyG-RealSense, 2022-02-22:
I recently posted some pictures at https://github.com/IntelRealSense/librealsense/issues/10179#issuecomment-1037996775 that show the positions of pins 5 (sync, the orange wire) and 9 (ground, the grey wire).  From the perspective of the pictures, the front of the camera would be facing towards you.  

The pin set is very hard to see even when looking in the recess closely as the pins are very thin.  Your highlighted port does match the sync pin slot on a retail camera though.

ilyak93, 2022-02-23:
@MartyG-RealSense , I see, thank you very much!

ilyak93, 2022-03-29:
@MartyG-RealSense, hi again. I looked over through the info, and the strange thing is in the pins hole of my rs camera model  there are 20 metal pins, 10 each side: top and bottom, so I wonder how it correspond to the 9 pins in D435/D455 rs cameras ? 
Or maybe it can be assembled to the 9 pins connector somehow according to this [guide](https://vcl3d.github.io/VolumetricCapture/docs/hardware/rs2_hardware/) or similarly.

MartyG-RealSense, 2022-03-29:
I looked again at this issue but without any technical references for this old apparent prototype D465 model there is not any advice that I can offer about the specific functions of the pins on your camera, unfortunately.