realsenseai/librealsense #10139: How to check if cameras are busy and restart them at start up?

joshuawillman, 2022-01-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 / D435i / L515  | 
| Firmware Version                   | v2.50.0 | 
| Operating System & Version |  Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  5.11.0-43-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                         | 
| Segment			|  Robot                         | 

----------------------------------------------------------------------------------------------------

Hi, 

I am using pyrealsense2 and three different RealSense cameras (D415 / D435i / L515). I am currently creating a vision system that can be accessed remotely. 

One issue that I have found when working with RealSense cameras is that if a Python script is run and then encounters an error, then the next time the script is run, the camera is busy. The only way to fix that issue is to unplug the camera and plug it back in. But this poses an issue in a remote system. 

What I need is to check for all available cameras upon start up. If any of them are busy, I need to restart the camera programmatically (perhaps by catching the error at start up and resolving it) . 

I have looked through the documentation and tried finding some help online for this issue. I have looked at checking what cameras are available using `context.query_devices()`. I have tried using `config.disable_stream()`, but that hasn't worked. (If that is an option, I could also be coding it incorrectly.)

Before sharing any code, I was simply wondering if there are any suggestions, pyrealsense2 methods, or other links that would be useful to solve this problem. Thank you!

chris-durand, 2022-01-07:
Hi,

I don't know if it helps in your case, but there is a [`hardware_reset()` method](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.device.html#pyrealsense2.device.hardware_reset):

```
import pyrealsense2 as rs
context = rs.context()
devices = context.query_devices()
for device in devices:
    device.hardware_reset()
```

MartyG-RealSense, 2022-01-07:
Thanks very much @chris-durand for your suggestion to @joshuawillman about hardware_reset()

Hi @joshuawillman   If all three cameras are attached to the same computer then a simple start-up approach would be to automatically build a list of the attached cameras, and check the length value of the list.  If there are meant to be 3 successfully detected cameras on the list and the list length is <3 (less than 3) then perform a reset.  https://github.com/IntelRealSense/librealsense/issues/2332#issuecomment-419959696 provides a Python scripting reference for a list size check.

A problem though is **how** to perform the reset.  Resetting all the cameras in the list with hardware_reset() would not reset the ones that are missing from the list.  A better solution may therefore be to use a Linux **USB reset script** to reset the entire USB port instead of specific cameras, as this method does not depend on being able to detect a camera.  https://github.com/IntelRealSense/librealsense/issues/8393 discusses this subject in regard to Python.

joshuawillman, 2022-01-07:
@MartyG-RealSense and @chris-durand Thank you so much for your quick responses!
I will test it out over the next few days and get back to you with any questions/status updates/solutions that I have.

MartyG-RealSense, 2022-01-07:
You are very welcome, @joshuawillman - I look forward to your next report.  Good luck!

MartyG-RealSense, 2022-02-01:
Case closed due to no further comments received.

closed 2022-02-01 by MartyG-RealSense

joshuawillman, 2022-02-01:
Hi @MartyG-RealSense,

Sorry for the late response. The issue I had was resolved using a combination of your suggestions. 
1. I added each of the camera's serial numbers to a list
2. At startup, I detected if all cameras were already detected and available. 
3. If a device was detected but would not start, I performed the hardware reset (using a Python try-except clause). This was generally caused by my application not shutting down properly, causing the cameras to still be "busy".
4. If, finally, a device was not found, I used the Python `re` and `subprocess` modules to query the USBs for connected to devices and handle the problem accordingly. 

The beginning code for querying devices connected to my computer:
```
import re
import subprocess

device_re = re.compile(b"Bus\s+(?P<bus>\d+)\s+Device\s+(?P<device>\d+).+ID\s(?P<id>\w+:\w+)\s(?P<tag>.+)$", re.I)
dev_found = subprocess.check_output("lsusb")
devices = []
for dev in dev_found.split(b'\n'):
    if dev:
        info = device_re.match(dev)
        if info:
            dinfo = info.groupdict()
            dinfo['device'] = '/dev/bus/usb/%s/%s' % (dinfo.pop('bus'), dinfo.pop('device'))
            devices.append(dinfo)
            
print(devices)
```
I've added it here in case anyone else wants to check for their devices in code. 

Again, sorry for the late response. Thanks!

MartyG-RealSense, 2022-02-01:
No problem at all.  Thanks very much for the update and for sharing your detailed solution with the RealSense community. :)