realsenseai/librealsense #10140: Issues powering/accessing D435i when running in Docker

Brac24, 2022-01-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.11.0-44-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.5                       | 
| Language                            |  ROS 2, python                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am following the the instructions from here https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_odometry. This runs a docker container with all the necessary dependencies (ROS 2, RealSense SDK, etc..). Within this container I can run all the realsense tools like the realsense-viewer, rs-depth, etc. The issue is that it looks like it is having trouble actually powering on the camera when running these tools. It shows that it detects the camera but for some reason it can't power it on. Not sure if this a docker access issue or actual power issue with my usb ports. All my usb ports power all other peripherals just fine. I can connect the camera to my windows laptop and run all the realsense tools there just fine. I just upgraded the firmware through my laptop and tried again but that did not seem to work.

Here is what the realsense viewer log shows
[image]

This is the message from rs-enumerate-devices:
Could not create device - failed to set power state . Check SDK logs for details
 08/01 20:25:12,696 ERROR [140458703419136] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS

I have also  added 99-realsense-libusb.rules but still didn't work.
Any other suggestions would could be causing this issue?

Thank you

Brac24, 2022-01-09:
I solved by installing librealsense onto my machine.

closed 2022-01-09 by Brac24

iraadit, 2022-08-16:
Hi @Brac24, did you ever try to get it working through Docker afterwards, I'm in the exact same position as you were.

It is mentioned in the troubleshooting from [isaac_ros_common](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docs/troubleshooting.md#realsense-ros2-launch-realsense2_camera-rs_launchpy-error), but following their solution doesn't work for me. :(

Brac24, 2022-08-16:
I did but not using Isaac ros. I use the containers for Jetson Nano from here https://github.com/dusty-nv/jetson-containers and also had to change the script to run the container with privileged access