realsenseai/librealsense #10142: Can I save rs-pcl.cpp as a .pcd file?

gitttable, 2022-01-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Camera Model                       | { D400 } | 
| Operating System & Version |    Linux (Ubuntu 18.04) |                                          
| Platform                                 | PC |                  | 

hey all. 
We need to make a normal vector with L515 and D435. I can make a .pcd file with xyz rgb, but I don't know how to make a .pcd with only xyz values. I want to create a .pcd using librealsense/wrappers/pcl//pcl/rs-pcl . Help.

MartyG-RealSense, 2022-01-10:
Hi @gitttable  The RealSense case https://github.com/IntelRealSense/librealsense/issues/4210#issuecomment-503064696 provides an example of Python scripting for saving xyz to PCD by importing **pclpy**

Alternatively, I wonder if you could adapt the xyzrgb version of the rs-pcl example - **rs-pcl-color** - that includes a pcd saving function by changing all of the **pcl::PointCloud<pcl::PointXYZRGB>** references to **pcl::PointCloud<pcl::PointXYZ>** so that it only saves xyz information to the pcd.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl-color/rs-pcl-color.cpp

gitttable, 2022-01-10:
Thanks you @MartyG-RealSense 
I'm trying to pcl::PointCloudpcl::PointXYZRGB references to pcl::PointCloudpcl::PointXYZ
you said change PointXYZRGB -> PointXYZ in rs-pcl-color? 
I tried, but it did not works. 

May i change all RGB? 
What should i do just like RGB_Texture? It can just delet? (65 line of rs-pcl-color code) 
 
std::tuple<int, int, int> RGB_Texture(rs2::video_frame texture, rs2::texture_coordinate Texture_XY)
{
    // Get Width and Height coordinates of texture
    int width  = texture.get_width();  // Frame width in pixels
    int height = texture.get_height(); // Frame height in pixels

MartyG-RealSense, 2022-01-10:
Thanks very much for your patience.  Aside from the RGB references mentioned above, another line that may need editing is 215:

**pc.Map_to(RGB);**

Since we do not want to map the point cloud to RGB, the appropriate course of action may be to map to the depth-frame reference instead.

**pc.Map_to(depth);**

gitttable, 2022-01-11:
Thank you for your reply. What I want to do is do normal estimation.
Normal estimation is possible with only xyz, but I can't quite figure it out with xyzrgb. Do you know an example of normal estimation using xyzrgb?

gitttable, 2022-01-13:
Thanks, @MartyG-RealSense . I Can fix own my way with your help. Thanks you. Have a good day. :)

MartyG-RealSense, 2022-01-13:
You are very welcome, @gitttable   :)

MartyG-RealSense, 2022-01-18:
Case closed due to satisfaction with outcome and no further comments received.

closed 2022-01-18 by MartyG-RealSense