realsenseai/librealsense #10145: How the rs::colorizer compress a depth map to RGB

dyd1234, 2022-01-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Ubuntu |
| SDK Version                            |  {  2.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  { }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi, I have seen the [white paper](https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras#section-4compression-using-image-codecs) and a related closed issue #6466. I noticed that the equation in chapter 2.1 is somehow setting some values in RGB negatively. The equation shows below:
[image]

I wish to know how you guys manage the displays of the negative RGB values? 

And why the RGB data changes to RGB24(RGB 8bit x 3) when it is in the recovery of depth images?
[image]

BTW: Sorry for posting an unfinished issue. I have closed it.

MartyG-RealSense, 2022-01-11:
Hi @dyd1234  It looks as though the values are negative because **inverse** (minus-value) colorization is being applied, as described by the section of the **Depth image compression by colorization** white-paper document that the above formula is quoted from.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras#section-2-1-uniform-colorization-and-inverse-colorization

The paper states that as the inverse colorization method uses **disparity** values instead of depth, it results in finer quantization at close range and therefore improved image quality.  The paper adds, "Inverse colorization is particularly well suited to depth cameras, like the D400 series, that derive depth from triangulation, which inherently has an inverse relationship between resolution and distance away".

In regard to RGB 24, my intepretation of the information in the **Depth Image Colorization** section linked to below is that use of RGB 24 helps to avoid loss of detail during **lossy compression**.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras#section-2depth-image-colorization

dyd1234, 2022-01-12:
Thank you for the reply! One more question for the colorization equation and the c++ implementation.

I noticed that the c++ implementation of colorization from chapter 3.1 in the white paper is using the SDK post-processing filter. Does the post-process with Hue colormap use the colorization equation in chapter 2.1? I sourced a bit down in the SDK  
files but I did not find the implementation of the **colorizer.process()**.

dyd1234, 2022-01-12:
@MartyG-RealSense Thank you for the reply! One more question for the colorization equation and the c++ implementation.

I noticed that the c++ implementation of colorization from chapter 3.1 in the white paper is using the SDK post-processing filter. Does the post-process with Hue colormap use the colorization equation in chapter 2.1? I sourced a bit deeper down in the SDK files but I did not find the implementation of the colorizer.process().

MartyG-RealSense, 2022-01-12:
A RealSense team member explains at https://github.com/IntelRealSense/librealsense/issues/6466#issuecomment-637385906 that Hue is configured by setting the **color scheme** to '9' with the line **color_filter.set_option(RS2_OPTION_COLOR_SCHEME, 9.0f);**

dyd1234, 2022-01-13:
Thanks! I think I have all my questions answered.

closed 2022-01-13 by dyd1234

MartyG-RealSense, 2022-01-13:
You are very welcome, @dyd1234 - thanks very much for the update!

cheind, 2024-10-17:
Although this comes late in the game, for future reference, here is a vectorized Python implementation with the equations corrected 

https://github.com/cheind/hue-depth-encoding