realsenseai/librealsense #10146: Color frame in .ply file

TheMarsupilami, 2022-01-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   {Win (10)   |   | 
| Platform                                 | PC  |
| Language                            | python/                         | 

Hello Everyone

Actually the goal is the same as in this issue I#6194 saw all this post but without any solution. Since some time has passed, I was wondering if there is now a solution ? 
Or is there a work around ?

MartyG-RealSense, 2022-01-11:
Hi @TheMarsupilami  As far as I am aware, the method in https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293 for saving a ply with color and without normals is still the only known successful method of doing so in Python code.

TheMarsupilami, 2022-01-11:
Thanks for your reply.
As far I understand that code snippet he is applying the color of the deepth stream and lays this into the ply file. 
The color comes from the deepth stream. I want to produce #6194(comment) the right picture with python.

1. Is Intel looking into this problem, because I saw alot of issues in this forum which are exactly about this problem ? Because it's a bit frustrating when this function is possible with realsenseviewer but then this option is not enabled with Python.

2. Is this just a problem just with python ?

MartyG-RealSense, 2022-01-11:
1.  There is not a currently active Intel investigation of the issue.  My recollection is that it was looked at in the past though.

2.  It is primarily a Python problem, as there are C++ cases - such as https://github.com/IntelRealSense/librealsense/issues/9071 - where a ply has been successfully exported with color.

Any investigation about saving a color ply with Python always ends up coming back to the **export_to_ply** based example linked to at https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293 unfortunately, as it is the only confirmed example that works (I have researched Python color ply exporting extensively over past years). 

This is demonstrated by a script at https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152 that uses the SDK's other ply export method **save_to_ply**.  The script works but did not export color for the script's creator.  You can of course try it yourself to see whether you get different results.

closed 2022-01-24 by MartyG-RealSense