realsenseai/librealsense #10154: doesn't support architecture 'arm64'

HusamZain, 2022-01-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   ubuntu  20.04 | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                        | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  not installed                         | 

### Issue Description
I'm trying to install SDK and when i run this code in **step 2** 
`sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u`
I got this 
`Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'arm64'`
 and when i run command in step 3 
`sudo apt-get install librealsense2-utils`
I got this error 
**E: Unable to locate package librealsense2-utils**
>
Does anyone know how to solve this ? 
I'm trying to use realsense with Ros2

MartyG-RealSense, 2022-01-13:
Hi @HusamZain  I note that you are using Ubuntu 20.04 (Focal).  This error is a known issue with librealsense installation from **packages** on Focal and is discussed at https://github.com/IntelRealSense/librealsense/issues/9040

Intel looked at this issue during the second half of 2021 and my understanding is that it was planned that it would be addressed in the SDK at a future date, though I do not currently have a specific time schedule to provide.

You could instead build the SDK on Jetson and Focal from source code instead of installing from packages.

HusamZain, 2022-01-19:
@MartyG-RealSense thank you for your help.
I am trying to bulid it from source using RSUSB backend method, however I got this error when I enable Cuda using this **-DBUILD_WITH_CUDA=true** as mentioned here [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-rsusb-backend](url)

### error 
[image]

### is there a way to solve ?

MartyG-RealSense, 2022-01-19:
https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13#issuecomment-573976359 suggests making an edit to the /bashrc file and then reloading the terminal to correct this error.

HusamZain, 2022-01-20:
that was helpful , thank you.

closed 2022-01-20 by HusamZain