realsenseai/librealsense #10157: Suitable camera for a close distance

LadaOndris, 2022-01-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 series  | 
| Firmware Version                   |  | 
| Operating System & Version |   Linux | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 |  |
| SDK Version                            |                           | 
| Language                            |                           | 
| Segment			|                       | 

### Issue Description

I don't specifically have a problem, but rather a question of what camera would suit our use case the best. We need a depth camera to capture **moving** hands in the range of **10 cm to 100 cm**. The use case is pose estimation of the hand (fingers and palm).

At the moment I have two cameras at my disposal---SR305 and D415. SR305 is no longer being sold, so I don't consider this an option anymore. The advantage of SR305 was its depth accuracy. As for the D415, I am able to decrease min-z by decreasing resolution and increasing disparity shift which gets me to min-z of 13 cm. The issue is, however, that at this distance the palm is too big to fit in the image. I suppose this is because of limited FOV. Experimentally, my hand needs to be at least at the distance of 22 cm so that it fits inside it whole; thus, the FOV is also limiting for our use case.

The following image demonstrates the above-mentioned scenario when my hand is captured by the D415 camera at 13 cm with a disparity shift of 30 and resolution 480x270.
[image]

The other issue with D415 is that the depth accuracy is too low when compared to SR305. I do not know to what extent settings can be adjusted or if other cameras in the D400 series are more accurate.

To summarize, the requirements are:
* range from 10 cm to 100 cm or more
* moving hands
* the whole palm should fit in the image (wide enough FOV)
* accurate depth (almost similar to SR305)

I explored a bit and found that D435, D435i, and D455 all have wider FOV. Moreover, I read that they support closer distances than D415. Therefore, I wonder whether either one of them could suit our case. 

Thank you for any advice.

MartyG-RealSense, 2022-01-14:
Hi @LadaOndris  The 400 Series cameras are significantly more capable than the older SR30x models.  As the SR30x is based on a different depth sensing technology to the stereo depth of the 400 series, there are a small number of applications that may produce a more desirable result on SR30x than the stereo depth technology can.  It is usually possible to compensate for this and enhance the image with the extensive range of powerful features such as post-processing and custom camera configuration that are available to the 400 Series though.

If the 0.3 meter minimum depth sensing distance of the D415 camera model is problematic for your project then the D455 would not be the most suitable choice as its minimum distance is 0.4 meters.  The D435 and D435i are well suited for close range sensing as they have a minimum depth distance of only 0.1 meters.

You can still use disparity shift with D435 and D435i to reduce the minimum distance below 10 cm, though there will be a distance below which the image begins blurring. 7 cm would be a good target for an absolute minimum distance with disparity shift.

An entire hand could fit within the field of view (FOV) size of D435 / D435i.

Because the RGB sensor on these models have a slower 'rolling' shutter whilst the depth sensors have a fast 'global' shutter, the RGB image may experience motion blur with a fast moving hand, as described in https://github.com/IntelRealSense/librealsense/issues/3554 but there are a couple of workarounds for reducing RGB blur.

If RGB auto-exposure is enabled then setting the FPS speed to 60 will help to negate the blurring.  If RGB auto-exposure is disabled though and manual exposure is being used, you can use 6 FPS (not 60) and set a manual RGB exposure value of 70.   https://github.com/IntelRealSense/librealsense/issues/3554 provides technical information that explains these settings.

LadaOndris, 2022-01-15:
Wonderful, thank you for the detailed information. We are not intending to use RGB, so the described motion blur should not be a problem in our case. From the information you provided, it seems that D435 and D435i are worth a try.

Could you shortly compare these two cameras and say which one may be more suitable?

MartyG-RealSense, 2022-01-15:
The main difference between D435 and D435i is that the D435i has an additional internal hardware component called an **IMU** that provides accelerometer (accel) and gyroscope (gyro) data.  If you do not require such motion data for your project and are only using depth then the IMU-less D435 would be fine.

A good visual demonstration of the fast motion capture capability of the D435 / D435i depth sensor is the YouTube video in the link below, where a D435 is mounted to a car traveling at full speed.

https://m.youtube.com/watch?v=OwJmCyAn3JQ

roelofvandijk, 2022-01-18:
Here is a similar issue that might help you: https://github.com/IntelRealSense/librealsense/issues/4550

Additionally, if you have not read them yet, I would suggest you take a look at these two Intel whitepapers:

[Tuning depth cameras for best performance](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance)

[Depth Post-Processing for Intel® RealSense™ Depth Camera D400 Series](https://dev.intelrealsense.com/docs/depth-post-processing)

MartyG-RealSense, 2022-01-18:
Thanks very much @roelofvandijk for the list of resources for @LadaOndris   :)

LadaOndris, 2022-01-18:
Thank you, @roelofvandijk, I really appreciate it. It is a good idea to tune the camera a little.

LadaOndris, 2022-01-18:
And also thank you, @MartyG-RealSense, for explaining the difference between the two cameras. We'll probably give the D435 a try.

MartyG-RealSense, 2022-01-24:
Hi @LadaOndris  Do you require further assistance with this case, please?  Thanks!

LadaOndris, 2022-01-24:
I do not think so, thank you!

closed 2022-01-24 by LadaOndris

MartyG-RealSense, 2022-01-24:
You are very welcome.  Thanks for the update!