realsenseai/librealsense #10159: WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

hichamhendy, 2022-01-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Linux (Ubuntu 18.04)| 
| Kernel Version (Linux Only)    |  Linux 5.4.0-92-generic  | 
| Platform                                 | Laptop XMG  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  ROS | 

### Issue Description
<Describe your issue / question / feature request / etc..>

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: l515
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
[... 123 lines cut ...]
 17/01 15:57:54,820 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:57:54,872 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:57:54,927 WARNING [140592778786560] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1642431474.945089136]: 
[ WARN] [1642431474.945223538]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[ INFO] [1642431474.970517233]: SELECTED BASE:Depth, 0
 17/01 15:57:54,970 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ WARN] [1642431475.021540778]: Publishing dynamic camera transforms (/tf) at 20 Hz
[ INFO] [1642431475.023347132]: RealSense Node Is Up!
 17/01 15:57:55,630 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:57:55,771 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:57:57,023 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:57:58,274 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:57:59,526 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:58:00,777 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:58:02,029 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:58:03,280 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 17/01 15:58:04,532 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/01 15:58:05,783 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/01 15:58:07,034 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/01 15:58:08,285 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/01 15:58:09,536 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/01 15:58:10,788 WARNING [140592482850560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

I suspect that this error is related to freezing/crushing the Laptop I reported before without success in solving

hichamhendy, 2022-01-17:
@ #8020

MartyG-RealSense, 2022-01-17:
Hi @hichamhendy  If the **control_transfer returned** warning is generating continuously in the log then this can indicate that there is a serious communication problem with the device.

Could you try adding the instruction **initial_reset:=true** to the end of your roslaunch instruction to see whether it improves stability by resetting the camera at launch, please?  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

MartyG-RealSense, 2022-01-17:
Are you **closing down** the rs-enumerate-devices program before performing the next launch in order to eliminate the possibility that the rs-enumerate-devices program is still holding onto a claim on the camera's access and so preventing a roslaunch from accessing it?

Under the **multi-streaming-model** rules of librealsense, once a program claims access to a particular stream such as depth then another program cannot access that particular stream afterwards until the first program releases its lock on the stream (for example, by closing that program).

hichamhendy, 2022-01-17:
As I said,  the rs-enumerate-devices program is one of two cases that sent me into freezing. However, I execute the command on a terminal and expect and that it gets automatically killed, after it shows the results!

What I if I shut down (ctl + c) the node, what is the going on in that case? However, reinitiating the node doesn't reproduce the freezing. Yet, changing any argument in the configurations leads to the freezing/crushing.

MartyG-RealSense, 2022-01-18:
Rather than editing the launch file contents, have you tried setting the changes to parameters as a custom configuration in the roslaunch instruction instead so that the launch file does not need to be edited?  For example, to configure both depth and color at 640x480 and 30 FPS:

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

From ROS wrapper 2.2.22 onwards, a custom stream configuration must include three factors for each stream type used in order for the launch to consider it to be valid: width, height and FPS.

On the L515 model, firmware **1.5.8.1** is required for RGB to support 640x480 color, though you already have this firmware installed and so should be able to use this resolution.

caoluming, 2022-01-19:
Mine is in a similar situation. Have you solved it

hichamhendy, 2022-01-19:
@MartyG-RealSense  by the way I get the same message by the way when I call `realsense-viewer`. So I wouldn't suggest that the problem is related to ROS.

@caoluming  I still get the error! The freezing problem crushed yesterday the whole OS and I needed to reinstall a new Ubuntu 18.04. I will be examining that and report. 

Till then, my main issue stays by **messenger-libusb**. It's a warning yet it's not understandable.

hichamhendy, 2022-01-19:
@MartyG-RealSense  I attached a crash report maybe that can help
[linux-image-5.4.0-94-generic.256950.txt](https://github.com/IntelRealSense/librealsense/files/7897939/linux-image-5.4.0-94-generic.256950.txt)

MartyG-RealSense, 2022-01-19:
@hichamhendy   I was just looking at your case when you updated.  What method did you use to install the librealsense SDK, please (packages or build from source code).  

If you built with source code, did you use the **RSUSB backend** installation method?  If it is a kernel related issue as your new log suggests then building with the RSUSB method could help with stability, as it is not dependent on Linux versions or kernel versions and does not require patching.

hichamhendy, 2022-01-19:
I installed as indicated here

Install the libraries (see section below if upgrading packages):
sudo apt-get install librealsense2-dkms
´sudo apt-get install librealsense2-utils´
The above two lines will deploy librealsense2 udev rules, build and activate kernel modules, runtime library and executable demos and tools.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

MartyG-RealSense, 2022-01-19:
A package build will still be making use of the kernel, so if there is still a kernel related conflict even after switching kernels then a package build will not correct that.  Building from source with the RSUSB backend method is necessary if one wants to bypass the kernel and help to avoid such conflicts.  Having said that, the 5.4 kernel that you were originally using at the start of this case is officially supported by librealsense, so it is strange that you would be having problems with it.

Reading the case again from the beginning, I note that **align_depth** is set to True.  Do the control_transfer returned messages reduce if align_depth is False?

hichamhendy, 2022-01-19:
Should I follow the instructions here https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk ?

MartyG-RealSense, 2022-01-19:
The instructions in the quoted link are for a non-RSUSB build that will require kernel patching.  Instructions for an RSUSB source-code build that is independent of the kernel can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

hichamhendy, 2022-01-20:
I followed the instructions indicated here (https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692) but nothing changed

MartyG-RealSense, 2022-01-20:
The **multiple udev rules** message can appear if librealsense was built from both packages and source code on the same computer.  Deleting the rule at **etc/udev/rules.d** should make the error disappear.

hichamhendy, 2022-01-23:
I tried the camera on different Laptop with same Ubuntu version and kernel, and it worked fine. I even tried it on Jetson Xavier Tegra and it went well, besides I had troubles with rtabmap but this is not the issue here.  There was no freezing. I doubted that that might be a problem with ubuntu but as I said the frequent crushing leaded me to new installations of Ubuntu. Yet, I still suffer from the same issue. The only thing I can think of is the driver compatibility with hardware of my Laptop.

Could anyone from support tell my if XMG (https://bestware.com/de/xmg-pro-17.html) is not compatible with L515?!

MartyG-RealSense, 2022-01-24:
I cannot see anything in the XMG specification on the provided web-page that would obviously cause an Ubuntu freeze / crash.  I would recommend running the RealSense SDK's **rs-hello-realsense** example program to see whether it is a graphics related problem, as rs-hello-realsense does not use graphics and has a text output.  If Ubuntu still freezes when rs-hello-realsense is used then this would suggest that it is not a graphics related problem.

If the SDK examples were built when you installed librealsense on Ubuntu then a pre-built executable version of rs-hello-realsense should be at the folder location **usr/local/bin**

hichamhendy, 2022-01-24:
Regarding the first suggestion, I concluded that the issue is related to graphics because the rs-hello-realsense script doesn't reproduce the issue. Where is the relevance with the graphics here? haw can it be fixed?

All scripts are there:
[image]

MartyG-RealSense, 2022-01-24:
The RealSense SDK makes use of **OpenGL** graphics.  If there are problems when running graphical RealSense programs then updating the graphics drivers on your PC may help.  This is because the drivers for popular-brand video chips such as Nvidia include OpenGL support.

hichamhendy, 2022-01-24:
I just figured out the OS freezed after while even though I've already plugged out the camera and stopped operating on it. I haven't noticed this behavior before. I think the USB-peripheral clams the process power that don't get released every time you finish the operation and maybe expands!

hichamhendy, 2022-01-24:
I just found out that the issue can be reproduced on another Labtop/tablet with Ubuntu 1804

Details

Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 18.04.6 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
git: 		| git version 2.17.1
cmake: 		| cmake version 3.10.2 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost: 		| Version: 1.65.1.0ubuntu1
eigen3: 	| Version: 3.3.4-4

otherwise I installed everything the same way.

MartyG-RealSense, 2022-01-24:
Does the crash still occur if you unplug the micro-sized end of the USB cable from the base of the L515, reverse the connector's orientation and plug it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).

hichamhendy, 2022-01-24:
No that doesn't free the freeze/lock

MartyG-RealSense, 2022-01-24:
Does it freeze if you boot the computer without the camera plugged in and then launch the RealSense Viewer without the camera inserted?

hichamhendy, 2022-01-24:
It doesn't freeze. I therefore presume a connection with the camera causes this behavior

MartyG-RealSense, 2022-01-24:
The next test that you could try is to enable the camera streaming in the Viewer with the camera plugged in and run an Ubuntu system monitoring tool such as **htop**.  If you monitor memory usage then you could see whether the available memory of the computer is being consumed over time (a 'memory leak') until it crashes.

hichamhendy, 2022-01-25:
I finally became able to specify the problem. The connection with camera manges some how to double the process /usr/lib/xorg/Xorg which uses GPU Memory in Ubuntu 18 Server

Normal case:
[image]

I closed later teams.

[image]

I closed the viewer
[image]

I even closed the terminal which shouldn't really change alot
[image]
sed

MartyG-RealSense, 2022-01-25:
As your L515 camera was able to work normally with another PC and with a Jetson, it does sound like an issue with Ubuntu Server on the affected machines rather than a problem with the camera hardware, its firmware or its USB cable.

Googling for the term **ubuntu memory leak /usr/lib/xorg/Xorg** shows that there is a high number of reports of memory leaks related to Xorg.

Disabling Xorg does not seem to be a viable option due to the amount of other systems in Ubuntu that it is connected to and stops if Xorg is killed, as you mentioned.  A popular suggestion is to instead edit the file **/etc/X11/xorg.conf** in some way.

MartyG-RealSense, 2022-01-25:
That's excellent news that you found a solution.  Congratulations, and thanks so much for sharing your method with the RealSense community!