realsenseai/librealsense #10160: Make sync YouTube tutorial videos listed

roelofvandijk, 2022-01-18:
Would it be possible to make the synchronization tutorial videos linked to in this comment publicly searchable on YouTube?
https://github.com/IntelRealSense/librealsense/issues/3084#issuecomment-465825421
It would make it much easier to find them.

MartyG-RealSense, 2022-01-18:
Of the three videos listed, the one in the link below relates to setting Inter Cam Sync Mode.

https://www.youtube.com/watch?v=DSm7NYG5bZE

This RealSense GitHub support forum cannot update the link to make that particular video findable without a link though as we are not involved in administrating the RealSense YouTube channel and do not have an admin contact for the channel.

roelofvandijk, 2022-01-18:
Thank you or the quick response, that is unfortunate. The videos are quite useful, and are not linked to anywhere else as far as I see. If they cannot be made listed, maybe they could be added as a link to the synchronization whitepaper? 
https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

MartyG-RealSense, 2022-01-18:
I will add a **Documentation** label to this case so that a RealSense team member involved with documentation can review it at a future date.  This issue should be kept open, but you do not need to take any further action with it.  Thanks for the suggestion!

closed 2026-10-23 by MartyG-RealSense