realsenseai/librealsense #10161: D435i can not connect

Lincolise, 2022-01-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10 | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0                         | 

### Issue Description
My camera can be detected correctly by pc, like what's shown below,btw i'm usint usb 3.0.
[image]
but when i use realsense viewer, it shows me some error and doesn't show any image, please help me with this, thanks!
[image]

[image]

Lincolise, 2022-01-18:
the complete content of error is: sensor.cpp:572 - acquire_power failed: MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "system could not find the specified path"

MartyG-RealSense, 2022-01-18:
Hi @Lincolise  Please check in your Windows privacy settings that permission for the camera hardware to be accessed and permission for applications to access the camera are both set to On.  You can do so using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

Lincolise, 2022-01-18:
> Hi @Lincolise Please check in your Windows privacy settings that permission for the camera hardware to be accessed and permission for applications to access the camera are both set to On. You can do so using the instructions at [#3381 (comment)](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789)

thanks for your fast reply. I've already checked these settings before asking this question, and they are both set to On now. but nothing changes. T T
error said that :"system could not find the specified path", is there any thing i can do around this?

MartyG-RealSense, 2022-01-18:
As your Device Manager lists a USB HD webcam, are you able to unplug it to eliminate the possibility that it is causing a conflict?

Lincolise, 2022-01-18:
> As your Device Manager lists a USB HD webcam, are you able to unplug it to eliminate the possibility that it is causing a conflict?

I'm using laptop now, this USB HD webcam can't be unplugged.

MartyG-RealSense, 2022-01-18:
Another way to check for a conflict with a non-removable camera is to right-click on its Device Manager entry and select **Disable device** from the menu.  You can easily re-enable it by right-clicking and selecting **Enable device**.

Lincolise, 2022-01-18:
> Another way to check for a conflict with a non-removable camera is to right-click on its Device Manager entry and select **Disable device** from the menu. You can easily re-enable it by right-clicking and selecting **Enable device**.

Thanks for your advice, but it shows the same error again. 
[image]
[image]
[image]

MartyG-RealSense, 2022-01-18:
If you have installed the full RealSense SDK usin the autpmated installer file on the SDK **Releases** page, could you check please whether the camera works with the **rs-capture** example program using the instructions at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656 please?

Lincolise, 2022-01-18:
> If you have installed the full RealSense SDK using the automated installer file on the SDK **Releases** page, could you check please whether the camera works with the **rs-capture** example program using the instructions at [#8024 (comment)](https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656) please?

i tried to open it, but it turned white and soon closed.
[image]

MartyG-RealSense, 2022-01-18:
This suggests to me that the camera is not being detected by rs-capture either, as this white screen will appear if the camera is not plugged in.

Next, please run the text-only program **rs-hello-realsense**, which displays a textual read-out of the depth measurement.  This will determine whether the problem is related to graphics or whether the camera is not being detected at all.

[image]

Lincolise, 2022-01-18:
[image]
it shows nothing.

MartyG-RealSense, 2022-01-18:
If you launch the RealSense Viewer without the camera plugged in and then insert the camera, does it ever appear if you insert and unplug it multiple times (waiting a couple of seconds after removal before inserting it again).

Instead of a quick insertion, you may find that it has a greater chance of being detected if you just insert the tip of the connector into the edge of the port and then push it the rest of the way in instead of performing a single pauseless insertion motion.

Do you have another USB port on your computer that you can try the camera in if you have not done so already?

Lincolise, 2022-01-19:
I've tried these methods, but also failed.
Then I used my friend's pc, it worked perfectly..I guess the problem is from my computer.
This problem made me depressed for a long time, thanks for all the patient answer and help of you!
Hope you and your team gets better!

MartyG-RealSense, 2022-01-19:
You are very welcome, @Lincolise - it's excellent to hear that you made progress in finding the cause of your problem!

Lincolise, 2022-01-19:
> You are very welcome, @Lincolise - it's excellent to hear that you made progress in finding the cause of your problem!

It seems like my usb interface cann't afford enough power to the camera? 
because my raspberrypi has the same error.
So I want to know what voltage does d435i need?

MartyG-RealSense, 2022-01-19:
I would recommend budgeting 2 watts (W) for each camera.  A mains electricity powered USB 3 hub can supply around 12 W to be distributed to all the devices attached to it.

Lincolise, 2022-01-19:
> I would recommend budgeting 2 watts (W) for each camera. A mains electricity powered USB 3 hub can supply around 12 W to be distributed to all the devices attached to it.

Thanks for answering!

MartyG-RealSense, 2022-01-26:
Hi @Lincolise  Do you require further assistance with this case, please?  Thanks!

Lincolise, 2022-01-28:
> Hi @Lincolise Do you require further assistance with this case, please? Thanks!

Yes I do.  I'm trying to port my project to raspberry PI these days, but my raspberry pi 4b can't detected the camera. What can I do to find the problem?

MartyG-RealSense, 2022-01-28:
Is your Pi using Raspbian or Ubuntu as its OS, please?  And which version of Raspbian or Ununtu is it using (for example, Raspbian Buster / Bullseye or Ubuntu 18.04 / 20.04).

closed 2022-02-09 by MartyG-RealSense