realsenseai/librealsense #10162: Can Link object file? (ex. .ply -> .pcd)

gitttable, 2022-01-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

hello, 
Can you make a single program by linking the executable files that came out after Cmake and make in each file?
For example, there is a .ply executable file in the first directory and a .pcd executable file in the second directory.
Is it possible to execute .ply in directory 1 and connect directly to .pcd in directory 2 to output the result as .pcd?
I tried using gcc, but there are more than 130 header files per .cpp file. Is there a way?

MartyG-RealSense, 2022-01-18:
Hi @gitttable  Could you provide further description please about what you are hoping to achieve to aid my understanding of your question?  Thanks very much.

As your previous question at https://github.com/IntelRealSense/librealsense/issues/10142 was about PCL, is this a question too about the PCL wrapper?

gitttable, 2022-01-19:
Yes. It is related to the pcl wrappers.
If you run make after cmake, an executable file is created, but I don't know how to link the executable file with other executable files. Any similar examples?

MartyG-RealSense, 2022-01-19:
I researched your question carefully but did not find a definitive solution.  The link below has a non-RealSense discussion about CMake that explored how to link executables together.  A couple of commenters suggested methods for doing it but always ended up with segmentation faults.

https://gitlab.kitware.com/cmake/cmake/-/issues/19527

closed 2022-02-01 by MartyG-RealSense