realsenseai/librealsense #10164: realsense-viewer throws HW error with L515

sivabandapalli, 2022-01-18:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Linux|

Error:  
` 18/01 16:38:29,258 WARNING [140115981969152] (rs.cpp:310) null pointer passed for argument "device"
 18/01 16:38:29,258 WARNING [140115981969152] (rs.cpp:2691) Couldn't refresh devices - hwmon command 0x2b( 4 0 0 0 ) failed (response -29= Value out of range)
`

MartyG-RealSense, 2022-01-18:
Hi @sivabandapalli  When you built the librealsense SDK, did you install the **udev rules**, please?  If you didn't then https://github.com/IntelRealSense/librealsense/issues/3697#issuecomment-482006790 describes how to do so for package builds and source-code builds.

sivabandapalli, 2022-01-18:
Hi Marty, I can confirm the udev-rules were installed.
`librealsense2-udev-rules is already the newest version (2.50.0-0~realsense0.6128).
`

MartyG-RealSense, 2022-01-18:
Thanks very much for the confirmation.  And is there any difference if you unplug the micro-sized end of the USB cable from the base of the camera, turn the connector around the other way and re-insert it into the camera base (USB-C cables are two-way insertion at the micro-sized end).

sivabandapalli, 2022-01-18:
When I try by connecting the other way round, it doesn't throw any error, but realsenseviewer doesn't show it in "ADD SOURCE". When I disconnect it and try it connecting it in the same way, the error is consistent.

MartyG-RealSense, 2022-01-18:
If the camera is detected correctly by the Viewer then it should automatically be listed in the side-panel without having to use the Add Source button.

[image]

Can you confirm that the camera is plugged into a USB 3 port?  Although it is capable of being used in a USB 2 port, it has to have its firmware driver updated first and so the camera would not work if it had been inserted accidentally into a USB 2 port before the firmware update had been performed.

Also, are you using the official 1 meter USB cable that was supplied with the camera?  If an own choice of cable was being used then it would be easily to be unknowingly using a USB 2 cable.

sivabandapalli, 2022-01-18:
Thanks for the prompt responses @MartyG-RealSense.
Yes,
- Camera is plugged into a USB 3 port
- I am using the cable that was supplied with the camera

sivabandapalli, 2022-01-18:
@MartyG-RealSense  I am not sure why it can't be detected in realsense-viewer, I can see it and use it like a webcam in other simple applications.

MartyG-RealSense, 2022-01-18:
Could you test the L515 in the **rs-capture** example program please to confirm whether the camera is capable of being accessed by a RealSense application?

The image below is rs-capture runnig with my own L515.

[image]

sivabandapalli, 2022-01-18:
Blank screen as below:
[image]

MartyG-RealSense, 2022-01-18:
Do you have access to any other computers that the camera could be tested on (Ubuntu or Windows).  

If not, could you confirm which type of computer you are using the camera on (Nvidia Jetson, Raspberry Pi, PC laptop / desktop, etc)

sivabandapalli, 2022-01-18:
It's Laptop: Dell Precision 5540

MartyG-RealSense, 2022-01-18:
As you are using Ubuntu, which kernel version are you using?

sivabandapalli, 2022-01-18:
It's **Ubuntu 20.04.2 LTS** and kernel is **5.11.0-46-generic**

MartyG-RealSense, 2022-01-18:
Support for kernel 5.11 was only added in SDK 2.50.0 via a patch submitted by a RealSense user and tested on a D455 - see https://github.com/IntelRealSense/librealsense/pull/9727

The most recent officially supported kernel is **5.8**.  So it may be that the unofficial 5.11 support has an issue with L515.  

If you built librealsense from packages and do not want to downgrade the kernel on your PC, an option may be to build librealsense from source code with CMake using the **RSUSB backend** installation method.  This method is not dependent on Linux versions or kernel versions and does not require kernel patching.

MartyG-RealSense, 2022-01-24:
Hi @sivabandapalli  Do you require further assistance with this case, please?  Thanks!

sivabandapalli, 2022-01-24:
Hi @MartyG-RealSense ! I am planning to test this on a different laptop that has 5.8 kernel. Also, I am trying the different installation method that you suggested. I will update you. If it works, I may not require further assistance. Thanks.

MartyG-RealSense, 2022-01-24:
Thanks very much @sivabandapalli for the update.  Please do update here about whether your tests succeed or not.  Good luck!

sivabandapalli, 2022-02-02:
@MartyG-RealSense  I tried both the methods, but none were successful. When I connected a different L515 camera it works fine. This says that something is wrong with the camera itself. Any suggestions?

MartyG-RealSense, 2022-02-02:
I reviewed the case again from the beginning, especially taking into account the results described in https://github.com/IntelRealSense/librealsense/issues/10164#issuecomment-1015673669 when inserting the micro-sized end of the USB cable into the camera one way or the other.

If trying the USB cable from the camera that works fine with the camera that has the problems makes no difference then it is likely a camera hardware fault.  You could therefore consider returning the camera to the retailer that you purchased it from using the **Return Material Authorization (RMA)** procedure of that retailer.

sivabandapalli, 2022-02-02:
I tried the camera that doesn't work with the USB cable from the camera that works, still, it shows the same error. @MartyG-RealSense

MartyG-RealSense, 2022-02-02:
Okay, thanks very much for checking. I would therefore recommend trying to return the camera to the retailer that you purchased it from as the symptoms indicate that there is nothing that could be done to correct the problem using software or drivers.

closed 2022-02-14 by MartyG-RealSense