realsenseai/librealsense #10167: Is it possible to obtain previous frame and current frame using pyrealsense2

ranjitkathiriya, 2022-01-19:
Hello there,

I am doing coding in python and I want to do the code for motion detection using a real sense camera and for that, I need every previous and current frame.

I have created code in opencv but I want to create same code using pyrealsense2 because at the detected motion I need ply points for that motion,

code is here,

```
import cv2

capture = cv2.VideoCapture(4)

while capture.isOpened():
    _, img_1 = capture.read()  # previous frame
    _, img_2 = capture.read()  # Current frame

    diff = cv2.absdiff(img_1, img_2)

    diff_gray = cv2.cvtColor(diff, cv2.COLOR_BGR2GRAY)

    diff_blur = cv2.GaussianBlur(diff_gray, (5, 5), 0)

    _, thresh_bin = cv2.threshold(diff_blur, 50, 255, cv2.THRESH_BINARY)

    contours, hierarchy = cv2.findContours(thresh_bin, cv2.RETR_TREE, cv2.CHAIN_APPROX_SIMPLE)
    font = cv2.FONT_HERSHEY_SIMPLEX

    if len(contours) <= 0:
        cv2.putText(img_1,'Motion Not Detection',(50, 50),font, 1,(0, 255, 255),2,cv2.LINE_4)
    else:
        cv2.putText(img_1,'Motion Detection',(50, 50),font, 1,(255, 0, 0),2,cv2.LINE_4)
    cv2.drawContours(img_1, contours, -1, (0, 255, 0), 2)

    cv2.imshow("Detecting Motion...", img_1)

    key = cv2.waitKey(20)
    if key == ord('q'):
        break

capture.release()
cv2.destroyAllWindows()
```

Any Idea?

Thanks in advance for help.

ranjitkathiriya, 2022-01-19:
> Wrong timestamp using frame callback in #5143, has a different issue.

```
_, img_1 = capture.read()  # previous frame
 _, img_2 = capture.read()  # Current frame
diff = cv2.absdiff(img_1, img_2)
```

Every time in the while loop I wanted to obtain the previous and current frame. Is it possible with pyrealsense2?

MartyG-RealSense, 2022-01-19:
How about putting a small **sleep** inbetween the prev and curr lines so that they are not both reading the frames instantly?

```
 prev = frames.get_color_frame()
 time.sleep(0.1)
 curr = frames.get_color_frame()
```

You should also add **import time** to your script's import list if you try this.

ranjitkathiriya, 2022-01-20:
Thanks for providing the solution..

```
    frames = pipeline.wait_for_frames()
    prev = frames.get_color_frame()
    time.sleep(0.1)
    frames2 = pipeline.wait_for_frames()
    curr = frames2.get_color_frame()
```

THis is working for me, we have to wait for the second frame also.

closed 2022-01-20 by ranjitkathiriya

MartyG-RealSense, 2022-01-20:
You are very welcome, @ranjitkathiriya - it's great to hear that the method worked for you.  Thanks very much for the update!