realsenseai/librealsense #10168: L515 Depth Bias

Cpeck71, 2022-01-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Linux - Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                  | 
| Platform                                 | Jetson Nano  |
| SDK Version                            |  2.49.0                          | 
| Language                            |  C++                       | 
| Segment			| Computer vision/Navigation                      | 

### Issue Description

Howdy Realsense team,
I've started to notice a discrepancy of the depth data in the form of a bias. I ran the depth-quality tool and with a known distance of ~1.93m the depth output was ~1.81m. I've noticed about the same value bias in other experiments, for example a test that had real-world data (from a motion tracking system) providing a distance of ~1.3m and the L515 was reading on average 1.11-1.15m.

My efforts so far have been to simply subtract out the bias from my measurements to help refine them, but I was wondering if y'all could help determine a source of this bias or something I can do to calibrate the lidar? I've done some searches in the github issues and believe I've read that there's not a specific calibration tool for the L515, but I also don't think I've found another issue mentioning a bias in the Lidar's measurement.

Thanks so much for the help!

MartyG-RealSense, 2022-01-20:
Hi @Cpeck71  Would it be possible to provide an RGB image of the scene that is being observed, please?  This will help to identify factors that may be negatively affecting the measurement.

For example, a difference between the L515 and the 400 Series models is that the L515 will see through transparent surfaces such as a plastic water bottle and pick up the depth data in the background behind the bottle instead of the bottle itself.

Using the **Short Range** preset can help to improve the image in some situations, as it reduces the **Receiver Gain** and **Laser Power** values.

Cpeck71, 2022-01-20:
Howdy Marty,

Here's a representative example of our current setup in the lab. Right now I'm using checkerboards for calibration and other purposes with a light shining on the board to ensure good visibility. The blue in the depth data was ~1.1m while the red was looking like ~1.16m (the device was at a slight tilt). I measured the distance from the board to the face of the device by hand and got 51.25" or 1.3m.

[image]
[image]

 I see what you mean about the different surfaces having different effects on the quality of depth data, however I've noticed a similar bias value on different surfaces. For example, the 1.9m I was talking about earlier was just pointing the L515 up at the ceiling which is that square foam-core/carboard ceiling tile which i suspect is rougher than the checkerboard.

The black floor of our lab absorbs a significant amount of the Lidar's light, which you can see in the depth data.

Thanks so much for your time!

MartyG-RealSense, 2022-01-20:
You are very welcome  :)

Yes, it is a general physics principle (not specific to RealSense) that dark gray and black colors absorb light, making it difficult for depth cameras to obtain detail from them.  Projecting a strong light onto such surfaces can help to bring out depth detail from them if it is required, like in https://github.com/IntelRealSense/librealsense/issues/6757 where a light-source is cast onto a black leather office chair.

In the recent case https://github.com/IntelRealSense/librealsense/issues/10131 the use of **apriltags** for L515 calibration instead of a checkerboard was discussed.  

If the scene is in a room with no natural sunlight and only artificial light-sources then the preset **No Ambient Light** could be tried.  Of the artificial light sources present, halogen lights and some LED lights may reduce L515 performance.

Cpeck71, 2022-01-20:
I should have mentioned that the images were used (and in my normal operations) with the "short range" setting. Our lab has a fair amount of overhead lighting, and I was using a spotlight to help visibility. I just did another test with the L515 a meter from the wall next to me (in a well lit space) and it was reading ~0.85m, using both "short range" and "no ambient light" settings. My point being that it seems the bias is always in that ~0.1-0.15 range.

The checkerboard has been used for detection and estimation of the extrinsic parameters of the camera to provide another sanity check on "truth" distance, similar to what I'd do with apriltags. Doing it that way provides similar measurements (~1.3 compared to ~1.1-1.15 in the depth data). So I'm thinking there's just some inherent bias going on and I'm curious if theres a way i can correct/calibrate the device itself or if i should just take it into account in my measurements.
Thanks!

P.s. this is completely unrelated to this problem and im sorry for hijacking this issue but I'm receiving a "control_transfer returned error, index 768, error: No data available, number: 61" error usually every second or so, but sometimes it arrives in bursts. I've looked at other issues talking about this and i suspect it might be the cable from the device to the computer. I know the devices are somewhat particular to the 3ft cable that it comes with, are there good replacements for this usb-c/usb cable that y'all recommend?

MartyG-RealSense, 2022-01-20:
The FAQ on Intel's official information website for the L515 states that depth error average at 1m distance from camera is <5mm at 95% reflectivity.  It recommends that the camera operates at an ambient temperature of 0 to 30 degrees C.

https://www.intelrealsense.com/lidar-camera-l515/

The link below has an L515 question about calibration where the user was reporting an error of up to 20 cm.  This seems consistent with the range of error that you are experiencing (1.3 m real distance versus 1.1 to 1.15 m depth measurement).  the RealSense team member in that case recommended use of the Short Range preset, which you have already been using.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000628961-How-to-calibrate-L515-

A small number of RealSense users calibrate with a tool called **Kalibr** instead of using official tools, and the links below discuss calibrating L515 with Kalibr.

https://github.com/rpng/open_vins/issues/147
https://github.com/ethz-asl/kalibr/issues/484

The company **Newnex** supply industrial grade USB cables and equipment that have been tested with RealSense cameras, though their web page does not mention the L515 model specifically.  I just tested that a 400 Series USB cable physically fits in an L515.  Newnex also supply premium USB cables in general.

https://www.newnex.com/realsense-3d-camera-connectivity.php

Cpeck71, 2022-01-20:
Thanks for the info! I'll take a look at these options and re-open the issue if necessary.

I really appreciate your time!

closed 2022-01-20 by Cpeck71

MartyG-RealSense, 2022-01-21:
You are very welcome.  :)  Please do feel free to re-open this issue at a future date if you have further questions.  Good luck!