realsenseai/librealsense #10170: D435I Cannot open motion module in SDK

ghhhgfgh, 2022-01-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Operating System & Version |  Linux ubuntu18.04 | 
| Kernel Version (Linux Only)    |  4.18.0-15-generic  | 
| Platform                                 | PC |
| SDK Version                            |  v2.50.0   |     | 

### Issue Description
Cannot open motion module in SDK. Some warnings below are shown in terminal.

 **20/01 19:04:19,287 WARNING [140141695858432] (backend-hid.cpp:631) iio_hid_sensor: Frames didn't arrived within the predefined interval
 20/01 19:04:19,290 WARNING [140141687465728] (backend-hid.cpp:631) iio_hid_sensor: Frames didn't arrived within the predefined interval**

Anyone knows what happens? There are one error when adding Repositories:
**Err:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease**

Are they related?

ghhhgfgh, 2022-01-20:
In the first installing, maybe I used the old one. When I add the keys, it shows no name, so I find the new one instruction to install:

**https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages**

So, the add key part is no mistakes:
**sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
Executing: /tmp/apt-key-gpghome.64s44WRrI3/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: public key ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" imported
gpg: Total number processed: 1
gpg:               imported: 1**

But I still have the error problem when adding repository

**Err:2 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease
  403  Forbidden [IP: 52.218.121.113 80]**

Is that something I should delete first (maybe the old key) to avoid the problems?

ghhhgfgh, 2022-01-20:
I delete the old key, and redo the steps. Although it still has the err, it works now in the SDK, and shows the IMU now. Thank you.

MartyG-RealSense, 2022-01-20:
Thanks very much @ghhhgfgh for the update - it's great news that it is working for you now!

MartyG-RealSense, 2022-01-27:
Hi @ghhhgfgh  Do you require further assistance with this case, please?  Thanks!

ghhhgfgh, 2022-01-28:
It seems everything is normal now. Thank you for your helping.

取得 iOS 版 Outlook<https://aka.ms/o0ukef>
________________________________
寄件者: MartyG-RealSense ***@***.***>
寄件日期: Thursday, January 27, 2022 3:54:57 PM
收件者: IntelRealSense/librealsense ***@***.***>
副本: ghhhgfgh ***@***.***>; Mention ***@***.***>
主旨: Re: [IntelRealSense/librealsense] D435I Cannot open motion module in SDK (Issue #10170)

Hi @ghhhgfgh<https://github.com/ghhhgfgh> Do you require further assistance with this case, please? Thanks!

MartyG-RealSense, 2022-01-28:
Thanks very much @ghhhgfgh for the update.  As the prolem is resolved, I will close this case.  Thanks again!

closed 2022-01-28 by MartyG-RealSense